The-OpenROAD-Project/OpenROAD-flow-scripts #1429: Create 0_prep.log to view logs related to design flow preparation

vijayank88, 2023-09-01:
### Description

ORFS flow starts recording logs from synthesis stage with `1_1_yosys.log`. Snippet of logs
```
1. Executing Verilog-2005 frontend: ./designs/src/gcd/gcd.v
2. Executing Liberty frontend.
3. Executing Liberty frontend.
4. Executing Liberty frontend.
```
But previously there is design flow preparation with `dont use cells` and creating `merged.lib`, it was missing part of logs.

It is useful when trying with different corners of ASAP7 and to make sure updated `DONT_USE_CELLS` processed properly or not.

### Suggested Solution

As we start synthesis with `1_1_yosys.log` either append platform preparation log in same file or create another new file `0_prep.log`. 
The following format should be captured in log.
```
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
./util/markDontUse.py -p "*x1p*_ASAP7* *xp*_ASAP7* SDF* ICG* DFFH*" -i platforms/asap7/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib
Opening file for replace: platforms/asap7/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib.gz
Marked 19 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib
./util/markDontUse.py -p "*x1p*_ASAP7* *xp*_ASAP7* SDF* ICG* DFFH*" -i platforms/asap7/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib
Opening file for replace: platforms/asap7/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib.gz
Marked 6 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib
./util/markDontUse.py -p "*x1p*_ASAP7* *xp*_ASAP7* SDF* ICG* DFFH*" -i platforms/asap7/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib
Opening file for replace: platforms/asap7/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib.gz
Marked 19 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib
./util/markDontUse.py -p "*x1p*_ASAP7* *xp*_ASAP7* SDF* ICG* DFFH*" -i platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib
Opening file for replace: platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz
Marked 19 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib
./util/markDontUse.py -p "*x1p*_ASAP7* *xp*_ASAP7* SDF* ICG* DFFH*" -i platforms/asap7/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib -o objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
Opening file for replace: platforms/asap7/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
Marked 22 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
./util/mergeLib.pl asap7_merged  ./objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib ./objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib > objects/asap7/gcd/base/lib/merged.lib
mkdir -p ./results/asap7/gcd/base ./logs/asap7/gcd/base ./reports/asap7/gcd/base
```
It can be reproduced with
```
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk synth
```

socclosure, 2024-08-15:
@maliberty 
This can be supported by adding below line

"puts "INFO: Lib Cell $cell is in the list of DONT_USE_CELLS so it will not be used."

in  flow/scripts/synth_preamble.tcl while processing list of DONT_USE_CELLS ( line# 75)
Can I make this change?

maliberty, 2024-08-15:
That is only part of the request and doesn't capture `creating merged.lib`.  I'm fine with your suggestion though.

socclosure, 2024-08-15:
@maliberty 
Thanks for pointing it out. Actually, synthesis flow is not merging liberty
 files instead, it is reading a merge lib specified by user or flow setting in 
config.mk of asap7 platform through DONT_USE_SC_LIB variable.

If user want to print which liberty & constraint files are read in abc then we can add below two lines

"
puts "INFO: abc will read $::env(DONT_USE_SC_LIB) liberty model."
puts "INFO: abc will read $::env(OBJECTS_DIR)/abc.constr constraint."
"
after "abc_args" are set (i.e. set abc_args [list -script $abc_script ... )

We can put

maliberty, 2024-08-15:
It seems the request is to capture what happens before yosys runs.  I'll let @vijayank88 comment on his request.

socclosure, 2024-08-16:
@vijayank88 , @maliberty 

If merge.lib file is not exists before synthesis run then Makefile is generating it and printing it's message in Standard output like below

"
Default PVT selection: BC model: NLDM
mkdir -p results/asap7/gcd/base/
echo 390 > results/asap7/gcd/base/clock_period.txt
/OpenROAD-flow-scripts/flow/util/preprocessLib.py -i /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib
Opening file for replace: /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib.gz
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib
/OpenROAD-flow-scripts/flow/util/preprocessLib.py -i /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib
Opening file for replace: /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib.gz
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib
/OpenROAD-flow-scripts/flow/util/preprocessLib.py -i /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib
Opening file for replace: /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib.gz
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib
/OpenROAD-flow-scripts/flow/util/preprocessLib.py -i /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz -o objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib
Opening file for replace: /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib
/OpenROAD-flow-scripts/flow/util/preprocessLib.py -i /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib -o objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
Opening file for replace: /OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib
**/OpenROAD-flow-scripts/flow/util/mergeLib.pl asap7_merged  ./objects/asap7/gcd/base/lib/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib  ./objects/asap7/gcd/base/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib ./objects/asap7/gcd/base/lib/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib > objects/asap7/gcd/base/lib/merged.lib**
mkdir -p ./results/asap7/gcd/base ./logs/asap7/gcd/base ./reports/asap7/gcd/base ./objects/asap7/gcd/base
"

Will that work?

vijayank88, 2024-08-19:
@socclosure 
With latest commit few files got renamed/updated. I am not able to reproduce logs I shown as per my issue template. Almost all terminal logs got printed in 1_1_yosys.log as it is.
Also in the log I can see dont use list as
```
abc -script PII/OpenROAD-flow-scripts/flow/scripts/abc_speed.script -liberty ./objects/asap7/gcd/base/lib/merged.lib -constr ./objects/asap7/gcd/base/abc.constr -dont_use *x1p*_ASAP7* -dont_use *xp*_ASAP7* -dont_use SDF* -dont_use ICG* -D 390
```
I believe this enough for me to get the lists.

socclosure, 2024-08-22:
@vijayank88 @maliberty 

It seems my last message might not be sent in readable format.

Can we close this issue?

maliberty, 2024-08-22:
Is there an associated pull request to resolve this?

socclosure, 2024-08-22:
I have made a small code change to print the "dont-use" cell list.

But I need some babysitting help on how to pull the
latest code from the master branch and push my changes.

Currently I have set up the OpenRoad flow on Docker.
Below is the output of "git status" command

"
***@***.***:/OpenROAD-flow-scripts/flow# git status
fatal: not a git repository (or any parent up to mount point
/OpenROAD-flow-scripts)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
"

I donot know how to resolve it.
Can any team member help me to setup the GIT work environment
of openRoad?

maliberty, 2024-08-22:
How did you create this area?  It seems like there is no git information.  Do you have a /OpenROAD-flow-scripts/.git directory?

socclosure, 2024-08-22:
Thanks.

I have installed openRoad suites by following instruction from the
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md

To Start the work I run below two commands
"
1: docker run --rm -it -u root -v /home/jdshah/openroad/OpenROAD-flow-scripts/flow:/OpenROAD-flow-scripts/flow \
  -e DISPLAY=${DISPLAY} \
  -v /tmp/.X11-unix:/tmp/.X11-unix  \
  -v ${HOME}/.Xauthority:/.Xauthority \
  --network host  \
  --security-opt seccomp=unconfined openroad/flow-ubuntu22.04-builder

2: source env.sh

"

Then I run "git status" which is giving errors.

Can you please guide what i am missing?

maliberty, 2024-08-22:
I don't think the .git directory is mounted inside the container.  Can you do `git status` outside the container?

socclosure, 2024-08-22:
Thanks.I can run “git status” and “git pull “ commandoutside the docker container.

socclosure, 2024-08-26:
@maliberty 
I need to commit a small change in synth_preamble.tcl to print a list of DO NOT USE cells.
How do i commit that changes to master branch?
Do i need create another git branch and submit the code change for PR to release it into master branch?

Please let me know process of submitting the change in flow's code .

maliberty, 2024-08-26:
You fork the repo and make the change in any branch in your repo.  From there you make a pull request, see https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

socclosure, 2024-09-02:
@maliberty 
It took long time to fork and create a pull request as my repository was not setup with required SSH & GPG keys. With online document and some struggle i am able to finish it.

I have filed a PR for this issue.
Please review and give your feedback.

socclosure, 2024-09-22:
@vijayank88 , @maliberty 
Please close or discard this issue as discussed earlier.

closed 2024-09-22 by maliberty