The-OpenROAD-Project/OpenROAD-flow-scripts #1434: asap7/riscv32i adder map logic and optimization with resizer

vijayank88, 2023-09-04:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

I am working on `asap7/riscv32i` design with different manual macro placement for lab exercise tutorial. Trying to fix timing violations with multiple variables.

The following max path not optimized during placement stage,
```
Startpoint: instr[4] (input port clocked by clk)
Endpoint: riscv/dp/_18865_ (falling edge-triggered flip-flop clocked by clk')
Path Group: clk
Path Type: max
```

With default setup timing reports as follows,

Placement: 3_4_place_resized.log
```
resizer pre report_tns
--------------------------------------------------------------------------
tns -2397732.00

===========================================
resizer pre report_wns
--------------------------------------------------------------------------
wns -2213.90

resizer report_tns
--------------------------------------------------------------------------
tns -53198.47

==========================================
resizer report_wns
--------------------------------------------------------------------------
wns -122.88
```

CTS report: 4_1_cts.log

```
cts post-repair check_setup
--------------------------------------------------------------------------

===========================================
cts post-repair report_tns
--------------------------------------------------------------------------
tns -53319.00

===========================================
cts post-repair report_wns
--------------------------------------------------------------------------
wns -123.62
```
Post Route Report: 6_report.log
```
finish report_tns
--------------------------------------------------------------------------
tns -40196.54

===========================================
finish report_wns
--------------------------------------------------------------------------
wns -96.24

```

If I remove [adder map logic](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/yoSys/cells_adders_R.v), there is no negative slack.

Placement: 3_4_place_resized.log
```
resizer pre report_tns
--------------------------------------------------------------------------
tns -4643115.50

============================================
resizer pre report_wns
--------------------------------------------------------------------------
wns -4067.38

resizer report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================
resizer report_wns
--------------------------------------------------------------------------
wns 0.00

===========================================
resizer report_worst_slack
--------------------------------------------------------------------------
worst slack 87.63
```

CTS report: 4_1_cts.log

```
cts post-repair report_tns
--------------------------------------------------------------------------
tns 0.00

============================================
cts post-repair report_wns
--------------------------------------------------------------------------
wns 0.00

===========================================
cts post-repair report_worst_slack
--------------------------------------------------------------------------
worst slack 113.32

```

Post Route Report: 6_report.log
```
finish report_tns
--------------------------------------------------------------------------
tns 0.00

============================================
finish report_wns
--------------------------------------------------------------------------
wns 0.00

============================================
finish report_worst_slack
--------------------------------------------------------------------------
worst slack 52.40

```

### Expected Behavior

So where the issue coming from?
1. Yosys `ADDER_MAP_FILE`?
2. Resizer has issue with optimization of FA/HA logic?

Fix resizer to meet better timing optimization. 

### Environment

```shell
With latest master
```

### To Reproduce

Use https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/asap7/riscv32i/config.mk and default setup end up slack violated.

If updated with `export ADDER_MAP_FILE :=` it meets positive slack

### Relevant log output

```shell
Updated in the description with both the logs
```

vijayank88, 2023-09-04:
cc/ @maliberty

openroadie, 2023-09-14:
@vijayank88 Can you please enlighten me on what the ADDER_MAP_FILE does? Does it actually use FA/HA cells?

maliberty, 2023-09-14:
@openroadie it is supposed to direct the techmapping of the fa/ha cells to tech specific primitives in yosys/abc.

openroadie, 2023-09-15:
@vijayank88 There seems to be only one full adder and one half adder in the library and if those have large delays then this result might be expected. The timing optimization pass currently does not look at mutli output cells like HA and FA. So I am not exactly sure what code can do about what to me looks like a limitation of the library. I would suggest that removing the ADDER_MAP_FILE (as you are already doing) is a valid path forward.

vijayank88, 2023-09-15:
@openroadie 
Thanks for the update. If it is limitation of library, we need to go with disabling ADDER_MAP_FILE or an other cell swapping to achieve the timing.

rovinski, 2023-09-15:
There was some point in time that it was discovered that mapping to FA/HA led to better results than not doing so. Things have changed so it may be worth revisiting it. It could also be enabled/disabled on a design-by-design basis.

As I recall, this is a two-fold problem. One is that once adders are mapped to FA/HA, Yosys doesn't really optimize them and cannot decide to remap them to other logic. Another is that the library is limited and may not have different drive strengths for FA/HA cells.

There is also a third issue which is that we don't understand Yosys's synthesized adder architecture. Some people have run experiments where manually curated adder architectures beat Yosys in both timing and area.

openroadie, 2023-09-18:
@rovinski Thanks for the insightful explanation. @vijayank88 I don't think there is any further action required on this issue for now. For your purposes, I would suggest not mapping to FA/HA.

closed 2023-09-18 by openroadie