The-OpenROAD-Project/OpenROAD-flow-scripts #1455: OpenROAD-Flow-scripts post synthesis Fails Specifically for ASAP7 Platform

Bynaryman, 2023-09-12:
### Subject

[Stage]: Synthesis.

### Describe the bug

Hello, I am experiencing an issue while running OpenROAD flow specifically for the asap7 platform. The flow works perfectly fine for other platforms like sky130 and nandgate45, but it fails during the ABC execution step when I try to use it for asap7. It works for none designs, and I tried with ubuntu20.04 and ubuntu22.04 dockers where I changed nothing in dockerfiles. I just cloned the repo, init submodules and run the install script.

The flow fails during the ABC execution with the following error:
```bash
18.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
18.1.1. Executing ABC.
ERROR: ABC: execution of command "/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-v6GVWs/abc.script 2>&1" failed: return code 134.
Command exited with non-zero status 1
Elapsed time: 0:04.37[h:]min:sec. CPU time: user 4.04 sys 0.17 (96%). Peak memory: 256512KB.
make: *** [Makefile:451: results/asap7/gcd/base/1_1_yosys.v] Error 1
[{WIFEXITED(s) && WEXITSTATUS(s) == 2}], 0, NULL) = 291632
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7fb06ce3e710}, {sa_handler=0x5568bcdc6f70, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7fb06ce3e710}, 8) = 0
ioctl(2, TIOCGWINSZ, {ws_row=64, ws_col=141, ws_xpixel=2256, ws_ypixel=2048}) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=291632, si_uid=1000, si_status=2, si_utime=3 /* 0.03 s */, si_stime=1 /* 0.01 s */} ---
wait4(-1, 0x7fff3d7e96d0, WNOHANG, NULL) = -1 ECHILD (No child processes)
rt_sigreturn({mask=[]})                 = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(2)                           = ?
+++ exited with 2 +++
```

### Expected Behavior

I expected the flow to complete successfully for the asap7 platform, similar to how it does for sky130 and nandgate45.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.2-arch1-1
os: Arch Linux 
cmake version 3.27.4
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 13.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10111-gb8c71cc9a
-- System name: Linux
-- Compiler: GNU 13.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.11 (found version "3.11.5") found components: Interpreter 
[... 85 lines cut ...]
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found SWIG: /usr/bin/swig (found suitable version "4.1.1", minimum required is "3.0")  
-- Using SWIG >= 4.1.1 -flatstaticmethod flag for python
-- Found Boost: /usr/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")  
-- boost: 1.83.0
-- Found Python3: /usr/include/python3.11 (found version "3.11.5") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /usr/lib/libz.so (found version "1.3")  
-- Configuring incomplete, errors occurred!
```

### To Reproduce

after proper installation, cd into flow folder then:
`OR_IMAGE=openroad/flow-ubuntu22.04-builder ./util/docker_shell make DESIGN_CONFIG=./designs/asap7/gcd/config.mk`

### Screenshots

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12585630/build_openroad.log)

vijayank88, 2023-09-12:
@vvbandeira 
The error reproduced with default `centos7` docker build also.

Bynaryman, 2023-09-12:
@vijayank88 your error seems to be related to the X-server. Mine seems to be about ABC processed being killed with an assertion failed and resulting in finishing the flow, however only for asap7 platform onyl this behavior is happening.

vvbandeira, 2023-09-13:
@Bynaryman 
What are the resources available for your docker containers? Note that depending on you native OS (where you installed docker) and the way it is configured, docker might not have access to all the CPU/RAM from you machine.

maliberty, 2024-03-06:
Are you still having issues?  If so please try with the latest code.

maliberty, 2024-03-13:
Please reopen if still relevant

closed 2024-03-13 by maliberty

EGelphman97, 2024-07-02:
Hi OpenROAD team,

I have experienced an identical issue with the asap7 pdk. I have run in the exact same environment using the exact same settings as Bynaryman. I have attempted to locate the bug in the source code but have failed. Any thoughts?