The-OpenROAD-Project/OpenROAD-flow-scripts #1456: coyote_tc macro placement fails

vijayank88, 2023-09-12:
### Subject

[Stage]: Floorplaning.

### Describe the bug

Working on sky130hd/coyote_tc design: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/designs/sky130hd/coyote_tc for upgrading IO PAD placement with latest `place_pad`.

If I use automatic macro placement with MPL runs infinitely with following message
```
[WARNING MPL-0061] Parquet area 567 x 5393.79 exceeds the partition area 144.9 x 1392.64.
[WARNING MPL-0061] Parquet area 567 x 5393.79 exceeds the partition area 144.9 x 1392.64.
```
Reproducible for MPL-0061: https://drive.google.com/drive/folders/1mvFyPlavMN5i_AZCnjf3ErUGf3LrTNlK?usp=sharing

If I use `RTLMP_FLOW` fails with following error.
```
[WARNING STA-0337] port 'en_i' not found.
HierRTLMP Flow enabled...
Call Macro Placer  -halo_width 40 -report_directory ./objects/sky130hd/coyote_tc/mpl1/rtlmp
Floorplan Outline: (0.0, 0.0) (5200.0, 4609.14),  Core Outline: (210, 212) (4989, 4387)
Traversed logical hierarchy
	Number of std cell instances : 152654
	Area of std cell instances : 1322824.25
	Number of macros : 15
	Area of macros : 6637524.00
	Area of macros with halos : 39717180.00
	Total area : 7960348.00
	Design Utilization : 0.40
	Core Utilization: 0.10
	Manufacturing Grid: 5

[ERROR MPL-0016] The instance area with halos 41040004.0 exceeds the core area 19952324.0
Error: macro_place_util.tcl, 136 MPL-0016
```
RTLMP test case: https://drive.google.com/file/d/1AuBDLVfyaLcC2ybm8JS0qHlAdY21M_l8/view?usp=drive_link

### Expected Behavior

Complete automatic macro placement

### Environment

```shell
With latest master
```

### To Reproduce

Refer the description

maliberty, 2023-09-12:
We are not enhancing the old macro placer anymore.  For rtlmp the error is quite clear:
```
[ERROR MPL-0016] The instance area with halos 41040004.0 exceeds the core area 19952324.0
```
You don't have enough area to place the macros.

vijayank88, 2023-09-12:
Yes. With default Halo and channel 40 & 80 spacing am getting error.

If I reduce to 10 Halo and channel spacing it is get struck at detail placement error.

maliberty, 2023-09-12:
Before we add a switch, have you looked at why it is required?  It sounds like something has gone wrong.

vijayank88, 2023-09-13:
> We are not enhancing the old macro placer anymore. For rtlmp the error is quite clear:
> 
> ```
> [ERROR MPL-0016] The instance area with halos 41040004.0 exceeds the core area 19952324.0
> ```
> 
> You don't have enough area to place the macros.

```
export DIE_AREA    = 0.0 0.0 5200 4609.14
export CORE_AREA   = 210 210 4990 4389.14
```
Above fixed size used. How much value have to be increased for die area based on error thrown by tool?
Basically how to convert that value and to increase the die size to fix that error?

vijayank88, 2023-09-13:
> Before we add a switch, have you looked at why it is required? It sounds like something has gone wrong.

I will fix the RTLMP issue, then based on error I'll open a new issue with test case.

vijayank88, 2023-09-19:
@maliberty 
Based on below information from RTLMP flow, how to calculate required die size?
```
	Number of std cell instances : 152654
	Area of std cell instances : 1322824.25
	Number of macros : 15
	Area of macros : 6637524.00
	Area of macros with halos : 39717180.00
	Total area : 7960348.00
	Design Utilization : 0.40
	Core Utilization: 0.10
	Manufacturing Grid: 5
```

vijayank88, 2023-09-20:
@maliberty 
I am working with latest OpenROAD head. With last week commit design completed macro placement with 5000x5000 itself.
With recent changes with mpl2 area required for macro placement never meet and failed at RTLMP stage itself.
Maybe following commit causes more area requirement to complete the macro placement:
https://github.com/The-OpenROAD-Project/OpenROAD/commit/b2dd9eb8094d159086f48abc94d2d9981ecc5e1f. After this changes coyote_tc never pass macro placement stage.

maliberty, 2023-09-20:
How are you past pad in this flow?  I'm still working on making pad pass.  The area check is new but I don't see how you could have gotten a good result before.

vijayank88, 2023-09-20:
@maliberty 
As per latest commits, created pad.tcl and now it's got stuck at pdn stage.
Mostly the issue related to pad placement only.
Here is pdn issue test case link https://drive.google.com/drive/folders/1QBQCwg8NAO2CBM6NF82ShIuKyVolUV4l

vijayank88, 2023-09-21:
@maliberty 
I just merged latest commits from ORFS and ran the flow. It was failed at pad placement itself.
```
[WARNING STA-0335] pin 'u_clk.u_in/IN' not found.
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[WARNING STA-0337] port 'en_i' not found.
[WARNING STA-0337] port 'en_i' not found.
[WARNING STA-0337] port 'en_i' not found.

==========================================================================
Floorplan check_setup
--------------------------------------------------------------------------
Warning: There is 1 input port missing set_input_delay.
Warning: There are 53469 unclocked register/latch pins.
Warning: There are 55151 unconstrained endpoints.
number instances in verilog is 319543
[WARNING IFP-0028] Core area lower left (250.000, 250.000) snapped to (250.240, 250.240).
[INFO IFP-0001] Added 1506 rows of 10216 site unithd with height 1.
[ERROR PAD-0026] Filling IO_WEST (4380.000um -> 4398.035um) will result in a gap.
Error: pad.tcl, 242 PAD-0026
Elapsed time: 0:38.74[h:]min:sec. CPU time: user 37.25 sys 1.37 (99%). Peak memory: 1312012KB.
make[1]: *** [Makefile:544: do-2_1_floorplan] Error 1
make: *** [Makefile:544: results/sky130hd/coyote_tc/base/2_1_floorplan.odb] Error 2
```
Also clock pin not found error `[WARNING STA-0335] pin 'u_clk.u_in/IN' not found.`. So timing report not properly generated post synthesis.

maliberty, 2023-09-21:
I'm aware that it still needs work.  This was just getting it past pad/pdn.

vijayank88, 2023-09-22:
@maliberty 
I have two different directory installed with ORFS. One with latest OR head and other one default ORFS setup.
With latest OR head am not getting any error while running coyote_tc and failing at GRT stages as  congestion issue. 
GRT Test case attached: https://drive.google.com/file/d/1SB639IkNac2bJo86HiMaizWTitY9Rvhc/view?usp=drive_link

Other one with default ORFS setup failing at pad placement itself with following error.
```
[ERROR PAD-0026] Filling IO_WEST (4380.000um -> 4398.035um) will result in a gap.
Error: pad.tcl, 242 PAD-0026
```
Are you facing above IO filling error?

maliberty, 2023-09-22:
You need the latest to get a big fix

closed 2024-03-11 by maliberty