The-OpenROAD-Project/OpenROAD-flow-scripts #1467: `./build_openroad.sh` fails because "git -C" is not supported inside Docker

oharboe, 2023-09-14:
### Subject

[Documentation] for documentation errors.

### Describe the bug

```
$ git status
HEAD detached at origin/master
nothing to commit, working tree clean
$ ./build_openroad.sh 
[ 89%] Building techlibs/sf2/synth_sf2.o
[ 99%] Building yosys-config
[ 99%] Building abc/abc-bafd2a7
Unknown option: -C
usage: git [--version] [--help] [-c name=value]
           [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p|--paginate|--no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           <command> [<args>]
ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!
make: *** [Makefile:745: abc/abc-bafd2a7] Error 1
make: *** Waiting for unfinished jobs....
[ 99%] Building passes/techmap/filterlib.o
passes/techmap/abc.cc: In member function 'virtual void {anonymous}::AbcPass::execute(std::vector<std::basic_string<char> >, Yosys::RTLIL::Design*)':
passes/techmap/abc.cc:1825:15: warning: 'g_argidx' may be used uninitialized in this function [-Wmaybe-uninitialized]
      cmd_error(args, g_argidx, stringf("Unsupported gate type: %s", g.c_str()));
      ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: Leaving directory '/OpenROAD-flow-scripts/tools/yosys'
The command '/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}' returned a non-zero code: 2
```

### Expected Behavior

It should build

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-32-generic
os: Ubuntu 23.04 (Lunar Lobster)
cmake version 3.25.1
-- The CXX compiler identification is GNU 12.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10111-gb8c71cc9a
-- System name: Linux
-- Compiler: GNU 12.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 12.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3 (found version "3.11.4") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
[... 113 lines cut ...]
-- Configuring incomplete, errors occurred!
```

### To Reproduce

See above.

### Relevant log output

```shell
$ git status
HEAD detached at b8c71cc9a
You are currently bisecting, started from branch '68113315c'.
  (use "git bisect reset" to get back to the original branch)

nothing to commit, working tree clean
$ etc/Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-32-generic
os: Ubuntu 23.04 (Lunar Lobster)
cmake version 3.25.1
^C
```

oharboe, 2023-09-15:
Hmmm... running " rm -rf tools/yosys/abc" fixed the issue...  Why would there be files in "tools/yosys/abc"??

rovinski, 2023-09-15:
That's where the Yosys Makefile builds abc.

oharboe, 2023-09-16:
> That's where the Yosys Makefile builds abc.

I see. So somehow that folder got poisoned and deleting it fixed the "./build_openroad.sh" docker build command.

Strangely, "git clean -fdx && git submodule foreach git clean -fdx ." didn't clean out that folder...

rovinski, 2023-09-17:
Yeah, I've run into that struggle before. I don't know why it doesn't get cleaned.

oharboe, 2023-09-17:
> Yeah, I've run into that struggle before. I don't know why it doesn't get cleaned.

`git submodule foreach` does enter the tools/yosys folder:

```
$ git submodule foreach pwd
Entering 'tools/LSOracle'
/home/oyvind/docker-ORFS/tools/LSOracle
Entering 'tools/OpenROAD'
/home/oyvind/docker-ORFS/tools/OpenROAD
Entering 'tools/yosys'
/home/oyvind/docker-ORFS/tools/yosys
```

oharboe, 2023-09-17:
```
$ ./build_openroad.sh --latest
[deleted, abc was also built here]
-- Installing: /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
-- Set runtime path of "/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad" to "/OpenROAD-flow-scripts/dependencies/lib:/OpenROAD-flow-scripts/dependencies/lib64:/opt/or-tools/lib64:/opt/or-tools/lib"
Removing intermediate container bd48610af1dd
 ---> ac4d3762e1ab
Successfully built ac4d3762e1ab
Successfully tagged openroad/flow-centos7-builder:latest
$ ls tools/yosys/abc
ls: cannot access 'tools/yosys/abc': No such file or directory
```

So tools/yosys/abc is not needed to build the docker image...

oharboe, 2023-09-17:
And if I try to create a file in tools/yosys/abc/def, it is cleaned up...

```
$ mkdir tools/yosys/abc
$ touch tools/yosys/abc/def
$ git submodule foreach git clean -fdx .
Entering 'tools/LSOracle'
Entering 'tools/OpenROAD'
Entering 'tools/yosys'
Removing abc/
$ ls tools/yosys/abc
ls: cannot access 'tools/yosys/abc': No such file or directory
```

oharboe, 2023-09-17:
So here is a sequence that creates `tools/yosys/abc` that is not cleaned out, after which `./build_openroad.sh` fails:

```
$ ./build_openroad.sh --local
$ git submodule foreach git clean -fdx .
$ ls tools/yosys/abc/
abc-bafd2a7  abc.rc        CMakeLists.txt  i10.aig   readmeaig
abcexe.dsp   abcspace.dsw  copyright.txt   lib       README.md
abclib.dsp   arch_flags.c  depends.sh      Makefile  src
```

oharboe, 2023-09-17:
Something copies abc into tools/yosys/abc and, I think, because tools/yosys/abc/.git exists, then the tools/yosys/abc folder is not cleaned out by `git clean -fdx .`

```
$ git submodule
 cf0a0eb29523cde8664e9313f31d7f7406d35d31 tools/LSOracle (v0.3.0-rc1-54-gcf0a0eb2)
 b8c71cc9a5bdcc182f98e06e2e473241eef5ecf5 tools/OpenROAD (v2.0-10111-gb8c71cc9a)
 bc027b2cae9a85b887684930705762fac720b529 tools/yosys (yosys-0.13-16-gbc027b2ca)
$ cd tools/yosys
$ git submodule
$ ls abc/.git
branches  description  HEAD   index  logs     packed-refs
config    FETCH_HEAD   hooks  info   objects  refs
```

oharboe, 2023-09-17:
@maliberty @vvbandeira I now understand the series of unfortunate events that leads to a mysterious failure when trying to build the Docker image:

1. Run `./build_openroad.sh --local`. When Yosys is built, it clones tools/yosys/abc
2. Now run `./build_openroad.sh` to build the docker image, this fails because yosys uses `git -C`, which is not supported by the git version in the docker image, to check if tools/yosys/abc is the right version.

Proposed solution: upgrade git in the docker image so that it supports `git -C`.

closed 2024-03-20 by oharboe