The-OpenROAD-Project/OpenROAD-flow-scripts #1485: Unable to Run Autotuner due to "adarray" not being an attribute in Numpy?

dalyles, 2023-09-21:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I'm currenlty running the Autotuner but have been impeded by this message. I didn't modify any script files either.

```
:~/OpenROAD-flow-scripts/flow/util$ python3 distributed.py --design picorv32 --platform sky130hs  --config ../designs/sky130hs/picorv32/autotuner.json tune
Traceback (most recent call last):
  File "/home/dlyles/OpenROAD-flow-scripts/flow/util/distributed.py", line 887, in <module>
    config_dict, SDC_ORIGINAL, FR_ORIGINAL = read_config(abspath(args.config))
  File "/home/dlyles/OpenROAD-flow-scripts/flow/util/distributed.py", line 320, in read_config
    config[key] = read_tune(value)
  File "/home/dlyles/OpenROAD-flow-scripts/flow/util/distributed.py", line 251, in read_tune
    np.adarray.tolist(
  File "/home/dlyles/mambaforge/lib/python3.10/site-packages/numpy/__init__.py", line 320, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'adarray'. Did you mean: 'asarray'?
```

Why would an error like this show up when there has not been an inssue in the past?

Thank you

### Expected Behavior

The Autotuner should execute.

### Environment

```shell
.
```

### To Reproduce

To reproduce I used this command: <code>python3 distributed.py --design picorv32 --platform sky130hs  --config ../designs/sky130hs/picorv32/autotuner.json tune</code>

vvbandeira, 2023-09-21:
It looks like you are using `mambaforge`, a Python package manager. Are you sure you have installed the requirements in the active environment from which you are trying to call the `distributed.py` script?

dalyles, 2023-09-21:
From my understanding, I always source <code>env.sh</code> so that everything runs smoothly. This is the first time I've come across this issue in all the times I've been running <code>distributed.py</code>

dalyles, 2023-09-21:
I did an update on numpy in <code>Mambaforge</code> however it still gives the same messsage.

luarss, 2024-03-15:
@dalyles This seems like a syntax error. Can you try `np.asarray` in place of `np.adarray`?

closed 2024-04-01 by vvbandeira