The-OpenROAD-Project/OpenROAD-flow-scripts #1489: Unable to finish gcd design on sky130hd_fakestack technology

wponghiran, 2023-09-21:
### Subject

[Stage]: GDS generation.

### Describe the bug

All expect the last stage `finishing` runs successfully for `gcd` design on `sky130hd_fakestack` technology. The log file indicate  

```bash
OpenROAD v2.0-9542-ge5a7ad7ae 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO] Deleted 0 routing obstructions
[INFO RCX-0431] Defined process_corner X with ext_model_index 0
[INFO RCX-0029] Defined extraction corner X
[INFO RCX-0008] extracting parasitics of gcd ...
[INFO RCX-0435] Reading extraction model file ./platforms/sky130hd_fakestack/rcx_patterns.rules ...
Signal 11 received
Stack trace:
 0# 0x0000556BD062E57B in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F483C4CD520 in /lib/x86_64-linux-gnu/libc.so.6
 2# rcx::extDistWidthRCTable::getWidthIndex(unsigned int) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# rcx::extMain::calcRes0(double*, unsigned int, unsigned int, unsigned int, int, int) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# rcx::extMain::getResCapTable() in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# rcx::extMain::makeBlockRCsegs(char const*, unsigned int, unsigned int, double, bool, double, int, char const*) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# rcx::Ext::extract(rcx::Ext::ExtractOptions) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# 0x0000556BD0C512D2 in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007F4840365028 in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# ord::tclAppInit(Tcl_Interp*) in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# 0x00007F483C4B4D90 in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in /home/wponghir/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
```

### Expected Behavior

Finish this stage with a report on area and timing. 
Running the same design on similar  `sky130hd` and  `sky130hs`  technology complete without any issue.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9542-ge5a7ad7ae
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
[... 110 lines cut ...]
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.3.9")
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1")
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.evJqB4yw07
```

### To Reproduce

`make DESIGN_CONFIG=./designs/sky130hd_fakestack/ariane133/config.mk`

wponghiran, 2023-09-21:
Freshly pulled and built OpenROAD on Red Hat 8 is also has the same problem.

maliberty, 2023-09-21:
@ravi-varadarajan do you have any plans to finish this off or should we just remove it?