The-OpenROAD-Project/OpenROAD-flow-scripts #1492: equivalence check failed for nangate/gcd

vijayank88, 2023-09-25:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

With latest ORFS docker installation, equivalence check failed during cts stage.
```
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 110   File "/usr/local/bin/eqy", line 520
    if m := p.regex.fullmatch(name):
          ^
SyntaxError: invalid syntax
Elapsed time: 0:06.63[h:]min:sec. CPU time: user 6.54 sys 0.09 (99%). Peak memory: 84852KB.
make[1]: *** [do-4_1_cts] Error 1
make: *** [results/nangate45/gcd/base/4_1_cts.odb] Error 2
```

### Expected Behavior

Pass equivalence check

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10367-g36112de99
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
[... 108 lines cut ...]
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.JdSCmS9duU
```

### To Reproduce

Update installation with docker.
Run `make` inside docker.

### Relevant log output

```shell
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 110   File "/usr/local/bin/eqy", line 520
    if m := p.regex.fullmatch(name):
          ^
SyntaxError: invalid syntax
Elapsed time: 0:06.63[h:]min:sec. CPU time: user 6.54 sys 0.09 (99%). Peak memory: 84852KB.
make[1]: *** [do-4_1_cts] Error 1
make: *** [results/nangate45/gcd/base/4_1_cts.odb] Error 2
```

vvbandeira, 2023-09-25:
Python 3.8 is required for `eqy`

closed 2023-09-25 by vvbandeira

vijayank88, 2023-09-25:
But for docker dependencies it should be handled automatically right?
Also python3 default to python3.8 only
@vvbandeira

vvbandeira, 2023-09-25:
@vijayank88 
Yes, docker handles that via the dependency installer. Ubuntu already ships python > 3.8 and for Centos7:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh#L377

vijayank88, 2023-09-26:
seems you're right. For Centos7 docker installation still it uses `python3.6.8`
```
bash-4.2$ which python3
/usr/bin/python3
bash-4.2$ python3
Python 3.6.8 (default, Jun 20 2023, 11:53:23) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-44)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
Needs update for Centos7 docker instalaltion.