The-OpenROAD-Project/OpenROAD-flow-scripts #1500: Antenna ANT-0011 Error

dalyles, 2023-09-26:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I get the following error messages when I run <code>make</code> on this circuit (platform is nangate45):

```
==========================================================================
check_antennas
--------------------------------------------------------------------------
[WARNING ANT-0011] -report_violating_nets is deprecated.
[INFO GRT-0043] No OR_DEFAULT vias defined.
Signal 11 received
Stack trace:
 0# 0x0000556213D66288 in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F5EB0C2B090 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbTechLayer::getRoutingLevel() in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# ant::AntennaChecker::findWireRoots(odb::dbWire*, std::vector<odb::dbWireGraph::Node*, std::allocator<odb::dbWireGraph::Node*> >&, std::vector<odb::dbWireGraph::Node*, std::allocator<odb::dbWireGraph::Node*> >&) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# ant::AntennaChecker::checkNet(odb::dbNet*, bool, bool, std::basic_ofstream<char, std::char_traits<char> >&, int&, int&) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# ant::AntennaChecker::checkAntennas(odb::dbNet*, bool) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# ant::check_antennas(char const*, bool) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# 0x0000556214887E04 in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# TclNRRunCallbacks in /home/dlyles/mambaforge/lib/libtcl8.6.so
 9# 0x00007F5EB4824286 in /home/dlyles/mambaforge/lib/libtcl8.6.so
10# Tcl_EvalEx in /home/dlyles/mambaforge/lib/libtcl8.6.so
11# Tcl_Eval in /home/dlyles/mambaforge/lib/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# ord::tclAppInit(Tcl_Interp*) in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# Tcl_MainEx in /home/dlyles/mambaforge/lib/libtcl8.6.so
15# main in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
Elapsed time: 33:25.75[h:]min:sec. CPU time: user 1999.37 sys 3.77 (99%). Peak memory: 6988088KB.
make: *** [Makefile:633: results/nangate45/FFT10_MixedRadix/base/5_1_grt.odb] Error 139
```

### Expected Behavior

The behavior expected should be an ininterrupted <code>make</code> flow.

### Environment

```shell
.
```

### To Reproduce

To reproduce the error: please use the following files:

RTL FILE: https://socks.lbl.gov/mvega/fftdesigns/-/blob/main/od_fft/fftmr/in_progress/FFT10_MixedRadix.v

<code>config.mk</code> file
```
export DESIGN_NAME = FFT10_MixedRadix
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/FFT10_MixedRadix.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
#export FASTROUTE_TCL = ./fastroute.tcl
export ABC_AREA      = 1

# Adders degrade GCD
export ADDER_MAP_FILE :=

export CORE_UTILIZATION ?= 55
export PLACE_DENSITY_LB_ADDON = 0.20
export TNS_END_PERCENT = 100

```

<code>constraint.sdc</code> file
```
current_design FFT10_MixedRadix
  
set clk_name  core_clock
set clk_port_name clk
set clk_period 2
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]

create_clock -name $clk_name -period $clk_period $clk_port

set non_clock_inputs [lsearch -inline -all -not -exact [all_inputs] $clk_port]

set_input_delay  [expr $clk_period * $clk_io_pct] -clock $clk_name $non_clock_inputs
set_output_delay [expr $clk_period * $clk_io_pct] -clock $clk_name [all_outputs]
```

vijayank88, 2023-09-26:
@dalyles 
Instead of giving repo path and `config.mk` to reproduce the issue simply run `make global_route_issue` and attach test case.
Are you using latest master?

dalyles, 2023-09-26:
I tried using the latest master, however it doesn't seem like everything was successfullly pulled. I found out how to do the update on the latest master and will do a rerun and get back to you.

vijayank88, 2023-09-26:
@dalyles 
Initial check update constraint file with clk_port_name as per top level clock port `input         clock,` not `clk`

dalyles, 2023-09-27:
I did this change and the fun successfully goes through.
Thanks again

vijayank88, 2023-09-27:
@dalyles 
I have made following changes with config.mk
```
export CORE_UTILIZATION ?= 35
export PLACE_DENSITY_LB_ADDON = 0.10
```
In constraint.sdc
```
set clk_name  core_clock
set clk_port_name clock
set clk_period 10
```
Flow completed with
```
finish report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
finish report_wns
--------------------------------------------------------------------------
wns 0.00

==========================================================================
finish report_worst_slack
--------------------------------------------------------------------------
worst slack 4.34
```
Now you can tighten constraints for your requirements. Antenna issue not reproduced, hence closing this.

vvbandeira, 2023-09-27:
@dalyles 
If you can still reproduce the error, please upload the tar file generated by `make global_route_issue`, crashes are always undesirable. With a proper test case @eder-matheus can turn this into an error, which is preferred. Thanks.