The-OpenROAD-Project/OpenROAD-flow-scripts #1505: ORFS dependencies installer fails

vijayank88, 2023-09-27:
### Subject

[Build] for compilation-related issues.

### Describe the bug

As there are update with `DependencyInstaller.sh` trying to run `sudo ./setup.sh`. It fails with following error.
```
Loaded plugins: fastestmirror, langpacks, product-id, search-disabled-repos
Cleaning repos: base centos-sclo-rh centos-sclo-sclo docker-ce-stable epel extras google-cloud-cli google-cloud-sdk google-compute-engine ius updates
Cleaning up list of fastest mirrors
To enable GCC-8 or Clang-7 you need to run:
    source /opt/rh/devtoolset-8/enable
    source /opt/rh/llvm-toolset-7.0/enable
/var/tmp/scljEWWjs: line 8: -u: command not found
```
Seems following line not executed
```
sudo -u $SUDO_USER "$DIR/etc/DependencyInstaller.sh" -common -prefix="$DIR/dependencies"
```
Also with clean setup, dependencies directory not created. 
@oharboe are facing similar issue with latest setup?

### Expected Behavior

`sudo ./setup.sh` should create `dependencies` directory with all required packages.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10111-gb8c71cc9a
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /opt/rh/rh-python38/root/usr/bin/python3.8 (found version "3.8.13") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
[... 103 lines cut ...]
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.OY4Z6Or2Je
```

### To Reproduce

sudo ./setup.sh

### Relevant log output

```shell
/var/tmp/scljEWWjs: line 8: -u: command not found
```

vvbandeira, 2023-09-27:
@vijayank88 
Are you sure you are on the latest code?

> [WARNING] Your current OpenROAD version is outdated.

And your log shows:

```
Cleaning up list of fastest mirrors
To enable GCC-8 or Clang-7 you need to run:
    source /opt/rh/devtoolset-8/enable
    source /opt/rh/llvm-toolset-7.0/enable
```

When a message for Python 3.8 is also expected, see https://github.com/The-OpenROAD-Project/OpenROAD/blob/37fa5450be8ff02e1f5591bb774ef54164404495/etc/DependencyInstaller.sh#L667-L671

vijayank88, 2023-09-27:
In Centos7 all above set in `.bashrc` to source automatically.

Here issue I'm facing is while running `sudo ./setup.sh` and unable to run `sudo -u $SUDO_USER "$DIR/etc/DependencyInstaller.sh" -common -prefix="$DIR/dependencies"` this command.

vvbandeira, 2023-09-27:
@vijayank88 
I am not saying to source these files; I am saying that the console output you posted here leads me to believe you are on an older version of ORFS/OR. I cannot reproduce it on a clean system.

vijayank88, 2023-09-27:
Till now I'm not monitored that `dependencies` or ran `sudo ./setup.sh`.
Due to addition of equivalence check I ran `sudo ./setup.sh`, but it is not creating `dependencies` directory both GCP cloud and my Centos7 laptop.
Sent you GCP machine path, please check once. Thanks.

vvbandeira, 2023-09-27:
This is not related to ORFS, likely due to some env or machine setup. Will handle it offline.

closed 2023-09-27 by vvbandeira