The-OpenROAD-Project/OpenROAD-flow-scripts #1507: OpenROAD `./etc/DependencyInstaller.sh` installs `yosys-0.33` while flow uses different version

mithro, 2023-09-27:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

OpenROAD's `./etc/DependencyInstaller.sh` installs `yosys-0.33`;

https://github.com/The-OpenROAD-Project/OpenROAD/blob/347459a5c91f473de611c0c0f403eb4880e8d339/etc/DependencyInstaller.sh#L12-L13

While the OpenROAD flow script seems to use a version [bc027b2](https://github.com/The-OpenROAD-Project/yosys/commit/bc027b2cae9a85b887684930705762fac720b529) [on Jan 19, 2022](https://github.com/The-OpenROAD-Project/yosys/commit/bc027b2cae9a85b887684930705762fac720b529) see;

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/tools

### Expected Behavior

`OpenROAD` and `OpenROAD-flow-scripts` use the same version of Yosys.

### Environment

```shell
N/A
```

### To Reproduce

N/A

mithro, 2023-09-27:
It also seems that the `OpenROAD-Flow-Scripts` is using https://github.com/The-OpenROAD-Project/yosys rather than https://github.com/YosysHQ/yosys

mithro, 2023-09-27:
One attempt at a fix at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1509

mithro, 2023-09-27:
Given that `OpenROAD-flow-scripts` has a  [`./etc/DependencyInstaller.sh`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/etc/DependencyInstaller.sh) which calls the OpenROAD version, it is unclear why `tools/yosys` is needed?

vvbandeira, 2023-09-27:
@mithro 
We initially added `tools/yosys` to complete ORFS RTL-GDSII (similar to `klayout`). Since `yosys` did not have a bin release and `klayout` did, `klayout` was added as a dependency in the script and `yosys` to the `tools/`.
Unfortunately, we were not diligent in making regular updates to `yosys`, and now the latest version of it breaks some designs in the ORFS CI, see #1468.
Recently, we added an equivalence check (i.e., `eqy`), which depends on a newer version of `yosys` than the one in `tools/yosys`. To mitigate this and not make it a blocker to add `eqy` we decided that until we can merge #1468, ORFS would keep using the old version that works for CI, and we would add a newer version in the dependency script to satisfy `eqy`. We know this is not ideal, but it was the best option that would not block `eqy` or break CI.

rovinski, 2023-09-28:
> not diligent in making regular updates to `yosys`

To be more precise, we tried updating several times over the past few years, but in each instance we found some severe degradations in the Yosys output that we ultimately didn't have the resources to investigate.

If the output of the current version is at least somewhat reasonable along the PPA curve, it is probably best to just update to the latest assuming it doesn't completely mess up the CI trends.

JuanSebastianMoya7, 2023-09-29:
Hi!
I have tried to install the Open Road flow scripts on two different PCs, one locally and the other as a Docker, and have the same error after running the make command for the nangate45 default project. Follows the error that appeared in both cases.
[image]

vijayank88, 2023-09-30:
@JuanSebastianMoya7 
I am also facing similar issue. Working on the fix. https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1513 follow-up GH issue.

JuanSebastianMoya7, 2023-10-02:
> @JuanSebastianMoya7 I am also facing similar issue. Working on the fix. #1513 follow-up GH issue.

Hi @vijayank88, I just followed the instructions that you proposed on #1513 and #4035 (https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035) . I downloaded the pre-built package and I set the environment as mentioned (export PATH="<extracted_location>/oss-cad-suite/bin:$PATH"). I run the Openroad make command on the Docker but I still get the same error. Do I have to uninstall yosys-0.33 (the actual version) or do I have to run a make command or a yosys command in the oss-cad-suite folder where the pre-buillt package is?

vijayank88, 2023-10-03:
@JuanSebastianMoya7 
have you set following variables before running make command?
```
# these variables are used in flow/Makefile. Do make sure the yosys path is sourced.
export OPENROAD_EXE=$(command -v openroad)
export YOSYS_CMD=$(shell command -v yosys)
```

JuanSebastianMoya7, 2023-10-03:
Hi @vijayank88 , the flow is running. 

No, I use the following commands.

docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow -v /home/user/Downloads/oss-cad-suite:/oss-cad-suite openroad/flow-centos7-builder

(the command above runs the make file correctly, it links the oss-cad-suite folder that I downloaded from the page you suggested to the /oss-cad-suite folder on the docker.) Now it runs correctly.

To get access to the GUI, the command that I used is:

docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow -v /home/user/Downloads/oss-cad-suite:/oss-cad-suite -e DISPLAY=${DISPLAY} -v /tmp/.X11-unix:/tmp/.X11-unix -v ${HOME}/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined openroad/flow-centos7-builder

and then at the end, I run the make gui_final.

Thanks for the help and support!

JuanSebastianMoya7, 2023-10-06:
> Note that we do provide a docker_shell script that is similar - see https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/README.md

Hi @maliberty, thank you for the information. I will explore that docker_shell script.

Kind regards,

maliberty, 2024-03-11:
@vvbandeira now that we have updated yosys is there more work to here?

vvbandeira, 2024-07-22:
OR and ORFS versions match now.

closed 2024-07-22 by vvbandeira