The-OpenROAD-Project/OpenROAD-flow-scripts #1513: GCD Nangate45 fails to run on fresh ORFS install

gudeh, 2023-09-28:
### Subject

[Stage]: CTS.

### Describe the bug

I simply did a fresh install of ORFS current version with the following commands:

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh

No error messages.
Than I did:

cd flow
make

And it fails to synthesize the example circuit:

[INFO RSZ-0094] Found 43 endpoints with setup violations.
[INFO RSZ-0040] Inserted 19 buffers.
[INFO RSZ-0041] Resized 80 instances.
[INFO RSZ-0043] Swapped pins on 31 instances.
[INFO RSZ-0049] Cloned 2 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 110 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 0:05.12[h:]min:sec. CPU time: user 5.08 sys 0.03 (99%). Peak memory: 110012KB.
make[1]: *** [Makefile:669: do-4_1_cts] Error 1
make: *** [Makefile:669: results/nangate45/gcd/base/4_1_cts.odb] Error 2

### Expected Behavior

Successfully synthesize GCD with nangate45.

### Environment

```shell
gudeh@snowflake:~/Desktop/OpenROAD-flow-scripts/flow$ ../tools/OpenROAD/etc/Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-33-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10427-g347459a5c
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[... 102 lines cut ...]
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.3.9") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.G3hesCo5vD
```

### To Reproduce

Run the sequence of commands in: https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

gudeh, 2023-09-28:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12753831/build_openroad.log)

vijayank88, 2023-09-30:
@gudeh 
Can you check whether `/usr/local/bin/eqy` exists?

gudeh, 2023-09-30:
> @gudeh Can you check whether `/usr/local/bin/eqy` exists?

No! There is no such directory:

```
gudeh@snowflake:~$ ls /usr/local/bin/
ccache-swig  cpack          lemon-0.x-to-1.x.sh  swig          yosys-filterlib
ccmake       ctest          lgf-gen              yosys         yosys-smtbmc
cmake        dimacs-solver  nsys                 yosys-abc     yosys-witness
cmake-gui    dimacs-to-lgf  nsys-ui              yosys-config

```

vijayank88, 2023-09-30:
Until it get fixed in master, go with pre-built binaries to use yosys/eqy/sby. Check my comment here for Ubuntu20/22: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035#issuecomment-1734753815

revollllt, 2023-11-04:
I am also experiencing the same issue. If you have solved this problem, could you please share your solution with me? I will be grateful.

gudeh, 2023-11-06:
Hi @revollllt, I downloaded the pre-built of yosys as indicated in the comment: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035#issuecomment-1734753815 . I downloaded the directory and set the path on my Linux environment, the installation says to do a 'export PATH="<extracted_location>/oss-cad-suite/bin:$PATH"', but I modified the file in ~/.bashrc for a permanent change.

I am not sure if the current version of ORFS already fixed this.

revollllt, 2023-11-06:
It really works for me. Thanks. 

> Hi @revollllt, I downloaded the pre-built of yosys as indicated in the comment: [The-OpenROAD-Project/OpenROAD#4035 (comment)](https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035#issuecomment-1734753815) . I downloaded the directory and set the path on my Linux environment, the installation says to do a 'export PATH="<extracted_location>/oss-cad-suite/bin:$PATH"', but I modified the file in ~/.bashrc for a permanent change.
> 
> I am not sure if the current version of ORFS already fixed this.

vvbandeira, 2024-01-16:
@gudeh @revollllt
Could you try again with the latest code? Please also make sure to `source env.sh` before running the flow if you used `sudo ./setup.sh` to install the dependencies.

gudeh, 2024-01-16:
Yes, I confirm the issue was fixed. It worked after doing `source env.sh` and using `sudo ./setup.sh`.

closed 2024-01-16 by gudeh

gudeh, 2024-01-16:
@vvbandeira Just some further information. Previously I was able to do `cd flow` and `make` without having to `source ./env.sh`. Also, we have these messages regarding elapsed time and eqy by the end of the flow run, this is for standard gcd with nangate45:
```
Log                       Elapsed seconds
1_1_yosys                          1
3_3_place_gp                       1
4_1_cts                            9
No elapsed time found in logs/nangate45/gcd/base/4_equivalence_check.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/combine.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/gate.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/gold.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/partition.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_0_clk.A/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_0_clk.Z/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_0__f_clk.A/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_0__f_clk.Z/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_1__f_clk.A/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_1__f_clk.Z/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_2__f_clk.A/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_2__f_clk.Z/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_3__f_clk.A/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clkbuf_2_3__f_clk.Z/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clknet_0_clk/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clknet_2_0__leaf_clk/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clknet_2_1__leaf_clk/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clknet_2_2__leaf_clk/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.clknet_2_3__leaf_clk/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.ctrl.state.out.1/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.ctrl.state.out.2/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.0/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.1/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.10/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.11/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.12/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.13/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.14/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.15/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.2/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.3/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.4/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.5/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.6/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.7/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.8/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin0.9/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.0/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.1/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.10/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.11/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.12/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.13/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.14/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.15/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.2/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.dpath.a_lt_bin1.3/basic/run.log
[... cut ...]
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.resp_msg.8/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.resp_msg.9/basic/run.log
No elapsed time found in logs/nangate45/gcd/base/4_eqy_output/strategies/gcd.resp_val/basic/run.log
5_1_grt                            2
5_3_route                         10
6_1_merge                          1
6_report                           1
Total                             25
```
Is this fix definitive, if so we need to remove this no elapsed time warning, right?

vvbandeira, 2024-01-16:
@gudeh, yes, the fix is definitive. Sourcing `env.sh` is required if you used `sudo ./setup.sh` or installed the dependencies in a non-default place (e.g., not in `/usr/bin`). The `env.sh` will add the `eqy` bin to the `PATH` variable.

I have started PR #1761, which should remove the warnings.