The-OpenROAD-Project/OpenROAD-flow-scripts #1527: Improve global placement performance

oharboe, 2023-10-04:
### Description

The example below was created with #1452 and `make DESIGN_CONFIG=designs/asap7/mock-registerfile/config.mk global_place_issue`

unzip  https://drive.google.com/file/d/1FGZ6Q66ZlWVqFuJ6IQj0ovF5nN2RDDlz/view?usp=drive_link

and run global placement, takes 1200 seconds or so. Ideally it should be faster...

```
./run-me-mock-registerfile-asap7-base.sh
```

rovinski, 2023-10-04:
I have asked around recently about implementing [this paper](https://infoscience.epfl.ch/record/273833/files/PID6218115.pdf). Which adds multithreading to GPL (albeit an older version from years ago). Haven't really gotten a clear reason why it hasn't been done - I think it may have had something to do with nondeterminism bugs. I or one of my students may start looking at that soon. There's some very low-hanging fruit in order to get decent speedups.

maliberty, 2023-10-04:
abk mentioned the same recently.  I don't see any sign that they made their code public.  There is some student work on a GPU version of replace.  I think all overlook the fact that calling grt/rsz/sta is a large fraction of the time that is not addressed.

rovinski, 2023-10-04:
@maliberty https://github.com/gessfred/Parallel-RePlAce