The-OpenROAD-Project/OpenROAD-flow-scripts #1547: Create a fake memory compiler for register files

maliberty, 2023-10-11:
### Description

In https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1452 the need for a register file was demonstrated.  https://github.com/ABKGroup/FakeRAM2.0 provides basic ram generation capability but doesn't satisfy the need for register files with arbitrary read/write ports.

### Suggested Solution

Enhance fakeram or develop an alternative solution to satisfy the requirements.

oharboe, 2023-10-17:
@louiic Just FYI of what Chipyard (that generates MegaBoom) outputs in order to describe the RAMs:

```
$ cat generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.mems.conf
name cc_dir_0_ext depth 1024 width 16 ports mrw mask_gran 16 
name cc_banks_0_ext depth 8192 width 64 ports rw  
name tag_array_0_ext depth 64 width 22 ports mrw mask_gran 22 
name array_0_0_0_ext depth 256 width 64 ports mwrite,read mask_gran 64 
name tag_array_0_0_ext depth 64 width 20 ports mrw mask_gran 20 
name dataArrayB0Way_0_ext depth 256 width 64 ports rw  
name hi_us_0_ext depth 128 width 1 ports mwrite,read mask_gran 1 
name table_0_ext depth 128 width 11 ports mwrite,read mask_gran 11 
name hi_us_0_0_ext depth 256 width 1 ports mwrite,read mask_gran 1 
name table_0_0_ext depth 256 width 12 ports mwrite,read mask_gran 12 
name table_0_1_ext depth 128 width 13 ports mwrite,read mask_gran 13 
name meta_0_0_ext depth 128 width 30 ports mwrite,read mask_gran 30 
name btb_0_0_ext depth 128 width 14 ports mwrite,read mask_gran 14 
name ebtb_ext depth 128 width 40 ports write,read  
name data_0_ext depth 2048 width 2 ports mwrite,read mask_gran 2 
name meta_0_ext depth 40 width 120 ports write,read  
name ghist_0_old_history_ext depth 40 width 64 ports write,read  
name ghist_0_current_saw_branch_not_taken_ext depth 40 width 1 ports write,read  
name ghist_0_ras_idx_ext depth 40 width 5 ports write,read  
name rob_debug_inst_mem_0_ext depth 32 width 32 ports mwrite,read mask_gran 32 
name l2_tlb_ram_0_ext depth 512 width 45 ports rw  
```

There are wafer thin wrappers around these SRAMs, such as:

```
module rob_debug_inst_mem_0(
  input [4:0]  R0_addr,
  input        R0_en,
               R0_clk,
  input [4:0]  W0_addr,
  input        W0_clk,
  input [31:0] W0_data,
  input        W0_mask
);

  wire [31:0] _rob_debug_inst_mem_0_ext_R0_data;
  rob_debug_inst_mem_0_ext rob_debug_inst_mem_0_ext (
    .R0_addr (R0_addr),
    .R0_en   (R0_en),
    .R0_clk  (R0_clk),
    .W0_addr (W0_addr),
    .W0_en   (1'h1),
    .W0_clk  (W0_clk),
    .W0_data (W0_data),
    .W0_mask (W0_mask),
    .R0_data (_rob_debug_inst_mem_0_ext_R0_data)
  );
endmodule
```

oharboe, 2023-10-17:
@tspyrou @maliberty @louiic So it would seem that hard macros are not generated for asynchronous(no read clock) SRAMs in Chipyard.

What will the fake memory compiler do w.r.t. asynchronous vs synchronous read SRAMs?

maliberty, 2023-10-17:
What is done instead?

oharboe, 2023-10-17:
> What is done instead?

flipflops. Asynchronous read SRAMs in the code are not considered SRAMs at all by Chipyard.

maliberty, 2023-10-17:
FF are clocked so how does that make it async?

oharboe, 2023-10-17:
> FF are clocked so how does that make it async?

asynchronous *read*. Write is clocked.

maliberty, 2023-10-17:
I'll leave it to @louiic to comment

maliberty, 2024-03-05:
@oharboe @louiic still relevant?

oharboe, 2024-03-06:
> @oharboe @louiic still relevant?

Not for my part. I am mocking memories now....

closed 2024-03-06 by maliberty