The-OpenROAD-Project/OpenROAD-flow-scripts #1553: Signal 11 received while running floorplan final check_setup

gkamendje, 2023-10-12:
### Subject

[Stage]: Floorplaning.

### Describe the bug

The tool crashes during the floorplan final check_setup phase when running `make floorplan` 

### Expected Behavior

Floorplan should complete and the tool should generate the expected `.odb` files.

### Environment

```shell
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.17.3
-- The CXX compiler identification is GNU 11.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10646-g0dff30da2
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3 (found version "3.10.12") found components: Interpreter
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
[... 89 lines cut ...]
10# sta::Search::seedArrivals() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# sta::Search::findArrivalsSeed() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# sta::Search::findArrivals1(int) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# sta::Search::findAllArrivals(bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# sta::Search::findAllArrivals() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# sta::Sta::checkTiming(bool, bool, bool, bool, bool, bool, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# check_timing_cmd(bool, bool, bool, bool, bool, bool, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# 0x000055EF5B997EF8 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
19# 0x00007FF51FEA0028 in /lib/x86_64-linux-gnu/libtcl8.6.so
20# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
21# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
22# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
23# 0x000055EF5B617D38 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
24# ord::tclAppInit(Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
25# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
26# main in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
27# 0x00007FF51BFEFD90 in /lib/x86_64-linux-gnu/libc.so.6
28# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
29# _start in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
Elapsed time: 0:06.79[h:]min:sec. CPU time: user 6.63 sys 0.16 (100%). Peak memory: 294256KB.
make[1]: *** [Makefile:559: do-2_1_floorplan] Error 139
make: *** [Makefile:559: results/smic018/ltag7/base/2_1_floorplan.odb] Error 2
```

maliberty, 2023-10-12:
Not the issue but you are missing a - in front of from:
```
set_false_path from [get_clocks nvm_clk] -to [get_clocks sys_clk]
```
In 
```
set_load 0.00261   -max -pin_load [get_ports [all_outputs]]
```
you should use
```
set_load 0.00261   -max -pin_load [all_outputs]
```
Likewise in
```
set_driving_cell -lib_cell BUFHDV2 -from_pin I -pin Z [get_ports [all_inputs]]
```

gkamendje, 2023-10-12:
Thanks for drawing my attention to the typos. 
BTW, I did notice that the crash only happens when I read the def file.

maliberty, 2023-10-12:
Thanks that's a good clue

maliberty, 2023-10-12:
In the PINS section of your floorplan DEF you are setting:
```
  + DIRECTION INOUT
```
which overrides what's in the Verilog and confuses the timer which has built the graph from the Verilog directions.  I think you should remove these statements.

I'll fix this by ignoring pin direction from floorplan DEF as I don't think that is the intended use for this capability.

gkamendje, 2023-10-13:
confirming that changing the port direction in the DEF file from `INOUT` to `OUTPUT` and `INPUT` respectively does not cause the tool to crash.

closed 2023-10-14 by maliberty