The-OpenROAD-Project/OpenROAD-flow-scripts #1561: Implement some way to test WRAP_LEFS/LIBS with non-NDA PDK

oharboe, 2023-10-13:
### Description

Only gf12 uses WRAP_LEFS/LIBS, this makes maintenance of ORFS awkward w.r.t. that functionality.

### Suggested Solution

Implement mock support for WRAP_LEFS/LIBS for e.g. asap7 such that these flows can be tested more easily.

This does not have to be for a real design.

`designs/asap7/mock-wrap-lefs/config.mk` ?

rovinski, 2023-10-13:
I would hazard that we don't want to develop that any further. The only point of the LEF wrapper was to shift macro pins to be on-grid because OR was having a lot of difficulty routing to off-grid pins in GF12. Further, it was a stop-gap (hacky) solution because of project deadlines.

Most technologies either 1) don't have that problem or 2) have a cleaner solution (e.g. simply move the pins on-grid). Development time would be better spent on a generalized solution inside the router if it is actually a problem.

maliberty, 2023-10-14:
I agree with @rovinski we want to less of it not more.  When we can find the time I would like to eliminate its use in gf12.  I think it is better to deal with your change as a one-off.

closed 2023-10-14 by maliberty