The-OpenROAD-Project/OpenROAD-flow-scripts #1579: Error when running openroad

chros098, 2023-10-23:
### Subject

[Documentation] for documentation errors.

### Describe the bug

运行出现error
[image]

### Expected Behavior

Error when running openroad 
[image]

### Environment

```shell
./etc/Env.sh
```

### To Reproduce

[image]

maliberty, 2023-10-23:
No legal way can be determined to access those pins.  Can you provide a test case?

rovinski, 2023-10-23:
@chros098 also please copy and paste text rather than posting a screenshot. It helps let others search for similar issues. Thanks!

chros098, 2023-10-26:
This problem has been solved after checking it