The-OpenROAD-Project/OpenROAD-flow-scripts #1580: Run stalled at CTS stage

dalyles, 2023-10-23:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I'm trying to run a variant of the fast Fourier transform, and the runtime is extremely long in the thousands of minutes, mainly in the CTS stage. There are no obvious signs of what the reason may be, but below I will provide a screenshot that better describes the situation.

### Expected Behavior

Run through OPFS in a reason able amount of time, not thousands of minutes.

### Environment

```shell
.
```

### To Reproduce

Run open road on cell. I'll provide files if needed.

### Relevant log output

```shell
==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273951

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
27.3768

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-16.4064

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-59.928114

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.16e-01   2.65e-03   2.43e-02   2.43e-01  45.3%
Combinational          1.64e-01   9.07e-02   3.81e-02   2.93e-01  54.7%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.79e-01   9.34e-02   6.22e-02   5.35e-01 100.0%
                          70.9%      17.5%      11.6%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17001.0 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538717.1 u
legalized HPWL       18775228.8 u
delta HPWL                    1 %

[WARNING RSZ-0069] skipping net _1107270_ with 152807 pins.
[WARNING RSZ-0069] skipping net _1114369_ with 112366 pins.
[WARNING RSZ-0069] skipping net _1199278_ with 100487 pins.
Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273951 endpoints with setup violations.
^Cmake[1]: *** [Makefile:672: do-4_1_cts] Error 130
make: *** [Makefile:672: results/nangate45/FFT40_MixedRadix/base/4_1_cts.odb] Interrupt

real    8162m12.779s
user    8041m25.869s
sys     117m29.941s
```
```

dalyles, 2023-10-23:
An additional note, I terminated the run. This was not an interrupt on the system itself.

maliberty, 2023-10-23:
How many instances are in this design?  Are these nets with fanout > 100k expected?  What is the tns/wns reported post-repair?

You might add -verbose option to repair_timing to get some idea if it is just progressing slowly.

dalyles, 2023-10-24:
I used the make flow in this process.  I added the option V=1 for the run and it doesn't provide any useful information. I will copy/paste to show what was put on the console after terminating the run.

```
==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273951

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
27.3768

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-16.4064

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-59.928114

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.16e-01   2.65e-03   2.43e-02   2.43e-01  45.3%
Combinational          1.64e-01   9.07e-02   3.81e-02   2.93e-01  54.7%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.79e-01   9.34e-02   6.22e-02   5.35e-01 100.0%
                          70.9%      17.5%      11.6%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17001.0 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538717.1 u
legalized HPWL       18775228.8 u
delta HPWL                    1 %

[WARNING RSZ-0069] skipping net _1107270_ with 152807 pins.
[WARNING RSZ-0069] skipping net _1114369_ with 112366 pins.
[WARNING RSZ-0069] skipping net _1199278_ with 100487 pins.
Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273951 endpoints with setup violations.
^Cmake[1]: *** [Makefile:672: do-4_1_cts] Error 130
make: *** [Makefile:672: results/nangate45/FFT40_MixedRadix/base/4_1_cts.odb] Interrupt
```

dalyles, 2023-10-24:
This file contains the RTL files FFT10, FFT20, FFT40. In this case I'm using FFT40
[FFT.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13125541/FFT.zip)

maliberty, 2023-10-24:
I'll need the sdc to run your design

maliberty, 2023-10-24:
the config.mk as well

maliberty, 2023-10-24:
The easiest way to make a complete reproducer is 'make cts_issue'

dalyles, 2023-10-24:
Here they are:

[FFT_const_config.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13125786/FFT_const_config.zip)

dalyles, 2023-10-25:
I see you closed this as completed. What are the next steps on my end to ensure that this works?

maliberty, 2023-10-25:
Build/run with the latest OR.

dalyles, 2023-10-25:
Thanks,

I'm going through the run again and it made some signification progress. But it looks like it is stalling again,
I think you may have to set the upper bound to a much higher value. Can this be done?

Thanks!

maliberty, 2023-10-25:
Do you see any "[WARNING RSZ-0069] skipping net" type messages?

dalyles, 2023-10-25:
Yes here they are:

```
[WARNING RSZ-0069] skipping net _1107270_ with 152807 pins.
[WARNING RSZ-0069] skipping net _1114369_ with 112366 pins.
[WARNING RSZ-0069] skipping net _1199278_ with 100487 pins.
Repair setup and hold violations...
```

maliberty, 2023-10-25:
Then you don't have the current version as the limit is now 200k.

vijayank88, 2023-10-26:
> Thanks,
> 
> I'm going through the run again and it made some significant progress. But it looks like it is stalling again, I think you may have to set the upper bound to a much higher value. Can this be done?
> 
> Thanks!

How you updated it?
Can you share steps to confirm, whether its updated to latest OR head?
Check the resizer log file with OpenROAD commit id

dalyles, 2023-10-26:
So how I updated it was to do this process:
```
git checkout master
git pull
./build_openroad.sh --local --clean
```

I am currently running the flow on a freshly cloned repo to ensure things are working as expected.

maliberty, 2023-10-26:
Use ./build_openroad.sh --local --clean --latest

dalyles, 2023-10-26:
Alright, i'll make sure to do that in my previous repo after I see the result of the run in my new repo.

dalyles, 2023-10-26:
After doing a fresh clone of the repo, I'm currently stuck here on the console. So I'm not sure if the boundary is even high enough? Surely with a fresh repo clone the update would also be in the relevant file to increase the pin count from 100k to 200k.

```
==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273951

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
27.3768

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-16.4085

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-59.935785

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.15e-01   2.65e-03   2.43e-02   2.42e-01  45.6%
Combinational          1.64e-01   8.75e-02   3.81e-02   2.89e-01  54.5%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.79e-01   9.01e-02   6.22e-02   5.31e-01 100.0%
                          71.3%      17.0%      11.7%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17328.9 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538658.0 u
legalized HPWL       18775147.0 u
delta HPWL                    1 %

[WARNING RSZ-0069] skipping net _1107270_ with 152807 pins.
[WARNING RSZ-0069] skipping net _1114369_ with 112366 pins.
[WARNING RSZ-0069] skipping net _1199278_ with 100487 pins.
Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273951 endpoints with setup violations.
```

maliberty, 2023-10-26:
As long as you see "[WARNING RSZ-0069] skipping net _1107270_ with 152807 pins." you aren't on the latest code.  Have you built with --latest?

dalyles, 2023-10-26:
Yes I did, I still get the same result. Is it possible to verify on your end?

maliberty, 2023-10-26:
I already verified it when I closed the issue.

dalyles, 2023-10-26:
I'll come back in a few hours to see if there is anything different. My run is still in progress and will update you.

maliberty, 2023-10-26:
If you continue to have problem please report the OR commit id you are using.

dalyles, 2023-10-27:
I'm currently stalling at this point for the last few hours.

```
==========================================================================
cts post-repair max_fanout_violation_count
--------------------------------------------------------------------------
max fanout violation count 0

==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273940

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
830.6546

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-822.2574

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-98.989086

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.33e-01   2.06e-03   2.35e-02   2.59e-01   6.7%
Combinational          3.46e+00   2.00e-02   4.16e-02   3.53e+00  91.4%
Clock                  2.18e-02   5.86e-02   9.97e-04   8.14e-02   2.1%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.71e+00   8.04e-02   6.56e-02   3.86e+00 100.0%
                          96.2%       2.1%       1.7%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17328.9 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538658.0 u
legalized HPWL       18775147.0 u
delta HPWL                    1 %

Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273939 endpoints with setup violations.
```

I do not see any of the warnings any more, would you assume this is still in runtime or 
is there another underlying issue? Especilly since  [INFO RSZ-0094] is present at the end?

Thanks

dalyles, 2023-10-27:
I found the actual file "Resizer.hh" and it has the update to 200K. 
I'll leave this run overnight and update you if anything happens. 
We may have stumbled upon another issue.

maliberty, 2023-10-27:
When I ran it the entire flow completed overnight.  How big a machine are you using?  I'm wondering if you are getting a lot of swapping.

dalyles, 2023-10-27:
Hmm that's a good sign I guess. I'm still stalling at the sam position I copied/pasted yesterday. Maybe there needs to be more time?

Here are the characteristics of the machine I'm using:

CPU: 2x AMD EPYC 7453 28-Core Processor @ 2.75GHz (28 Core, 2 Thread, 64MB L3)
Memory: 512GB (16x 32GB DRR4 2667 MT/s)
Disk Space: 1.7TB (/scratch)

dalyles, 2023-10-27:
I'm still at this position in my run:

```
==========================================================================
cts post-repair max_fanout_violation_count
--------------------------------------------------------------------------
max fanout violation count 0

==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273940

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
830.6546

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-822.2574

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-98.989086

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.33e-01   2.06e-03   2.35e-02   2.59e-01   6.7%
Combinational          3.46e+00   2.00e-02   4.16e-02   3.53e+00  91.4%
Clock                  2.18e-02   5.86e-02   9.97e-04   8.14e-02   2.1%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.71e+00   8.04e-02   6.56e-02   3.86e+00 100.0%
                          96.2%       2.1%       1.7%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17328.9 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538658.0 u
legalized HPWL       18775147.0 u
delta HPWL                    1 %

Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273939 endpoints with setup violations.
```

Are the characteristics of my machine adequate to handle a run like this, or is it possible that something else is wrong on my end?

dalyles, 2023-10-27:
If this helps here is a copy of my workspace:
https://drive.google.com/file/d/1U_UvXme2yuRjWmBSUcFlstlIc-9UDECY/view?usp=drive_link

The command I use to run the makeflow is:  make DESIGN_CONFIG=./designs/nangate45/FFT40_MixedRadix/config.mk

Thanks!

maliberty, 2023-10-30:
From your OpenROAD-flow-scripts/flow/logs/nangate45/FFT40_MixedRadix/base/4_1_cts.tmp.log file:
```
OpenROAD v2.0-10770-g6719c92e6 
```
this corresponds to commit id 6719c92e6 which is from Oct 19.  The fix was on Oct 25 so you still haven't run with the fix.

I already advised you run with --latest.  You need to do the work to update and verify that you have updated.

dalyles, 2023-10-31:
Hi @maliberty,

I did the update as instructed I tried to look for the commit ID, but it looks like nothing shows for fixes on October 25 <code>git log</code>. Am I looking at the right place? I'm in the home directory of ORFS.

```
commit 400b5477c13f5d5989289a3e402dc386dab683a2
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Sun Oct 29 23:32:43 2023 -0700

    fix the default design
    
    Signed-off-by: Matt Liberty <mliberty@precisioninno.com>

commit ddb88c659b9ba1135f93190a29371c7464c9b8c0
Author: KrzysztofHerman <herman@ihp-microelectronics.com>
Date:   Thu Oct 26 09:15:28 2023 +0200

    Metadata generated for each IHP supported design
    
    Signed-off-by: KrzysztofHerman <herman@ihp-microelectronics.com>

commit b711f8f692bf7e32aeaf95715993153bc1df2b9b
Merge: 8a0f9b75 46cdb712
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Tue Oct 24 23:12:47 2023 -0700

    Merge pull request #1585 from The-OpenROAD-Project-staging/aes-block-metrics
    
    update asap7/aes-block metrics
```

maliberty, 2023-10-31:
You are looking at ORFS but the fix is in OR.  You need to look at tools/OpenROAD which is a git submodule.

dalyles, 2023-10-31:
Hi @maliberty 
I have these in the commit logs
```
commit 446bd29261f5794d6dd9cf8ffdd8b2f8e1b795ee (master)
Merge: e2217ada5 1315af98a
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Wed Oct 25 08:29:24 2023 -0700

    Merge pull request #4172 from The-OpenROAD-Project-staging/rsz-bump-max_steiner_pin_count
    
    rsz: increase max_steiner_pin_count_ 100k -> 200k

commit 1315af98a4a5229baf9f573f4a286e11c3a5b35e
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Wed Oct 25 07:12:27 2023 -0700

    rsz: increase max_steiner_pin_count_ 100k -> 200k
    
    When we remove_buffers in floorplan initialization we may produce very
    high fanout nets.  rsz then refuses to work on them leading to
    failures.  I've bumped the limit based on a user design but a better
    long term solution is needed.
    
    Fixes The-OpenROAD-Project/OpenROAD-flow-scripts#1580
    
    Signed-off-by: Matt Liberty <mliberty@precisioninno.com>
```

Thanks for helping me verify if I had the latest update. Unfortunatley, this newly found issue after the update still presist: here's a snippet below:

```
==========================================================================
cts post-repair max_fanout_violation_count
--------------------------------------------------------------------------
max fanout violation count 0

==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273940

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
830.6546

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-822.2574

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-98.989086

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.33e-01   2.06e-03   2.35e-02   2.59e-01   6.7%
Combinational          3.46e+00   2.00e-02   4.16e-02   3.53e+00  91.4%
Clock                  2.18e-02   5.86e-02   9.97e-04   8.14e-02   2.1%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.71e+00   8.04e-02   6.56e-02   3.86e+00 100.0%
                          96.2%       2.1%       1.7%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17328.9 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538658.0 u
legalized HPWL       18775147.0 u
delta HPWL                    1 %

Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273939 endpoints with setup violations.
```

I had this running last night but still stalling in this position.
Thanks!

maliberty, 2023-10-31:
Perhaps you haven't compiled then?  The logs you provided were older as I've pointed out.

dalyles, 2023-10-31:
I executed these commands to do the update as of yesterday.
Is there something i'm missing here?

```
git checkout master
git pull
./build_openroad.sh --local --clean --latest
```

Thanks

maliberty, 2023-10-31:
That should do it.  You can verify that the commit id when running OR matches what you see in git log.

dalyles, 2023-10-31:
Before rerunning these commands i'll show you a snapshot of the most recent commit ids:

```
commit 536cd01eaa0d5e7a67e87fcee9296c555036b66b (HEAD, origin/master, origin/HEAD)
Merge: 2126c1671 d4f96d5b4
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Mon Oct 30 17:12:48 2023 -0700

    Merge pull request #4190 from eder-matheus/grt_fix
    
    grt: check net layer range when calculating NDR costs

commit 2126c1671d4de8e253e0c9c8b6b111d6f19c3237
Merge: bb8df437c 6d652f114
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Mon Oct 30 16:02:18 2023 -0700

    Merge pull request #4174 from eder-matheus/fix_gitignore
    
    fix gitignore and report file generation

commit bb8df437cdfabac29866dfbb97aee9c1b3a38bfb
Merge: 9964d69c3 b9204ed82
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Mon Oct 30 15:52:12 2023 -0700

    Merge pull request #4189 from The-OpenROAD-Project-staging/cts-name-fix
    
    cts: fix names of generated clock nets

commit d4f96d5b4c563f297dc8b6d1a67568428c9e430d
Author: Eder Monteiro <emrmonteiro@inf.ufrgs.br>
Date:   Mon Oct 30 19:01:29 2023 -0300

    grt: check net layer range when calculating NDR costs
    
    Signed-off-by: Eder Monteiro <emrmonteiro@inf.ufrgs.br>
```

Does this seem to align with your setup?

maliberty, 2023-10-31:
That's after my fix so it should be fine.

dalyles, 2023-10-31:
Ok sounds good, then the iussie I mentioned here exists with the proper update.

```
==========================================================================
cts post-repair max_fanout_violation_count
--------------------------------------------------------------------------
max fanout violation count 0

==========================================================================
cts post-repair max_cap_violation_count
--------------------------------------------------------------------------
max cap violation count 61

==========================================================================
cts post-repair setup_violation_count
--------------------------------------------------------------------------
setup violation count 273940

==========================================================================
cts post-repair hold_violation_count
--------------------------------------------------------------------------
hold violation count 0

==========================================================================
cts post-repair critical path delay
--------------------------------------------------------------------------
830.6546

==========================================================================
cts post-repair critical path slack
--------------------------------------------------------------------------
-822.2574

==========================================================================
cts post-repair slack div critical path delay
--------------------------------------------------------------------------
-98.989086

==========================================================================
cts post-repair report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             2.33e-01   2.06e-03   2.35e-02   2.59e-01   6.7%
Combinational          3.46e+00   2.00e-02   4.16e-02   3.53e+00  91.4%
Clock                  2.18e-02   5.86e-02   9.97e-04   8.14e-02   2.1%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.71e+00   8.04e-02   6.56e-02   3.86e+00 100.0%
                          96.2%       2.1%       1.7%

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 2800602 u^2 36% utilization.

Placement Analysis
---------------------------------
total displacement      17328.9 u
average displacement        0.0 u
max displacement           10.7 u
original HPWL        18538658.0 u
legalized HPWL       18775147.0 u
delta HPWL                    1 %

Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 273939 endpoints with setup violations.
```

Is [INFO RSZ-0094] An error of concern here? Or is there something that needs to be updated in the script to handle this design?

maliberty, 2023-10-31:
Please upload the full log from this run.  How long did you let it run?  How much memory do you have?  Please add -verbose to see any progress.

dalyles, 2023-10-31:
ok, I will terminate the run and provide the log information: (FYI, i executed this with make soI couldn't invoke a -verbose option in real time)

[FFT40_MixedRadix.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13221315/FFT40_MixedRadix.tar.gz)

maliberty, 2023-10-31:
at least this looks like a good commit: OpenROAD v2.0-10891-g536cd01ea 

You didn't answer any of the other questions so its hard to know if it is truly hung or just not done yet.

dalyles, 2023-11-01:
Hi Matt,
I dont have the exact information to those questions. All I really know is that it has been running for at least 15 hours (with the fast majoriy of the runtime hung in aforementioned stage) and I don't know how to access the memory usage. 
I just suspect something is not right with the run but I don't have the expertise to properly diagnose the issue.

dalyles, 2023-11-12:
I'm following up on this particular issue (still a hanging run):
here is the information for my machine.

HW Characteristics:
CPU: 2x AMD EPYC 7453 28-Core Processor @ 2.75GHz (28 Core, 2 Thread, 64MB L3)
Memory: 512GB (16x 32GB DRR4 2667 MT/s)
Disk Space: 1.7TB (/scratch)
SW Characteristics:
OS: Ubuntu 20.04.6 LTS
GUI: vnc

I hope this helps.
Thanks