The-OpenROAD-Project/OpenROAD-flow-scripts #1586: [ERROR GUI-0013] Unable to find visible display control at Pins.

gkamendje, 2023-10-26:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

I get the following error after moving to commit b711f8f692bf7e32aeaf95715993153bc1df2b9b

```
Saving final image
[ERROR GUI-0013] Unable to find visible display control at Pins.
[ERROR GUI-0070] Error: save_images.tcl, 23 GUI-0013
Error: final_report.tcl, 76 GUI-0070
Command exited with non-zero status 1
Elapsed time: 0:07.66[h:]min:sec. CPU time: user 7.15 sys 0.33 (97%). Peak memory: 361144KB.
make[1]: *** [Makefile:809: do-6_report] Error 1
make: *** [Makefile:809: logs/smic018/test/base/6_report.log] Error 2
```

### Expected Behavior

The flow should complete without issue

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.28.0-rc2
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10770-g6719c92e6
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
[... 102 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Download the following testcase
https://drive.google.com/file/d/11ZBQpopI98jTpNsrpt8uEp-S1QSRzwxt/view?usp=share_link
 and run `make finish`

### Relevant log output

```shell
Saving final image
[ERROR GUI-0013] Unable to find visible display control at Pins.
[ERROR GUI-0070] Error: save_images.tcl, 23 GUI-0013
Error: final_report.tcl, 76 GUI-0070
Command exited with non-zero status 1
Elapsed time: 0:07.66[h:]min:sec. CPU time: user 7.15 sys 0.33 (97%). Peak memory: 361144KB.
make[1]: *** [Makefile:809: do-6_report] Error 1
make: *** [Makefile:809: logs/smic018/test/base/6_report.log] Error 2
```
```

SawyDust1228, 2023-12-29:
Have you fixed the problem?

gkamendje, 2023-12-29:
No I have just disabled that step for now.

maliberty, 2023-12-30:
Most likely you have an out of date version of the openroad app.  Please update and try again.  If problems persist please give the commit id you are using.

maliberty, 2024-03-05:
Stale - reopen if still relevant

closed 2024-03-05 by maliberty