The-OpenROAD-Project/OpenROAD-flow-scripts #1587: unclocked register despite clock definition on leaf pin

gkamendje, 2023-10-27:
### Subject

[Stage]: CTS.

### Describe the bug

I have a clock divider in the design and I would like the clock defined on leaf pins to be seen by the endpoints driven by the clock divider. This is not the case
```
check_setup -verbose
Warning: There are 2 unclocked register/latch pins.
  _3694_/CK
  _3695_/CKN
Warning: There are 7 unconstrained endpoints.
  _3694_/D
  _3695_/D
  _3712_/D
  _3787_/D
  clk_rst_inst.cg_nvm_clk_inst.CLKLANQHDV_inst/E
  clk_rst_inst.cg_nvm_clk_inst.CLKLANQHDV_inst/TE
  clk_rst_inst.cg_pump_clk_inst.CLKLANQHDV_inst/TE
```
In a more general sense, this issue is about how contrains to hierarchical pins should be defined.

### Expected Behavior

clock pins `_3694_/CK` driven by `mem_ctrl_inst.i_bit_period` should be part of the clock tree. 

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.28.0-rc2
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10770-g6719c92e6
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format -CMake Error at src/gpl/CMakeLists.txt:44 (find_package):
  By not providing "Findortools.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ortools", but
  CMake did not find one.

  Could not find a package configuration file provided by "ortools" with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
  "ortools_DIR" to a directory containing one of the above files.  If
  "ortools" provides a separate development package or SDK, be sure it has
  been installed.

 Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")
-- Found Boost: /usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0")
-- boost: 1.78.0
-- Found Python3: /usr/include/python3.10 (found version "3.10.12") found components: Development Development.Module Development.Embed
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- spdlog: 1.9.2
-- Found BISON: /usr/bin/bison (found version "3.8.2")
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot
-- STA version: 2.4.0
-- STA git sha: 1fed0491bb9e84c9622b7e984821e0711502a14e
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

download the following test case and run `make cts`

### Screenshots

[image]

maliberty, 2023-10-31:
Would you describe the path an input clock is supposd to take in reaching _3694_/CK?  Tracing back from that input leads to a big fanin cone:
[image]

gkamendje, 2023-11-01:
Here is a path from the clock input i_sys_clk to the `3694/CK`.  The sys_clk does not go through the DFF `3718` although the tool instantiates the clock inverter `_1882/ZN`. 
```
CLKBUFHDV0 clkbuf_0_i_sys_clk (.I(i_sys_clk),    .Z(clknet_0_i_sys_clk));
CLKBUFHDV0 clkbuf_1_1__f_i_sys_clk (.I(clknet_0_i_sys_clk),    .Z(clknet_1_1__leaf_i_sys_clk));
CLKBUFHDV0 clkbuf_leaf_9_i_sys_clk (.I(clknet_1_1__leaf_i_sys_clk),  .Z(clknet_leaf_9_i_sys_clk));
DRNQHDV0 _3718_ (.CK(clknet_leaf_9_i_sys_clk),    .D(_0123_),    .Q(\tx_inst.bit_time_counter[5] ),    .RDN(net70));
NAND4BBHDV2 _1881_ (.A1(\tx_inst.bit_time_counter[5] ),    .A2(_1527_),    .B1(_1529_),    .B2(_1540_),    .ZN(_1541_));
CLKINHDV0 _1882_ (.I(_1541_),    .ZN(\mem_ctrl_inst.i_bit_period ));
DRNQHDV0 _3694_ (.CK(\mem_ctrl_inst.i_bit_period ),    .D(\tx_inst.encoder_inst.bi_ph_reg2_d ),    .Q(\tx_inst.encoder_inst.bi_ph_reg2_q ),    .RDN(net69));
```

maliberty, 2023-11-01:
It would be quite odd for clock to go through a CK->Q arc of a FF.  Why do you think it should?

gkamendje, 2023-11-01:
For simple clock division by 2

maliberty, 2023-11-01:
It sounds like you need to use create_generated_clock then

gkamendje, 2023-11-02:
I have been exactly trying to use generated clocks for quite a while. I am having issues with SDC constrains (see #1570). I cannot figure out how to access internal pins while setting SDC constrains. I would like for example to create a generated clock on a hierarchical pin using a syntax similar to this: `create_generated_clock -name bit_clock    -source [get_ports i_sys_clk] -divide_by 2 [get_pins u1_inst?u2_inst?u3_inst/Z]`  I have tried variants such as `get_pins u1_inst*u2_inst*u3_inst/Z` or even `get_pins u1_inst.u2_inst.u3_inst/Z` .  I get the error `[WARNING STA-0335] pin '' not found.` at various steps of the flow.  I could not find any example of generated clock in the other designs.

maliberty, 2023-11-06:
Have you looked at the yosys output?  In the db you provided all the instances are given flat names (eg _<num>_).  There is no u1_inst* to match to.

gkamendje, 2023-11-12:
Thanks for the hints. Manually instantiating gates on which the contrains are to be applied and following Yosys's flattening naming convention,  `get_pins u1_inst.u2_inst.uleaf_inst/Q` did finally work. Closing this case.

closed 2023-11-12 by gkamendje