The-OpenROAD-Project/OpenROAD-flow-scripts #1589: High slack difference between OpenRoad and Commercial EDA tool

faisaladilquadery, 2023-10-31:
### Subject

[Stage]: Detail Router.

### Describe the bug

I have implemented three designs using a commercial EDA tool and OpenRoad. The netlist for all three designs were also taken from a commercial EDA Tool. The designs with implemented with a 65nm Technology ( Foreign PDK )
After the Implementation I have compared the setup timing of the wns paths of openroad and Innovus and noticed huge slack differences and a lot of differences in the structure of the timing path. Some of which are as follows :

1. Unncecessary detouring in data paths of OpenROAD.
2. Less complex cells used in OpenROAD.
3. Higher buffer counts in OpenROAD.
4. Higher Cell delay in OpenROAD.

Below I have attached screenshots of three design and a selected timing path for both to compare:
**Left Picture : Commercial EDA Tool, Right Picture : OpenRoad**

[image]
[image]
Fig 1 : Design 1 ( OpenSource Design )

[image]
[image]
Fig 2 : Design 2 ( OpenSource Design )

[image]
[image]
Fig 3 : Design 3 ( OpenSource Design )

[image]
Fig 4 : Data Summary

[image]
Fig 5 : Graphical Comparison

After analysing the three designs I have the following issues and questions: 
**1. It is noticeable that for all three designs, openroad made some detoures in the paths. Why is openroad doing this ? Is it related to its algorithm ?**
**2. All the designs has very high cell delay values, which is the root cause for the high slack differences. Why is this the case ?**
**Note : My observation is that there maybe a irregular correlation between placement and global routing**
**3. What could be some methods and solution to reduce these slack differences ?**
**4. Can CTS be synthesized with multiple clock root buffers? Currently it takes only one**

### Expected Behavior

Both tools to have a slack difference of 100ps.

### Environment

```shell
e (found version "1.80.0")  
-- boost: 1.80.0
-- Found Python3: /home/nsl/anaconda3/include/python3.9 (found version "3.9.12") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- spdlog: 1.8.1
-- Found BISON: /usr/bin/bison (found version "3.8.2") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.4.0
-- STA git sha: d26f41708ed89d467c033334b0a3895558c003b4
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/nsl/OST/nsl/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found re2: /opt/or-tools/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

./

maliberty, 2023-10-31:
We discourage the posting of results from commercial tools.  Often the license terms prohibit this so you should be careful.  I'll only respond to how you might improve the OR results in general terms.

Less complex cells used in OpenROAD : this is determined by synthesis which would be yosys not OR.  You can factor out the impact of synthesis by using a common post-synthesis netlist.  Issues with yosys should be filed with https://github.com/YosysHQ/yosys

You show "a selected timing path".  Is that the most critical path in OR?

"Can CTS be synthesized with multiple clock root buffers? Currently it takes only one" - there is only one root so what would be the purpose?

It is hard to make progress without a test case to look at in more depth.  While interesting it isn't very actionable to look at images.

faisaladilquadery, 2023-11-01:
@maliberty Thanks for the heads up, I was unaware at the moment regarding the license terms.
I will precisely keep the discussion on how to improve the OpenRoad results.

Can you please help me understand the below queries:
1. Why is there so much cell delay for the WNS paths mentioned for the three seperate designs?
2. What is the reasons for extra detouring ?
3. Regarding root buffer cells, I was talking about the following: 
[image]
OpenRoad uses only one drive strength clock buffer cell for these marked positions as being assigned to it. Can openroad use multiple cells ( multiple driver strength clock buffers) from the clock buffer list for these cells ?

I have uploaded test case of Design 2 in this link:
https://drive.google.com/file/d/13WJbibcuLEPsQsg3f0Yo_Tc_bv0IjXPI/view?usp=sharing

maliberty, 2023-11-01:
1 - Quite likely due to the selection of gates during techmapping in yosys/abc.  Possibly due to limited gate sizes in the library.  We'll have to look at the paths in detail.
2 - We'll have to look at the paths in detail.
3 - Those aren't root buffers.  The root buffer is above this point at the root of the tree.

faisaladilquadery, 2023-11-02:
1. You can load the gui from the test case i provided and look at the wns path.
Startpoint : u_id_ex/op1_ff/qout_r_reg_4
Endpoint : u_regs/regs_reg_11_17
For Reference: 
[image]
[image]
[image]
[image]

Path in gui : ( For Reference )
[wns path design 2 in gui.docx](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13234148/wns.path.design.2.in.gui.docx)

2. Reference above
3. Okay I understand. But what about the cells i have marked ? can openroad use multiple cells for them ?

maliberty, 2023-11-02:
3 - yes those are the regular clock buffer cells in -buf_list

faisaladilquadery, 2023-11-02:
[image]
This is my clock buff list but tools are only using ckbd12
[image]
Any particular reason ?