The-OpenROAD-Project/OpenROAD-flow-scripts #1617: digital route violation

chros098, 2023-11-13:
### Subject

[Stage]: Detail Router.

### Describe the bug

When I added a new process (FinFET process) according to the official documents, I ran to 5-2detail roate. The log showed that there were many via space violations, and then the program kept repairing, but the related errors would not be repaired. The lech is attached to the picture. For the lef via part, please help find the reason, thank you

### Expected Behavior

perfect run 

### Environment

```shell
no
```

### To Reproduce

暂无

### Relevant log output

```shell
暂无
```

### Screenshots

[image]
[image]
[image]

### Additional Context

暂无

maliberty, 2023-11-13:
Can you load the markers in the GUI to see if there is some pattern?  
What is your PDN structure on M1/M2 like?  Do you have parallel rails?

chros098, 2023-11-13:
We installed a new version of opneroad and some new problems appeared. We will solve the new problems first and then reproduce the problem.

chros098, 2023-11-15:
I reported a new error after updating the PDN. Please help me check it. Thank you.
[image]
[image]
[image]

maliberty, 2023-11-15:
Please provide a test case as the logs are insufficient to say what is wrong.

chros098, 2023-11-20:
Sorry, PDK cannot be made public. I will think of another way.

closed 2023-12-06 by chros098