The-OpenROAD-Project/OpenROAD-flow-scripts #1618: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1454

chros098, 2023-11-13:
### Subject

[Stage]: CTS.

### Describe the bug

During the process of running the nangate45 official use case, a problem occurred in the cts stage.

### Expected Behavior

During the process of running the nangate45 official use case, a problem occurred in the cts stage.

### Environment

```shell
The latest version of the official environment
```

### To Reproduce

nagate45 gcd test

### Screenshots

[image]
[image]

vijayank88, 2023-11-13:
similar to this issue: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1513

vvbandeira, 2024-01-17:
Could you try again with the latest code? Please also make sure to `source env.sh` before running the flow if you used `sudo ./setup.sh` to install the dependencies.

(also, closing as duplicated)