The-OpenROAD-Project/OpenROAD-flow-scripts #1621: coyote_tc config.mk in ORFS failing at PDN/PAD Placement

nchiolino, 2023-11-15:
### Subject

[Stage]: Floorplaning.

### Describe the bug

When running the sky130hg/coyote_tc/config.mk Makefile, it appears as though it fails when trying to floorplan the PDN grid to the supply pins of the I/O cells. Its not clear if the error is associated with the PDN.tcl file or the PAD.tcl file. When loading the ODB files for macro placement, it appears that both macros and the I/O have successfully placed. It just seems as though the power grid cannot "find" or complete due to an issue I cannot understand. I have all the latest example files as of 11/10/23. The last few lines of the 2_6_floorplan_pdn.log file can be seen below:

[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_vdd_8 is not connected to any net.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_vss_8 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_vss_8 is not connected to any net.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_vzz_20 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_vzz_20 is not connected to any net.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_v18_20 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_v18_20 is not connected to any net.
[INFO PDN-0001] Inserting grid: grid
Command terminated by signal 9

### Expected Behavior

I expect the example design to run successfully. 

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.2.0-36-generic

os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-10925-gb7c9b7e1c

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done
[... 238 lines cut ...]

-- TCL header: /usr/include/tcl/tcl.h

-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 

-- TCL readline enabled

-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.TKdR6bpmOW
```

### To Reproduce

run make DESIGN_CONFIG=./designs/sky130hd/coyote_tc/config.mk

### Additional Context

I am simply looking for a working example of a complete flow which includes both macros and I/O cells that fully routes and generates GDS. I have no preference on technology.

maliberty, 2023-11-15:
Is this just a duplicate of #1082 ?

nchiolino, 2023-11-15:
I took a long look at that issue and seems to be an old issue that I was assuming was fixed. I cannot seem to understand what the exact issue is with #1082 . As mentioned above, the ODB files that are generated in the flow seem to indicate that the pads and the macros are placing. The issue I am seeing with with the PDN step. Am I incorrect in assuming that?

maliberty, 2023-11-15:
The pdn problems are related to the pads so it is much of the same.

closed 2023-11-15 by maliberty

nchiolino, 2023-11-15:
Ok, thank you for your quick response.

vijayank88, 2023-11-16:
I think I have made some progress on this by fixing PDN issue and got stuck at global routing stage.
Let me check my codebase and make PR for PDN fix.

vijayank88, 2023-11-16:
I ran ORFS with latest commit, pdn stage goes beyond where you've raised the issue @nchiolino 
```
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_vzz_20 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_vzz_20 is not connected to any net.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_v18_20 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_v18_20 is not connected to any net.
[INFO PDN-0001] Inserting grid: grid
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u_coyote/r2f/rocket/RocketTile/icache/icache/T198/mem/macro_mem/macro_mem
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u_coyote/r2f/rocket/RocketTile/icache/icache/T212/mem/macro_mem/macro_mem
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u_coyote/r2f/rocket/RocketTile/icache/icache/T226/mem/macro_mem/macro_mem
````

nchiolino, 2023-11-16:
@vijayank88 Thank you very much for continuing to dig into this! I don't see any changes to the pdn.tcl script in github. How should I proceed with your fix?
Thank you very much in advance.

nchiolino, 2023-11-16:
I don't know if this is a legitimate issue or not, I am just learning all of this to be honest, but I did notice that there are no power or ground pads listed in the gate level netlist (1_synth.v). I was able to find all of the GPIO pads and they seem to be connected properly but there are no power or ground pads in that netlist. Shouldn't the power and ground pads exist in that netlist as well? That could possibly be why PDN is failing, because it cannot make a connection that is not in the top level verilog. It seems as thought the pad.tcl file is placing the I/O cells in there proper locations but the router cannot connect or "find" the power and ground nets because, I think, the top level Verilog file does not include the connections. Does that sound correct?

maliberty, 2023-11-16:
The verilog doesn't need to have pg pins as we connect them up by name

nchiolino, 2023-11-16:
I am seeing that now actually. So the power and ground nets are handled by ios.v?

maliberty, 2023-11-16:
They are handled by 
```
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDD$} -power
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDDPE$}
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDDCE$}
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {VPWR}
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {VPB}
add_global_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^VSS$} -ground
add_global_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^VSSE$}
add_global_connection -net {VSS} -inst_pattern {.*} -pin_pattern {VGND}
add_global_connection -net {VSS} -inst_pattern {.*} -pin_pattern {VNB}
```
in flow/platforms/sky130hd/pdn.tcl

vijayank88, 2023-11-17:
> @vijayank88 Thank you very much for continuing to dig into this! I don't see any changes to the pdn.tcl script in github. How should I proceed with your fix? Thank you very much in advance.

I am not made any changes in the coyote_tc setup. It is passing till CTS stage. Got stuck at global routing stage.
Maybe I'll raise new issue for routing stage.
What is system resource? like RAM,Swap Memory?

nchiolino, 2023-11-21:
@vijayank88 Thank you very much for your feedback. As it turns out, I was about 2GB shy from it passing through PDN and now I am getting the same failure that you are seeing. It seems to be getting stuck on global routing with too much congestion. 
I will keep pushing on it but wanted to at least let you know that I am getting the same result. 
Thank you

vijayank88, 2023-11-22:
@nchiolino 
Routing congestion issue already raised here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1630

nchiolino, 2023-11-22:
Excellent. I will keep an eye on it. I am also building my own smaller example to test some things out.