The-OpenROAD-Project/OpenROAD-flow-scripts #1630: coyote_tc facing [ERROR GRT-0119] Routing congestion too high

vijayank88, 2023-11-18:
### Subject

[Stage]: Global Router.

### Describe the bug

Built-in example `sky130hd/coyote_tc` facing global routing congestion issue. 
This is IO pad based test design. So it is good to find/fix what is causing routing congestion..
[image]

### Expected Behavior

Complete coyote_tc RTL-GDSII generation

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11117-g6061c4892
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /opt/rh/rh-python38/root/usr/bin/python3.8 (found version "3.8.13") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 124 lines cut ...]
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal   10061800       4863708          51.66%
met2       Vertical      7538478       4589953          39.11%
met3       Horizontal    5030900       2919748          41.96%
met4       Vertical      3016992       1612646          46.55%
met5       Horizontal    1006180        427911          57.47%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 1549727
[INFO GRT-0198] Via related Steiner nodes: 69538
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 1979103
[INFO GRT-0112] Final usage 3D: 9366721
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/coyote_tc/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 35 GRT-0119
Elapsed time: 18:28:12[h:]min:sec. CPU time: user 66477.02 sys 6.46 (99%). Peak memory: 2855508KB.
make[1]: *** [Makefile:726: do-5_1_grt] Error 1
make: *** [Makefile:726: results/sky130hd/coyote_tc/base/5_1_grt.odb] Error 2
```

vijayank88, 2023-12-08:
@eder-matheus
Any update on this?

vijayank88, 2023-12-26:
With latest run congestion on corner pad location
[image]

eder-matheus, 2024-03-25:
@vijayank88 @maliberty  I've run this design with the latest versions of our tools, and I found an odd thing about the nets with congestion. Some instances are not placed, and the nets with congestion are connected to pins of these instances:
[image]

Are these instances expected to be not placed? Should GRT fail for instances like that?

maliberty, 2024-03-25:
Unplaced IOs don't make sense but coyote_tc isn't really a working test case as io placment was never working for it.

eder-matheus, 2024-03-25:
> Unplaced IOs don't make sense but coyote_tc isn't really a working test case as io placment was never working for it.

And who should be responsible for placing these instances? Is it PPL or a different tool? I see that the instance type of these is `Inst`, so PPL would not make anything to these instances.

maliberty, 2024-03-25:
pad would place them

eder-matheus, 2024-10-23:
Closing it for now. Feel free to open it again or create a new issue once he PADs are properly placed in this design

closed 2024-10-23 by eder-matheus