The-OpenROAD-Project/OpenROAD-flow-scripts #1643: Document interaction of footprint, special nets and CTS

Syndace, 2023-11-23:
### Description

After adding a footprint to our design, TritonCTS suddenly skipped all clocks. The log read as follows:

```
[INFO CTS-0007] Net "p_clk" found for clock "p_clk".
[INFO CTS-0116] Special net "p_clk" skipped.
[INFO CTS-0007] Net "p_jtag_tck" found for clock "p_jtag_tck".
[INFO CTS-0116] Special net "p_jtag_tck" skipped.
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
```

We had our clocks configured in the SDC as usual, using:

```
create_clock [get_ports p_clk] -period 30 -waveform {0 15}
create_clock [get_ports p_jtag_tck] -period 100 -waveform {0 50}
```

First, I found this comment on a different issue, which suggests that the footprint setup adds pad signals to special nets: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1182#issuecomment-1610157940
Then, I noticed that the coyote_tc example uses `get_pins` instead of `get_ports` for their clocks only, avoiding the pad signal, likely to work around this issue. After switching our clock definitions to `get_pins` as well, the clock synthesis works as expected again.

### Suggested Solution

Document this apparently known interaction between footprint and CTS.

maliberty, 2023-11-24:
You should not use the old pad footprint code.  It is just for legacy support and there are new tcl commands that replace it.

As for clocks you either need to set them at pad outputs or have a timing model for the pad that allows for the clock to propagate through the pad.

Syndace, 2023-11-24:
Thank you,

> You should not use the old pad footprint code. It is just for legacy support and there are new tcl commands that replace it.

this is my footprint script, based on [the documentation](https://openroad.readthedocs.io/en/latest/main/src/pad/README.html) and the sky130hd/coyote_tc example: [footprint.tcl.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13455985/footprint.tcl.txt)
Is that approach considered legacy?

> [...] or have a timing model for the pad that allows for the clock to propagate through the pad.

Would that be part of the IO library's full liberty file? We currently use a structural liberty similar to [sky130_dummy_io.lib](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130io/lib/sky130_dummy_io.lib) while the full liberty is in the works.

maliberty, 2023-11-24:
That's fine - I thought you were using the older footprint config file.

The sky130 looks ok though I haven't actually tried it.  You should have a timing arc from the pad to the output driving the core that sta can propagate through.

Syndace, 2023-11-25:
Okay great, thanks :+1:

closed 2023-11-25 by Syndace