The-OpenROAD-Project/OpenROAD-flow-scripts #1646: Some pads missing from padframe (IHP PDK)

Syndace, 2023-11-24:
### Subject

[Stage]: I/O Placement.

### Describe the bug

Some pads are missing in the padframe. There is no warning/error in the log, the pads are simply not there.

### Expected Behavior

.

### Environment

```shell
I'm using the docker container and have last updated a few days ago. The command still does not succeed:

tee: build/openroad-env.log: Permission denied
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 5.10.0-26-amd64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

CMake Error at CMakeLists.txt:112 (message):
  Insufficient gcc version.  Found 4.8.5, but require >= 8.3.0.

-- The CXX compiler identification is GNU 4.8.5
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: HEAD-HASH-NOTFOUND
-- Configuring incomplete, errors occurred!
```
```

### To Reproduce

The following is a small, self-contained example project that reproduces the problem. (We're finally fully open-source, no more NDAs!!! :tada:)

[missing_pads.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13458432/missing_pads.tar.gz)

### Screenshots

This is how the project looks after building, a few pads missing (all of them input pads, don't know if that's just coincidence though): 
![2023-11-24-122108_2560x1440_scrot](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/assets/5029529/bc966ecd-71b9-4bf8-a53b-1cbcec4e7562)

(Also note how there are suspiciously fewer connections between the pads and the power ring on the left and right sides than on top and bottom. That might be a different issue though.)

### Additional Context

I don't think that's relevant for this issue, but I have the following minor changes to the IHP PDK locally:

```
$ git diff platforms/ihp-sg13g2/
diff --git a/flow/platforms/ihp-sg13g2/config.mk b/flow/platforms/ihp-sg13g2/config.mk
index 1d3f7e35..9c8b5505 100644
--- a/flow/platforms/ihp-sg13g2/config.mk
+++ b/flow/platforms/ihp-sg13g2/config.mk
@@ -33,8 +33,8 @@ export FILL_CELLS = sg13g2_fill_1 sg13g2_fill_2 sg13g2_fill_4 sg13g2_fill_8
 #  ----------------------------------------------------
 # Set the TIEHI/TIELO cells
 # These are used in yosys synthesis to avoid logical 1/0's in the netlist
-export TIEHI_CELL_AND_PORT = sg13g2_tiehi HI
-export TIELO_CELL_AND_PORT = sg13g2_tielo LO
+export TIEHI_CELL_AND_PORT = sg13g2_tiehi L_HI
+export TIELO_CELL_AND_PORT = sg13g2_tielo L_LO
 
 # Used in synthesis
 export MIN_BUF_CELL_AND_PORTS = sg13g2_buf_1 A X
diff --git a/flow/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef b/flow/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef
index c11d4c53..3eb01c4f 100644
--- a/flow/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef
+++ b/flow/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef
@@ -1163,7 +1163,7 @@ MACRO sg13g2_and4_2
 END sg13g2_and4_2
 
 MACRO sg13g2_antennanp
-  CLASS CORE ;
+  CLASS CORE ANTENNACELL ;
   ORIGIN 0 0 ;
   FOREIGN sg13g2_antennanp 0 0 ;
   SIZE 1.44 BY 3.78 ;
@@ -3829,7 +3829,7 @@ MACRO sg13g2_einvn_8
 END sg13g2_einvn_8
 
 MACRO sg13g2_fill_1
-  CLASS CORE ;
+  CLASS CORE SPACER ;
   ORIGIN 0 0 ;
   FOREIGN sg13g2_fill_1 0 0 ;
   SIZE 0.48 BY 3.78 ;
@@ -3859,7 +3859,7 @@ MACRO sg13g2_fill_1
 END sg13g2_fill_1
 
 MACRO sg13g2_fill_2
-  CLASS CORE ;
+  CLASS CORE SPACER ;
   ORIGIN 0 0 ;
   FOREIGN sg13g2_fill_2 0 0 ;
   SIZE 0.96 BY 3.78 ;
@@ -3889,7 +3889,7 @@ MACRO sg13g2_fill_2
 END sg13g2_fill_2
 
 MACRO sg13g2_fill_4
-  CLASS CORE ;
+  CLASS CORE SPACER ;
   ORIGIN 0 0 ;
   FOREIGN sg13g2_fill_4 0 0 ;
   SIZE 1.92 BY 3.78 ;
@@ -3919,7 +3919,7 @@ MACRO sg13g2_fill_4
 END sg13g2_fill_4
 
 MACRO sg13g2_fill_8
-  CLASS CORE ;
+  CLASS CORE SPACER ;
   ORIGIN 0 0 ;
   FOREIGN sg13g2_fill_8 0 0 ;
   SIZE 3.84 BY 3.78 ;
```

Syndace, 2023-11-29:
I spent a bit of time investigating but I haven't really made progress. What I can confirm now is that InOut pads never disappear. In the following build I've tried using InOut pads (with hardwired output enable depending on whether they're inputs or outputs) for all pads and all of the pads are always there. However, the flow still struggles to connect some of the pads to the power net, with error messages like this:

```
[ERROR PSM-0042] Unable to connect macro/pad Instance u_din to the power grid.
```

I've tried playing around with `attribute keep` in the Verilog in case there's some weirdness going on during synthesis, didn't help. I also tried playing around with the power net, removing stripes that might make it harder to connect pads to the core power ring.

And just as a side note, the same behaviour can be seen with one of IHP's closed source I/O libraries: a few missing input-only pads and a `PSM-0042` crash pretty late into the flow.

Here's the referenced build: [cant_connect.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13502638/cant_connect.tar.gz)
And a screenshot of that build:, which nicely shows how `u_din` is not connected to the power ring. 
[image]

maliberty, 2023-11-29:
Without remove_buffers:
[image]

Syndace, 2023-11-29:
Aaah thank you, what a funny bug :D

It seems the `PSM-0042` is an unrelated problem then, I'll open a new issue for that.

Syndace, 2023-11-29:
Also, do you not want to keep this issue open until there is a fix? In my eyes this is definitely a bug. The cells/macros placed during floorplanning should never be touched by the following steps. Or is my expectation wrong?

maliberty, 2023-11-29:
Real pad cells never look like buffers so this doesn't come up - they always have many control pins.  We could skip anything that is placed in remove_buffers if you feel this use case is important.

Syndace, 2023-11-29:
I have to disagree - I have the same issue with a proprietary closed-source I/O library that has been used for years in numerous successful tapeouts. The open-source cells I am using here are "real" too, we plan to tape out with them soon and they will become an official part of IHP's open source PDK. What kind of control pins would you expect on a pad whose sole purpose is to be a fast input or output cell?

maliberty, 2023-11-29:
Perhaps the name IOLib_dummy.lib is misleading then.  Perhaps I'm used to seeing more configurable cells than what you are using.  What do you think of my proposed solution?

The other option would be to add a dont-touch to the pad instances

Syndace, 2023-11-29:
> Perhaps the name IOLib_dummy.lib is misleading then.

The liberty is called dummy right now because it only contains the bare structural information needed by the flow to use the cells, inspired by [sky130's dummy_io.lib](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130io/lib/sky130_dummy_io.lib). A full liberty with all the bells and whistles is planned, however the current priority is general cell quality, DRC cleanness etc.
The closed-source I/O library I've tested with has fully fledged liberties for three process corners.

> Perhaps I'm used to seeing more configurable cells than what you are using.

Maybe :D I remember that `cut_rows` was not a thing either until I asked for it, because apparently most modern PDKs use tapcells and IHP is a bit of an outlier. Maybe their I/O pads are also unusually simplistic.

> What do you think of my proposed solution?

The solution that feels cleanest to me is marking cells placed during floorplanning as dont-touch, since that matches my expectations. I expect macros I've placed during floorplanning to appear at that exact position in the final GDS. However, I'm also fine with remove_buffers just skipping the placed cells, whatever you think is the better solution. Skipping `remove_buffers` or modifying the I/O library are fine as workarounds, but I wouldn't consider them solutions.

maliberty, 2023-11-29:
If I simply add dont-touch during the place_* commands then connect_by_abutment will fail (it modifies the iterms connections).  I could have connect_by_abutment override dont-touch but that seems dangerous as well.  

My other suggestion is for remove_buffer to skip over anything already placed.  That seems a bit simpler.

Your thoughts?

Syndace, 2023-11-29:
Sounds good to me!