The-OpenROAD-Project/OpenROAD-flow-scripts #1667: Flow is unable to connect some pads to the power net `ERROR PSM-0042` (ihp-sg13g2)

Syndace, 2023-11-29:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

The flow fails to connect certain pads to the power net and aborts with `ERROR PSM-0042`.

### Expected Behavior

-

### Environment

```shell
Using the docker container, always on the newest master commit.
```

### To Reproduce

The following is a minimal self-contained example design to reproduce the issue: [cant_connect.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13502638/cant_connect.tar.gz)

Notice how the flow aborts with

```
[ERROR PSM-0042] Unable to connect macro/pad Instance u_din to the power grid.
```

and inspect the result in the GUI to confirm that `u_din` and a few other pads are indeed not connected to the core power ring.

### Screenshots

[image]

### Additional Context

I've tried playing with the pdn and remove stripes on layers that could potentially cause routing issues, without success.

maliberty, 2023-11-30:
Just an aside as I started to look at your test case "CLASS PAD ;" could be more descriptive as the syntax allows for more specific types " PAD [INPUT | OUTPUT |INOUT | POWER | SPACER | AREAIO]" which we use in the GUI
[image]

maliberty, 2023-11-30:
@gadfort do you have some time to look at this?  I see some IO cells are tied to the core ring but others aren't.  The highlighted instance is the one psm complains about:
[image]

gadfort, 2023-11-30:
@maliberty I can take a look. But when I was fixing the other issue the other day, I noticed that these are things that check_power_grid should have complained about, so I might detour a little in the fix to fix this problem at a slightly deeper place.

maliberty, 2023-12-02:
Corner is not an AREAIO.  You could mark it SPACER or leave it unspecified.

Does the tri-state output act as an input too or is it just possible to float the output?

stafverhaegen-chipflow, 2023-12-03:
> Corner is not an AREAIO. You could mark it SPACER or leave it unspecified.

OK.
 
> Does the tri-state output act as an input too or is it just possible to float the output?

Only able to float the output.

maliberty, 2023-12-03:
Then it sounds like an output as there is no finer specification.  Its mostly for convenience in the GUI display controls.

maliberty, 2023-12-20:
Is the intention that every signal pad should be tied to the core rings or just the vdd/vss ones?

stafverhaegen-chipflow, 2023-12-21:
The intention is that the IO cells should make it possible to use the vdd/vss track rings inside the IO ring and don't need to make own vss/vdd ring arounf the standard cells; this is dione by allowing to connect to the vdd/vss pin in each of the IO cells and not only in the IOPadVdd/IOPadVss cells. The intention is that the cells are still compatible with a flow that generates it's own vdd/vss ring around the standard cells and only connects to the IO ring vdd/vss pins in the IOPadVss and IOPadVdd IO cells though.  
After further inspection the execution was buggy. The vdd/vss pin on Metal2 to TopMetal1 was supposed to touch the corresponding OBS layer so one should be able to route to the vdd/vss pins on these layers. This was not the case anymore as a change in layout generation was not reflected in the LEF generation. Additionally the vdd/vss pins on TopMetal1 layer were covered by a TopMetal2 OBS layer so one could not route to it on TopMetal2 and a TopVia2 either making the pins actually inaccessible.  
I now delivered a new set of IO cells so I guess @Syndace should now see if this problem has been resolved.

maliberty, 2023-12-21:
Do you think these IO cells should be added to ORFS for others to use?

maliberty, 2023-12-21:
@Syndace if problems persist please update with a new reproducer, otherwise close.

Syndace, 2023-12-21:
Issue persists with updated I/O cells. New reproducer: 
[cant_connect.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13744299/cant_connect.tar.gz)

> Do you think these IO cells should be added to ORFS for others to use?

The I/O cells will become part of IHP's open PDK, at that point it would be nice to also add them to ORFS.

maliberty, 2023-12-22:
Is it intentional how the pins change Y position on the different layers, eg
```
  PIN vdd
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER Metal3 ;
        RECT 0.000 160.000 80.000 178.000 ;
    END
    PORT
      LAYER Metal4 ;
        RECT 0.000 140.000 80.000 155.800 ;
    END
    PORT
      LAYER Metal5 ;
        RECT 0.000 140.000 80.000 158.000 ;
    END
    PORT
      LAYER TopMetal1 ;
        RECT 0.000 140.000 80.000 158.000 ;
    END
  END vdd
```

maliberty, 2023-12-22:
On the east side you have:
[image]

The Metal5 rings have core connections on TopMetal1.  You will also have to bring a connection from the inner VDD pin up to TopMetal2 and down onto the Metal5 ring.  The Y value at which you do so is constrained by how the core is connecting.

The current pad code doesn't try to pre-plan this and may run into conflicts, in which case it won't put in the strap.   We could cut back the core connection but that might not be good either given how sparse the straps are.  How do you envision this connection being made?

maliberty, 2023-12-22:
I think this is why we don't see the problem on N/S because the ring is using TopMetal1 and the core is connecting from below on Metal5

stafverhaegen-chipflow, 2023-12-22:
> Is it intentional how the pins change Y position on the different layers, eg

Yes, this is intentional to allow connection to vdd pin on Metal3 and to vss pin on Metal4 to TopMetal1.

Syndace, 2023-12-22:
> I think this is why we don't see the problem on N/S because the ring is using TopMetal1 and the core is connecting from below on Metal5

Indeed, disabling the horizontal TM1 stripes resolves the connection issue for all cells except for `IOPadVdd` and `IOPadVss`. @stafverhaegen-chipflow can you double-check those two specifically please?
 
[image]

> How do you envision this connection being made?

Good question. On one hand, there surely is enough space to make the connection, the flow could try to avoid collisions with the stripes. On the other hand, given how many connections there are already, it doesn't matter if one or two padcells are not connected to the core power ring.

On a different note, I am a little surprised that this is an error, since the vss/vdd of the cells should be considered connected by abutment no?

Syndace, 2023-12-22:
> Good question. On one hand, there surely is enough space to make the connection, the flow could try to avoid collisions with the stripes. On the other hand, given how many connections there are already, it doesn't matter if one or two padcells are not connected to the core power ring.

Actually, I want to take that back. In the long term, I think the flow should 100% look for a solution to connect the pads, rather than giving up on the first collision. The alternatives are:

- The user is required to carefully plan power stripe offset, pitch and padcell locations to avoid collisions
- The user is required to only connect stripes to the power ring from below
- The flow has to silently/with a warning ignore the missing connections

I don't think any of these options is acceptable, given that the connection is very well possible.

stafverhaegen-chipflow, 2023-12-22:
> 
> Indeed, disabling the horizontal TM1 stripes resolves the connection issue for all cells except for `IOPadVdd` and `IOPadVss`. @stafverhaegen-chipflow can you double-check those two specifically please?

`IOPadVss` has the same pins as the other cells only difference is that the pad pin is now part of the vss port as there internally is a direct connection from the pad to the vss track and thus the pin.
Same for `IOPadVdd` but there is an additional pin on TopMetal2 as I use TopMetal2 inside this cell to connect the bondpad connection to the vdd track.
AFAICS this is how it should be done in LEF unless somebody else sees a problem ?

gadfort, 2023-12-31:
@Syndace there are two sets of issues going on here.
1. PDN is rejecting the pad connections when you end up with a intermediate via (going from TM2 to M5 with the TM1 strap getting in the way). This could be addressed with a some refinement to detect this condition and attempt to move/resize the connection to get it to fit. I'll see what can be done for that.
2. PSM doesn't handle abutment connectivity well and should probably have failed on the `check_power_grid` call but it doesn't seem to perform some checks until the analyze call. I think PSM needs some fixing to help it work faster and be consistent in the checking.

@stafverhaegen-chipflow The way the LEF is constructed it it implies that all the PORTs are weakly connected and need to to be connected externally to deliver power to the cell. This is probably not true so the LEF probably needs to be corrected to reflect that. According to the LEF spec: 
```
Power routers should connect to every PORT statement, if there is more than one for a
given PIN. For example, if a block has several PORTs on the boundary for the VSS PIN,
each PORT should be connected by the power router.
```
You might also want to consider using the `CLASS` specification to define the port you want to connect to the pad cell via the `CLASS BUMP` since that would help PDN/PSM ignore that shape.

gadfort, 2024-01-07:
@Syndace @stafverhaegen-chipflow I think I have something working, except `u_vddcore` with the LEF as it is, PDN will insert TopVia2 vias onto the VDD pin (but in the GDS where is already vias there), so that would be an issue in DRC. I think the LEF needs to have the VIAs blocked otherwise there is nothing to stop PDN from blindly dropping vias.
[image]

So when you run `analyze_power_grid` it will still generate that error for this instance.

stafverhaegen-chipflow, 2024-01-08:
Not that easy for me to understand what is going on but I think what is happening is that you insert TopVia2 in a region that is covered both by TopMetal2 pin and obstruction in the LEF. I don't think it should be legal to insert TopVia2 in such a region and you should just contact the pin on TopMetal2 without inserting TopVia2.

stafverhaegen-chipflow, 2024-01-08:
> @stafverhaegen-chipflow The way the LEF is constructed it it implies that all the PORTs are weakly connected and need to to be connected externally to deliver power to the cell. This is probably not true so the LEF probably needs to be corrected to reflect that.

The only use case I support at the moment with my IO cells is the generation of a full IO ring using only cells provided in the library. The connection of each port of the power pins should be fulfilled by abutment of the cells and the weak connection is used to discourage the use of other cells/connections as much as possible.

maliberty, 2024-01-08:
If an area is covered by both a pin shape and an obstruction then the pin shape always wins (that's the LEF standard).  You can't go outside the shape but you can legal connect to the pin in a fully enclosed manner.

stafverhaegen-chipflow, 2024-01-09:
> If an area is covered by both a pin shape and an obstruction then the pin shape always wins (that's the LEF standard). You can't go outside the shape but you can legal connect to the pin in a fully enclosed manner.

I know but what we are discussing here is insertion of TopVia2 to connect to a pin on TopMetal1 in a region that is covered by pin and obstruction on TopMetal2 for the same signal. I don't see why you need to insert the TopVia2 as you are already connected to the pin on TopMetal2 anyway.  
I think this is a corner case that is not directly covered by the standard but I will add Via blockage layer under the pins to make clear that one should not insert vias in between such pins.

closed 2024-01-11 by maliberty