The-OpenROAD-Project/OpenROAD-flow-scripts #1669: Error when running megaboom

ApeachM, 2023-11-30:
### Subject

[Documentation] for documentation errors.

### Describe the bug

Running the flow with `make` at `flow` director for megaboom, the error is occurred.

```
ERROR: ABC: execution of command "/home/mik077/tmp/tmp.d5g0Zhroht/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-IVhrUr/abc.script 2>&1" failed: return code 127.
Command exited with non-zero status 1
Elapsed time: 36:07.87[h:]min:sec. CPU time: user 2104.34 sys 16.35 (97%). Peak memory: 10213528KB.
```

### Expected Behavior

Doing flow for megaboom

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 4.18.0-522.el8.x86_64
os: CentOS Stream 8
cmake version 3.25.1
-- The CXX compiler identification is GNU 9.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/tool/gcc/gcc-9.3.0/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10770-g6719c92
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/tool/gcc/gcc-9.3.0/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.6 (found version "3.6.8") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 97 lines cut ...]
-- Found Boost: /home/tool/boost/boost_1_78_0/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /home/tool/boost/boost_1_78_0/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.XoxGDtEN7M
```

### To Reproduce

```
cd flow
# Make the comment of `DESIGN_CONFIG=./designs/asap7/megaboom/config.mk` as real code
make -j
```

### Relevant log output

```shell
Here is the log [link](https://postechackr-my.sharepoint.com/:u:/g/personal/kmj0824_postech_ac_kr/ER_M0oZkQNBCh0PmDSzsAmkB_vBH0ibrfSk56YFsi0Fjjg?e=xOB4hJ), `for 1_1_yosys.log`.
```

maliberty, 2023-12-01:
@oharboe does this work for you?  This design isn't in the CI so I have no idea if it should work

rovinski, 2023-12-01:
Error code 127 is usually related to command or file not found. Can you double check that your Yosys and ABC are properly installed? Additionally, can you check that your `/tmp` folder has space? The `/tmp` disk may be full so the file `/tmp/yosys-abc-IVhrUr/abc.script` cannot be written.

oharboe, 2023-12-01:
> @oharboe does this work for you? This design isn't in the CI so I have no idea if it should work

I don't use megaboom in ORFS and I don't think it is realistic to work with 24hours+ designs in ORFS. I would recommend deleting it so as to not cause false expectations.

oharboe, 2023-12-01:
@ApeachM Try running `make DESIGN_CONFIG=designs/asap7/gcd/config.mk` first. This is a small design that can verify your OpenROAD and ORFS setup.

ApeachM, 2023-12-01:
Oh, I am sorry. I think it is not matter of megaboom. 

I did `make DESIGN_CONFIG=designs/asap7/gcd/config.mk` but there was an error like below.

```
[mik077@dgx flow]$ make DESIGN_CONFIG=designs/asap7/gcd/config.mk -j
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /home/tool/klayout/install/0.27.1-c8/bin/klayout -zz -rd design_name=gcd \
        -rd in_def=./results/asap7/gcd/base/6_final.def \
        -rd in_files="/home/mik077/tmp/orfs/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds  " \
        -rd config_file= \
        -rd seal_file="" \
        -rd out_file=./results/asap7/gcd/base/6_1_merged.gds \
        -rd tech_file=./objects/asap7/gcd/base/klayout.lyt \
        -rd layer_map= \
        -r /home/mik077/tmp/orfs/OpenROAD-flow-scripts/flow/util/def2stream.py) 2>&1 | tee ./logs/asap7/gcd/base/6_1_merge.log
/home/tool/klayout/install/0.27.1-c8/bin/klayout: error while loading shared libraries: libruby.so.2.5: cannot open shared object file: No such file or directory
Command exited with non-zero status 127
Elapsed time: 0:00.00[h:]min:sec. CPU time: user 0.00 sys 0.00 (100%). Peak memory: 2400KB.
make[1]: *** [Makefile:901: do-gds-merged] error 127
make: *** [Makefile:896: results/asap7/gcd/base/6_1_merged.gds] error 2

```

And when I see the files in  `OpenROAD-flow-scripts/tools/install`, there are file following below.
Is there anything that I missed?

```
[mik077@dgx install]$ pwd
/home/mik077/tmp/orfs/OpenROAD-flow-scripts/tools/install
[mik077@dgx install]$ tree .
.
├── OpenROAD
│   ├── bin
│   │   ├── openroad
│   │   └── sta
│   ├── include
│   │   └── sta
│   │       ├── ArcDelayCalc.hh
│   │       ├── ArrayTable.hh
│   │       ├── Bfs.hh
│   │       ├── ClkNetwork.hh
│   │       ├── Clock.hh
│   │       ├── ClockGatingCheck.hh
│   │       ├── ClockGroups.hh
│   │       ├── ClockInsertion.hh
│   │       ├── ClockLatency.hh
│   │       ├── ConcreteLibrary.hh
│   │       ├── ConcreteNetwork.hh
│   │       ├── Corner.hh
│   │       ├── CycleAccting.hh
│   │       ├── DataCheck.hh
│   │       ├── DcalcAnalysisPt.hh
│   │       ├── Debug.hh
│   │       ├── Delay.hh
│   │       ├── DelayCalc.hh
│   │       ├── DelayFloat.hh
│   │       ├── DelayNormal1.hh
│   │       ├── DelayNormal2.hh
│   │       ├── DeratingFactors.hh
│   │       ├── DisabledPorts.hh
│   │       ├── DispatchQueue.hh
│   │       ├── EnumNameMap.hh
│   │       ├── EquivCells.hh
│   │       ├── Error.hh
│   │       ├── ExceptionPath.hh
│   │       ├── FuncExpr.hh
│   │       ├── Fuzzy.hh
│   │       ├── Graph.hh
[... 346 lines cut ...]
                ├── lut_map.v
                ├── lutrams_map.v
                ├── mux_map.v
                ├── xc2v_brams.txt
                ├── xc2v_brams_map.v
                ├── xc3s_mult_map.v
                ├── xc3sa_brams.txt
                ├── xc3sda_brams.txt
                ├── xc3sda_dsp_map.v
                ├── xc4v_dsp_map.v
                ├── xc5v_dsp_map.v
                ├── xc6s_brams.txt
                ├── xc6s_brams_map.v
                ├── xc6s_dsp_map.v
                ├── xc7_brams_map.v
                ├── xc7_dsp_map.v
                ├── xc7_xcu_brams.txt
                ├── xcu_brams_map.v
                ├── xcu_dsp_map.v
                ├── xcup_urams.txt
                └── xcup_urams_map.v

48 directories, 356 files

```

maliberty, 2024-03-06:
Please reopen if still relevant.

closed 2024-03-06 by maliberty