The-OpenROAD-Project/OpenROAD-flow-scripts #1671: Corners outside of chip area in final GDS

Syndace, 2023-12-01:
### Subject

[Stage]: GDS generation.

### Describe the bug

When inspecting the final GDS with KLayout, I noticed the padframe corner cells are placed outside of the chip area. In the DEF (via `make gui_final`), everything looks fine. Screenshots below.

### Expected Behavior

.

### Environment

```shell
Using the docker container, always on the newest master commit. I'm also on the newest commit of OpenROAD, since it includes the fix for https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1646.
```

### To Reproduce

The following is a minimal self-contained example design to reproduce the issue: 
[corner_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13525398/corner_issue.tar.gz)

Included in the .tar.gz, in the `results` folder, are the full log, the final GDS and copies of the screenshots I've attached below.

### Screenshots

[image]
[image]

### Additional Context

I had a very similar issue a year ago, when I used OpenLane and called the "old" padframe commands: https://github.com/The-OpenROAD-Project/OpenROAD/issues/2253
This might be related.

maliberty, 2023-12-01:
The gds for corner has 0,0 at the lower right instead of the lower left.

maliberty, 2023-12-01:
I see you have in the LEF
```
MACRO Corner
  CLASS PAD ;
  ORIGIN 180.000 -0.000 ;
```

@gadfort might you look at this as well?

maliberty, 2023-12-14:
@gadfort ??

Syndace, 2023-12-14:
This might be a bug in the I/O library, it seems to me that the origin mismatches between LEF and GDS. I have asked the author of the I/O library and will report back once I hear from them.

gadfort, 2023-12-14:
@maliberty sorry I missed this. Like @Syndace it seems that the GDS has a different origin from the LEF

Syndace, 2023-12-18:
Okay solved it, it was indeed a bug in the I/O library. Sorry for opening this issue and not checking the I/O library first.

For reference, LEF and GDS don't need to have matching origins; the `FOREIGN` LEF statement is what glues LEF and GDS together. In our case, the `FOREIGN` statement was incorrect for the corner, it specified the GDS origin of the corner as (0,0), when it is (-180,0) in reality.

closed 2023-12-18 by Syndace