The-OpenROAD-Project/OpenROAD-flow-scripts #1672: Error running gcd

ApeachM, 2023-12-02:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I am trying to running ORFS in Window.
I installed WSL2 and Docker.

In the WSL2, 
```shell
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh 

docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-centos7-builder
```
And I could check the install is finished with below command.
```shell
source ./env.sh
yosys -help
openroad -help
```

To build examples, 
```shell

sudo ./setup.sh
cd flow
util/docker_shell make
```
but there was an error for running following belows.
```shell
...
21.8. Executing OPT_EXPR pass (perform const folding).
21.9. Rerunning OPT passes. (Maybe there is more to do..)
21.10. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
21.11. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
21.12. Executing OPT_MERGE pass (detect identical cells).
21.13. Executing OPT_DFF pass (perform DFF optimizations).
21.14. Executing OPT_CLEAN pass (remove unused cells and wires).
21.15. Executing OPT_EXPR pass (perform const folding).
21.16. Finished OPT passes. (There is nothing left to do.)
Using ABC speed script.
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 390
22. Executing ABC pass (technology mapping using ABC).
22.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
22.1.1. Executing ABC.
sh: line 1:   124 Aborted                 /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-6w717U/abc.script 2>&1
ERROR: ABC: execution of command "/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-6w717U/abc.script 2>&1" failed: return code 134.
Elapsed time: 0:06.13[h:]min:sec. CPU time: user 5.90 sys 0.22 (99%). Peak memory: 301692KB.
make: *** [results/asap7/gcd/base/1_1_yosys.v] Error 1
```

When I run
```shell
util/docker_shell make DESIGN_CONFIG=designs/asap7/gcd/config.mk
```
then the error is occurred below.
```shell
...

21.15. Executing OPT_EXPR pass (perform const folding).
21.16. Finished OPT passes. (There is nothing left to do.)
Using ABC speed script.
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 390
22. Executing ABC pass (technology mapping using ABC).
22.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
22.1.1. Executing ABC.
sh: line 1:   112 Aborted                 /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-m5Q7aE/abc.script 2>&1
ERROR: ABC: execution of command "/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-m5Q7aE/abc.script 2>&1" failed: return code 134.
Elapsed time: 0:05.97[h:]min:sec. CPU time: user 5.81 sys 0.16 (100%). Peak memory: 301600KB.
make: *** [results/asap7/gcd/base/1_1_yosys.v] Error 1
```

### Expected Behavior

Running correctly 

### Environment

```shell
docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-centos7-builder

Running the docker container, and I did 
```
cd tools/OpenROAD
./etc/Env.sh
```
But the message is abnormal. Is the reason that I used the docker? When I run ./openroad in tools/install/OpenROAD, it works well..

```
bash-4.2$ ./etc/Env.sh
tee: build/openroad-env.log: Permission denied
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 5.15.133.1-microsoft-standard-WSL2
os: CentOS Linux 7 (Core)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

CMake Error at CMakeLists.txt:112 (message):
  Insufficient gcc version.  Found 4.8.5, but require >= 8.3.0.

-- The CXX compiler identification is GNU 4.8.5
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: HEAD-HASH-NOTFOUND
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Mentioned above.

chinayly, 2023-12-07:
I have the same question in Ubuntu22.04, I still cannot solve it

chinayly, 2023-12-07:
I have the same question in running the OpenROAD-flow-script on Ubuntu22.04, I still cannot solve it

maliberty, 2023-12-07:
@vvbandeira please answer re: eqy

vijayank88, 2023-12-07:
Duplicate of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1513 
@chinayly  @ApeachM 
Check this comment: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035#issuecomment-1736369365
You may use prebuilt one as per this comment: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4035#issuecomment-1734753815

vvbandeira, 2024-01-16:
@chinayly @ApeachM 
Could you try again with the latest code? Please also make sure to `source env.sh` before running the flow if you used `sudo ./setup.sh` to install the dependencies.

maliberty, 2024-03-05:
Stale - reopen if still relevant

closed 2024-03-05 by maliberty