The-OpenROAD-Project/OpenROAD-flow-scripts #1673: Error when building with ubuntu 20

ApeachM, 2023-12-02:
### Subject

[Documentation] for documentation errors.

### Describe the bug

In the WSL2, I did run
```
./build_openroad --os=ubuntu20.04
```

But the building is failed with below error message

```
...
#8 472.5 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddGroup.c
#8 472.5 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddHarwell.c
#8 472.6 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddInit.c
#8 472.6 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddInteract.c
#8 472.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddLCache.c
#8 472.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddLevelQ.c
#8 472.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddLinear.c
#8 472.8 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddLiteral.c
#8 472.8 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddMatMult.c
#8 472.8 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddPriority.c
#8 472.8 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddRead.c
#8 472.9 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddRef.c
#8 472.9 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddReorder.c
#8 473.0 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSat.c
#8 473.1 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSign.c
#8 473.1 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSolve.c
#8 473.1 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSplit.c
#8 473.2 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSubsetHB.c
#8 473.2 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSubsetSP.c
#8 473.3 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddSymmetry.c
#8 473.3 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddTable.c
#8 473.4 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddUtil.c
#8 473.5 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddWindow.c
#8 473.5 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddCount.c
#8 473.6 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddFuncs.c
#8 473.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddGroup.c
#8 473.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddIsop.c
#8 473.7 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddLin.c
#8 473.8 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddMisc.c
#8 474.0 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddPort.c
#8 474.0 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddReord.c
#8 474.0 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddSetop.c
#8 474.1 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddSymm.c
#8 474.2 [ 94%] ABC: `` Compiling: /src/bdd/cudd/cuddZddUtil.c
#8 474.2 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddAuto.c
#8 474.2 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddCas.c
#8 474.3 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddImage.c
#8 474.4 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddKmap.c
#8 474.4 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddMaxMin.c
#8 474.5 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddMisc.c
#8 474.6 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddSet.c
#8 474.7 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddSymm.c
#8 474.7 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddThresh.c
#8 474.7 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddTime.c
#8 474.7 [ 94%] ABC: `` Compiling: /src/bdd/extrab/extraBddUnate.c
#8 474.7 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdApi.c
#8 474.8 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdCheck.c
#8 474.9 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdLocal.c
#8 474.9 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdMan.c
#8 475.0 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdProc.c
#8 475.0 [ 94%] ABC: `` Compiling: /src/bdd/dsd/dsdTree.c
#8 475.0 [ 94%] ABC: `` Compiling: /src/bdd/epd/epd.c
#8 475.0 [ 94%] ABC: `` Compiling: /src/bdd/mtr/mtrBasic.c
#8 475.1 [ 94%] ABC: `` Compiling: /src/bdd/mtr/mtrGroup.c
#8 475.1 [ 94%] ABC: `` Compiling: /src/bdd/reo/reoApi.c
[... 69 lines cut ...]
--------------------
ERROR: failed to solve: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 1

```

### Expected Behavior

Building Well

### Environment

```shell
I used the docker, so I could make normal log.
```

### To Reproduce

In the WSL2, ubuntu 22 which is installed only git and docker, 

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh  --os=ubuntu20.04

```

maliberty, 2023-12-02:
Probably you want to add --local to avoid building in docker.  If you really want to build in docker then you should remove your local build directory.

maliberty, 2024-03-06:
Please reopen if this is a still an issue

closed 2024-03-06 by maliberty