The-OpenROAD-Project/OpenROAD-flow-scripts #1677: Overlapping vias on signal connections to pads

Syndace, 2023-12-04:
### Subject

[Stage]: I/O Placement.

### Describe the bug

The flow generated overlapping vias on some signal connections to the padframe.

### Expected Behavior

.

### Environment

```shell
Docker on newest commit.
```

### To Reproduce

The following is a minimal self-contained example design to reproduce the issue: 
[corner_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13525398/corner_issue.tar.gz)

(nevermind the name - this is an example project for a different issue that just happens to also reproduce this issue)

Included in the .tar.gz, in the `results` folder, are the full log and the final GDS for your convenience.

To reproduce, build the project, inspect the final GDS, hide all layers but Via2 and check the signal connections to the padframe. In `results/6_final.gds`, the issue appears at 318.5 421.2, see screenshot below.

### Screenshots

[image]

maliberty, 2023-12-04:
@osamahammad21 I see that the macro pin has shapes on M2 & M3:
[image]

but all the APs on on M2 and drt puts a via to try to land on the M2 leading to the above.  The grt guide is just on M3 so it appears to be a pure drt issue.

maliberty, 2023-12-04:
@Syndace as a workaround you could convert the M2 to OBS instead of pin shapes.

osamahammad21, 2023-12-07:
@maliberty, why is it a drt issue? The M3 shape is not on track, thus making pa prefer on track m2 access points. As far as DRT aware, it is not violating anything.

maliberty, 2023-12-07:
There is a cut shape in the pin and we are putting in another via leading to the overlapping cuts.

osamahammad21, 2023-12-07:
But the cut shape isn't visible to the detailed router, right? That's as far as I could tell from running gui_route. There is only one via2 shape at that location which is coming from the net accessing the pin.

maliberty, 2023-12-07:
I believe there is a cut shape in the LEF pin but could you double check

osamahammad21, 2023-12-10:
No it's not  in the lef:
MACRO IOPadInOut4mA
  CLASS PAD ;
  ORIGIN -0.000 -0.000 ;
  FOREIGN IOPadInOut4mA 0 0 ;
  SIZE 80.000 BY 180.000 ;
  SYMMETRY X Y R90 ;
  SITE IOLibSite ;
  PIN c2p
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER Metal2 ;
        RECT 18.330 178.090 18.620 180.000 ;
      LAYER Metal3 ;
        RECT 18.225 179.710 18.725 180.000 ;
    END
  END c2p

maliberty, 2023-12-10:
@Syndace it looks like you LEF abstract is missing the cut shape

Syndace, 2023-12-13:
Maybe I'm misunderstanding something, but there is routing obstruction on Metal2 where the overlapping via appears:

```
  OBS
    LAYER Metal1 ;
      RECT 0.000 0.000 80.000 180.000 ;
    LAYER Metal2 ;
      RECT 0.000 0.000 80.000 180.000 ;
    LAYER Metal3 ;
      RECT 0.000 0.000 80.000 179.000 ;
    ...
```

[image]
[image]

osamahammad21, 2023-12-19:
The routing obstruction prevents the router from routing through the obstruction but it does not prevent macro pin access. So the router is allowed to have this metal 2 segment just to access the pin. If you do not want it to do so, you could just remove the metal2 pin shape.

stafverhaegen-chipflow, 2023-12-19:
Just chiming in. The cut shape is not in the LEF file indeed. But as the area is covered both by Metal2 and Metal3 OBS, I would think you should not be able to add a Via2 in this region as it involves also adding Metal2 and Metal3.
So it is really legal to add a cut in LEF regions that are covered by top and bottom OBS ?
Both Metal2 and Metal3 pins are touching the edge of the OBS layer so should just be able to be connected without adding a cut.

stafverhaegen-chipflow, 2023-12-19:
Sorry, I was wrong, area is not covered by Metal3 OBS and I agree this has to be fixed in the LEF.

Syndace, 2023-12-19:
Cool, thanks everybody, I'll close this then :)

closed 2023-12-19 by Syndace