The-OpenROAD-Project/OpenROAD-flow-scripts #1681: Metal2 minimum area violations

Syndace, 2023-12-04:
### Subject

[Stage]: Detail Router.

### Describe the bug

The flow creates Metal2 islands that are smaller than the minimum area required by the tech LEF. Here's the relevant excerpt from [the tech LEF](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/ihp-sg13g2/lef/sg13g2_tech.lef):

```
LAYER Metal2
  ...
  AREA      0.144 ;
  ...
END Metal2
```

And here's a GDS produced by the flow which has two Metal2 islands that are smaller than the required area: 
[6_final.gds.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13551076/6_final.gds.tar.gz)

The first violating island is at (62.495, 35.600), the second is at (52.330, 62.480). The screenshots below show the first island, once in KLayout and once in `make gui_route`.

### Expected Behavior

The router should create patch metal to meet the minimum area requirement. 

### Environment

```shell
Docker container, newest commit.
```

### To Reproduce

Is `make {script}_issue` actually a thing? I tried `make route_issue` but that doesn't exist.

To reproduce, simply build any of the example projects for the ihp-sg13g2 PDK. I don't know how consistently this issue appears, but even on the very small SPI project (GDS of my build attached above) I had two violating Metal2 islands. On larger projects there are a lot more, easily in the 100s. The problem is finding them; I have access to a proprietary tool that allows me to find Metal2 islands that don't meet the minimum area requirement. I am happy to run the tool on your GDSes and give you coordinates of violations if that helps. The KLayout DRC script included with ihp-sg13g2 is sadly not capable of finding them (yet).

### Screenshots

[image]
[image]

### Additional Context

I'd be happy about any sort of quick and dirty workaround for now. I'm also happy to assist in any way.

maliberty, 2023-12-04:
make detail_route_issue

Syndace, 2023-12-04:
Thanks: [detail_route_metal2_min_area.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13552193/detail_route_metal2_min_area.tar.gz)

Syndace, 2023-12-17:
FYI, this can be worked around by increasing the minimum area in the tech LEF. This means that metal patches are created and their size is affected by the area rule, the patches are just not large enough sometimes.

maliberty, 2023-12-17:
@refaay @osamahammad21 would you update on this?

refaay, 2023-12-18:
I have been working on debugging this issue for some time and discussing with Osama.
There's an issue with dr area patching as it considers all the vias used to be the default via, while sometimes this is not the case. I guess increasing the minArea in lef solution mentioned solved this problem as sometimes the gapArea calculated is less than the actual gapArea.
There's another issue with gc patching as it correctly adds the needed patch afterwards, but this patch completely disappears after that. I am trying to see why and debugging this issue that may be related to the patch being on border.

osamahammad21, 2024-04-25:
Fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/4994

closed 2024-04-25 by osamahammad21