The-OpenROAD-Project/OpenROAD-flow-scripts #1682: Error in placment stage for a design with only macros

b224hisl, 2023-12-05:
### Subject

[Stage]: Global Placement.

### Describe the bug

I'm hardening a design with just 4 macros. It fails in `make: *** [Makefile:635: results/sky130hd/fakeram4/fakeram4/3_1_place_gp_skip_io.odb]`.
I have check the source code, that error seems to come from 
[image]
And I would like to know how to handle it 

### Expected Behavior

finish without error

### Environment

```shell
kernel: Linux 3.10.0-1160.76.1.el7.x86_64
os: Ubuntu 20.04.6 LTS (Focal Fossa)
```

### To Reproduce

[global_place_skip_io_fakeram4_sky130hd_fakeram4_2023-12-05_11-34.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13554770/global_place_skip_io_fakeram4_sky130hd_fakeram4_2023-12-05_11-34.tar.gz)

### Relevant log output

```shell
OpenROAD 4759105234654bd6e30af9179e9523c38ede62a5 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 1380 2720
[INFO GPL-0005] CoreAreaUxUy: 153640 92480
[INFO GPL-0006] NumInstances: 70
[INFO GPL-0007] NumPlaceInstances: 0
[INFO GPL-0008] NumFixedInstances: 4
[INFO GPL-0009] NumDummyInstances: 66
[INFO GPL-0010] NumNets: 201
[INFO GPL-0011] NumPins: 377
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 155000 95000
[INFO GPL-0014] CoreAreaLxLy: 1380 2720
[INFO GPL-0015] CoreAreaUxUy: 153640 92480
[INFO GPL-0016] CoreArea: 13666857600
[INFO GPL-0017] NonPlaceInstsArea: 11313350400
[INFO GPL-0018] PlaceInstsArea: 0
[INFO GPL-0019] Util(%): 0.00
[INFO GPL-0020] StdInstsArea: 0
[INFO GPL-0021] MacroInstsArea: 0
[WARNING GPL-0136] No placeable instances - skipping placement.
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::shared_ptr<gpl::NesterovBase>; _Alloc = std::allocator<std::shared_ptr<gpl::NesterovBase> >; std::vector<_Tp, _Alloc>::reference = std::shared_ptr<gpl::NesterovBase>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x00005641ECC7C0A5 in /usr/bin/openroad
 1# 0x00007F47DF73E090 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x00005641ECCC4ACD in /usr/bin/openroad
 5# 0x00005641ECD2C95E in /usr/bin/openroad
 6# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007F47E3409924 in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
10# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /usr/bin/openroad
11# ord::tclAppInit(Tcl_Interp*) in /usr/bin/openroad
12# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# main in /usr/bin/openroad
14# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
15# _start in /usr/bin/openroad
Command terminated by signal 6
Elapsed time: 0:00.64[h:]min:sec. CPU time: user 0.37 sys 0.06 (67%). Peak memory: 83804KB.
```

maliberty, 2023-12-05:
Please package a test case with 'make global_place_skip_io_issue'

b224hisl, 2023-12-06:
> Please package a test case with 'make global_place_skip_io_issue'

I have updated this file here: 
[image]

maliberty, 2023-12-06:
Its best to open new issues for unrelated problems.  I've reopened this for the pdn issue.

maliberty, 2023-12-06:
There appear to be no vias between the power straps and the macro.  A starting problem is that -cell takes a master not an instance.  So -cells {fakeram7_256x32} not -cells {dmem0}.  If you really want to use instance names then use -instances ...

maliberty, 2023-12-06:
Then you see the real issue, these show no vias are inserted:
```
[WARNING PDN-0110] No via inserted between met4 and met5 at (316.8800, 97.1200) - (317.8800, 98.7200) on VSS
```
I'll look at that a bit further

closed 2023-12-06 by maliberty

b224hisl, 2023-12-07:
> instances

yes, but do you know how to handle this problem?

b224hisl, 2023-12-07:
> I think the problem is your fakeram. The power pins are too close together to drop a generated via onto.

So how can I adjust the power pins' location? It seems that no power pins like `input VDD; input VSS` defined in all of the verilog files.

b224hisl, 2023-12-07:
No matter how I modify the `pdn.tcl`, there is `no via inserted between met4 and met5`, Could you know what's wrong with it and how to handle this problems?

maliberty, 2023-12-07:
If this is just for prototyping you can suppress the check IR-drop check in order to finish the flow:
```
export GND_NETS_VOLTAGES      =
export PWR_NETS_VOLTAGES      =
```

maliberty, 2023-12-08:
Its not a netlist problem - the power pins of the fakeram are the problem.