The-OpenROAD-Project/OpenROAD-flow-scripts #1695: No Instance (or Cell) Count in 6_report.log

dalyles, 2023-12-13:
### Description

In the 6_report.log report, we get some final summary results, one critical result that we do not see is the instance count of the circuit. Is is possible to include an instance count in the final results that also differentiates between the synthesized cells, the cells added by clock tree synthesis, and the number of filler cells. I'm also assuming that instance count means cell count in this context.

Thank you

maliberty, 2023-12-13:
We can add more detailed reporting.  Some cells are also added for timing optimization, padring generation, endcap/tapcells, antennas, and other categories.  How do you intend to use this data could give a better idea of what would be more useful.

dalyles, 2023-12-13:
I think for 6_report.log, some summary values would be very useful, such as cell count post synthesis (no cts or filler) and then the cell count for CTS alone, and the cell count for filler alone, and then the total cell count of the entire circuit.
This would be useful for our team because although we have an area metric, we also want a cell count metric to have a clearer idea about not only the size, but the complexity of the circuit. Also we want to compare the cell counts between the circuit without the clock tree so we can see which counts dominate, (this also applies for the filler cells)

I hope this makes sense, feel free to follow up if anything I said was unclear.
Thanks!

arthurjolo, 2024-01-25:
Hi @dalyles I have created a pull request that adds a command to report the cell count as you specified. Can you use the command added by that PR and see if that is what you wanted? If you want something different please let me know and I can change it. The PR link is [#4588](https://github.com/The-OpenROAD-Project/OpenROAD/pull/4588), and the command name is report_cell_count

dalyles, 2024-02-26:
Great, 

I think the information provided is great, but what I requested was to include it in 6_report.log  just like the rest of the important summary results are. Another thing I'd like to add, it seems like in the most recent version of ORFS, the timing and power information is no longer there and has been migrated to the reports directory.

I also browsed the reports in the reports directory and the log in the logs directory and I do not see any cell/instance count. Can we please include that in the reports?
Thanks

dalyles, 2024-02-26:
Hi @arthurjolo I just wanted to also say that the previous version of ORFS had timing and power information in <code>6_report.log</code> but it no longer displays it there and was migrated to <code>6_finish.rpt</code>. Was this intended?

Since those metrics and summary results are migrated to <code>6_finish.rpt</code> would it be appropriate to place the data there?

Thanks!

arthurjolo, 2024-02-26:
> Hi @arthurjolo I just wanted to also say that the previous version of ORFS had timing and power information in `6_report.log` but it no longer displays it there and was migrated to `6_finish.rpt`. Was this intended?
> 
> Since those metrics and summary results are migrated to `6_finish.rpt` would it be appropriate to place the cell count data there?
> 
> Thanks!

This was intended, I talked with Matt and he advised me to add the comand on the final_report.tcl that shows the report on the 6_report.log. @maliberty do you think the way it is is better or should we change it to be reported on the 6_report.rpt?

dalyles, 2024-02-27:
I think if that's the case, it should be no problem. I just want to make sure that I'm still able to find the relevant metrics that used to be in <code>6_report.log</code>

Thanks

dalyles, 2024-02-29:
Also one last tiny request, 
For the cell count can you make a final row where all the cells are totaled up?

Thanks

dalyles, 2024-03-01:
> Also one last tiny request, For the cell count can you make a final row where all the cells are totaled up?
> 
> Thanks

Just following up on this.

maliberty, 2024-03-01:
@arthurjolo this sounds simple

closed 2024-03-06 by maliberty