The-OpenROAD-Project/OpenROAD-flow-scripts #1701: ARM-64 support

LiaV211, 2023-12-14:
### Description

I am currently working on setting up the OpenROAD environment on Ubuntu 22.04 LTS hosted on AWS, utilizing a virtual server of the instance type t4g.xlarge with a storage volume of 128 GiB.

I have diligently followed the steps outlined in the provided guide available at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildLocally.md . However, I encountered several challenges during the setup process, particularly with CMake version discrepancies, file paths, and formats.

To address these issues, I made specific modifications to scripts, notably in DependencyInstaller.sh, and additionally manually cloned and built KLayout and OR-Tools. While the setup process indicates completion, new challenges have surfaced during the build phase, specifically regarding OR-Tools.
My changes in DependencyInstaller.sh
```
# Download CMake for ARM64
wget https://cmake.org/files/v3.21/cmake-3.21.2-linux-aarch64.sh

# Strip .note.ABI-tag from libQt5Core.so
strip --remove-section=.note.ABI-tag /usr/lib/aarch64-linux-gnu/libQt5Core.so
```

I would sincerely appreciate any guidance or assistance you could provide to help resolve these issues and ensure a successful setup. If there are any insights, recommendations, or alternative steps that might aid in overcoming these obstacles, I am eager to learn and implement them.

Thank you very much for your time and consideration. I look forward to your valuable assistance.

### Additional Context

These are the issues
Could not find a package configuration file provided by "ortools" with any
of the following names:
ortoolsConfig.cmake
ortools-config.cmake
Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
"ortools_DIR" to a directory containing one of the above files. If
"ortools" provides a separate development package or SDK, be sure it has
been installed.
After setting the path, I get similar issue
CMake Error at /home/ubuntu/OpenROAD-flow-scripts/dependencies/share/cmake-3.21/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
Could not find a package configuration file provided by "absl" with any of
the following names:
abslConfig.cmake
absl-config.cmake
Add the installation prefix of "absl" to CMAKE_PREFIX_PATH or set
"absl_DIR" to a directory containing one of the above files. If "absl"
provides a separate development package or SDK, be sure it has been
installed.
Call Stack (most recent call first):
/home/ubuntu/OpenROAD-flow-scripts/or-tools/build/ortoolsConfig.cmake:33 (find_dependency)
src/gpl/CMakeLists.txt:44 (find_package)
-- Configuring incomplete, errors occurred!

vijayank88, 2023-12-14:
@LiaV211 
You aim is to work on PnR flow using ORFS or contribute to tool development also?

liav-vorak, 2023-12-15:
Hello @vijayank88,

We would like to add ARM64 architecture support for the OpenROAD, which will make the project more cost effective in cloud based environments. We would like to contribute to the tool development, including adding CI for the ARM64 architecture. 
A while ago we were able to build the project for ARM64 and created AWS AMI for it, but with the recent changes in the tool we are faceing issues mentioned in the issue description. Our goal is to add CI for ARM64 and make the tool officially supported on ARM64 architecture.

Regards,
Lia

maliberty, 2023-12-15:
We don't build or test on arm-64 but would be glad to take a PR that helps with that platform.  Have you installed or-tools somewhere?  It is an external dependency of OR.

liav-vorak, 2023-12-18:
Hello @maliberty 
Yes I have installed and build or-tools manually

maliberty, 2023-12-18:
I thought or-tools included an absl installation.   Maybe you have to do a separate install? @vvbandeira any ideas?

vvbandeira, 2023-12-19:
@liav-vorak could you try to add `-DBUILD_DEPS=ON`? Are you following the [official docs](https://developers.google.com/optimization/install/cpp/source_linux#build_the_source_code)?

liav-vorak, 2023-12-20:
Thanks for the responses.
The build completed successfully.

vvbandeira, 2023-12-20:
@liav-vorak, that is great news. Please consider contributing our changes with a PR to improve our support for ARM64.

closed 2023-12-20 by vvbandeira