The-OpenROAD-Project/OpenROAD-flow-scripts #1718: Coyote_tc SoC example not connecting power pads to core power grid

nchiolino, 2023-12-21:
### Subject

[Stage]: Floorplaning.

### Describe the bug

When executing the floorplan for the coyote_tc out of the box example, it appears as though the core power and ground (u_vdd and u_vss) pads are not routing any traces to the core pdn ring. It appears as though the rest of the pdn.tcl file is being executed properly for things like strap widths and spacings except for -connect_to_pads. I am receiving no errors or warnings in the logs for the floor planning steps for pdn.

### Expected Behavior

I expect the VDD pads to route a trace to the VDD net of the pdn core ring, likewise with VSS pads and nets.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.

It is recommened to pull the latest changes.

If problem persists, file a github issue with the re-producible test case.

kernel: Linux 6.2.0-39-generic

os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)

cmake version 3.24.2

-- The CXX compiler identification is GNU 11.4.0

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- OpenROAD version: v2.0-10925-gb7c9b7e1c

-- System name: Linux

-- Compiler: GNU 11.4.0

-- Build type: RELEASE

-- Install prefix: /usr/local

-- C++ Standard: 17

-- C++ Standard Required: ON

-- C++ Extensions: OFF

-- The C compiler identification is GNU 11.4.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/cc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 235 lines cut ...]
-- Tcl Extended disabled

-- Python3 enabled

-- Configuring done

-- Generating done

-- Build files have been written to: /tmp/tmp.G8BwQUAN6m
```

### To Reproduce

Head to ./flow and run "make floorplan DESIGN_CONFIG=./designs/sky130hd/coyote_tc/config.mk".
Once completed, results can be viewed in the gui with make gui_floorplan. You will see that no u_vdd or u_vss pad instances are connected to the internal core power grid.

### Relevant log output

```shell
It appears that this bug is somewhat related to issue #1667 but differs enough to create a separate issue.
```

### Screenshots

[image]

maliberty, 2023-12-22:
As I commented in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1082
```
This was created as a work in progress to make a public example. Unfortunately it never got completed before the original author left the project. I've left it around on the hopes that someone would finish it up but it hasn't happened. At this point it might be better to just remove it.
```

nchiolino, 2024-04-15:
It was discovered that the east west routing metal was not a metal layer in the power connection portion of the pads. Therefore, it could not make a connection to the power and ground pads on the east and west sides of the chip. A temporary metal layer was added to the pads to prove the flow out and that seemed to work.

closed 2024-04-15 by nchiolino