The-OpenROAD-Project/OpenROAD-flow-scripts #1724: Under the new technology, an error occurs when running the code

chros098, 2023-12-26:
### Subject

[Stage]: Synthesis.

### Describe the bug

报错 
[image]

### Expected Behavior

[image]

### Environment

```shell
https://drive.google.com/file/d/1mjY2Riwhyp2hQwq8y62tn4a900jJKSKh/view?usp=drive_link 链接在这里
```

### To Reproduce

[image]

### Relevant log output

```shell
https://drive.google.com/file/d/1mjY2Riwhyp2hQwq8y62tn4a900jJKSKh/view?usp=drive_link
```

### Screenshots

[image]

vijayank88, 2023-12-26:
Is the new technology is opensource? @chros098

chros098, 2023-12-26:
Not open source

vijayank88, 2023-12-26:
> Not open source

Remove the drive link and share test case directly with @maliberty

chros098, 2023-12-26:
@maliberty  Hello, how can I share test cases with you?

chros098, 2023-12-26:
Hello, how can I share test cases with you? @maliberty

maliberty, 2023-12-26:
We can discuss support and and NDA via [info@precisioninno.com](mailto:info@precisioninno.com)

chros098, 2023-12-27:
done thanks

chros098, 2023-12-28:
https://drive.google.com/file/d/1baj54YCxDoi5kA84cavB4zPThX2QSL1c/view?usp=sharing thamks

maliberty, 2024-01-22:
The actual crash is in [abc](https://github.com/berkeley-abc/abc) not openroad.  You might have better luck with an issue there.

I did look at the crash and it is related to an instance of ADH1CSCONV2_12TL40.  This appears to be a full adder cell.  You can provide customer mapping for adders.  For example in ng45:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/0ea230930b57a6fb517521d8c34cb3591815b596/flow/platforms/nangate45/config.mk#L38

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/nangate45/cells_adders.v

Perhaps that will work around the issue.

closed 2024-03-06 by maliberty