The-OpenROAD-Project/OpenROAD-flow-scripts #1725: make command fails for nangate45/gcd

vijayank88, 2023-12-26:
### Subject

[Stage]: CTS.

### Describe the bug

With latest ORFS build facing error during cts stage with Ubuntu20.04 machine.
```
OpenROAD v2.0-10182-gd8f34b68e 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -sink_clustering_enable -balance_levels
[ERROR CTS-0055] Missing argument -buf_list
Error: cts.tcl, 41 CTS-0055
Command exited with non-zero status 1
```

### Expected Behavior

Complete make command

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-91-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.25.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11432-gead3236e5
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.5 (found version "3.5.0") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
[... 107 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.l1bNPEy1nw
```

### To Reproduce

Build latest ORFS
```
source env.sh
cd flow
make
```

### Relevant log output

```shell
OpenROAD v2.0-10182-gd8f34b68e 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -sink_clustering_enable -balance_levels
[ERROR CTS-0055] Missing argument -buf_list
Error: cts.tcl, 41 CTS-0055
Command exited with non-zero status 1
```

maliberty, 2023-12-26:
In 'Environment'
```
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
```
Please try with the latest OR.

vijayank88, 2023-12-27:
With latest OR head, problem resolved.

closed 2023-12-27 by vijayank88