The-OpenROAD-Project/OpenROAD-flow-scripts #1733: Fail in synthesis stage with hierarchical designs of ASAP7 PDK

dearmrlv, 2023-12-30:
### Subject

[Documentation] for documentation errors.

### Describe the bug

Get failures when running design riscv32i and swerv_wrapper with ASAP7 PDK. And flow is OK for designs without hierarchy (e.g. ibex). 
For riscv32i,  the error is as blow:
```
...
32.23.19. Extracting gate netlist of module `\maindec' to `<abc-temp-dir>/input.blif'..                                                                                         [0/1872]
32.23.20. Extracting gate netlist of module `\regfile' to `<abc-temp-dir>/input.blif'..
32.23.21. Extracting gate netlist of module `\riscv' to `<abc-temp-dir>/input.blif'..
32.23.22. Extracting gate netlist of module `\riscv_top' to `<abc-temp-dir>/input.blif'..
32.23.23. Extracting gate netlist of module `\shifter' to `<abc-temp-dir>/input.blif'..
32.23.24. Extracting gate netlist of module `\signext' to `<abc-temp-dir>/input.blif'..
32.24. Executing OPT pass (performing simple optimizations).
32.24.1. Executing OPT_EXPR pass (perform const folding).
32.23.24. Extracting gate netlist of module `\signext' to `<abc-temp-dir>/input.blif'..
32.24. Executing OPT pass (performing simple optimizations).
32.24.1. Executing OPT_EXPR pass (perform const folding).
32.24.2. Executing OPT_MERGE pass (detect identical cells).
32.24.3. Executing OPT_DFF pass (perform DFF optimizations).
32.24.4. Executing OPT_CLEAN pass (remove unused cells and wires).
32.24.5. Finished fast OPT passes.
32.25. Executing HIERARCHY pass (managing design hierarchy).
32.25.1. Analyzing design hierarchy..
32.25.2. Analyzing design hierarchy..
32.26. Printing statistics.
32.27. Executing CHECK pass (checking for obvious problems).
33. Executing TECHMAP pass (map to technology primitives).
33.1. Executing Verilog-2005 frontend: /cyberpi/OpenROAD-flow-scripts/flow/platforms/asap7/yoSys/cells_adders_R.v
33.2. Continuing TECHMAP pass.
34. Executing TECHMAP pass (map to technology primitives).
34.1. Executing Verilog-2005 frontend: /cyberpi/yosys/install/usr/local/bin/../share/yosys/techmap.v
34.2. Continuing TECHMAP pass.
35. Executing DFFLIBMAP pass (mapping DFF cells to sequential cells from liberty file).
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFLx1_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFLx2_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFLx3_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFLx4_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFHx1_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFHx2_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFHx3_ASAP7_75t_R' - skipping.
Warning: Found unsupported expression 'D*!SE+!D*!SI+SE*!SI' in pin attribute of cell 'SDFHx4_ASAP7_75t_R' - skipping.
35.1. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
36. Executing ABC pass (technology mapping using ABC).
36.1. Extracting gate netlist of module `$paramod\flopr\WIDTH=s32'00000000000000000000000000100000' to `<abc-temp-dir>/input.blif'..
36.1.1. Executing ABC.
ERROR: Can't open ABC output file `/tmp/yosys-abc-rshShO/output.blif'.
```
The `/tmp` directory on my machine is mounted and with enough space:
```
$ df -h /tmp
Filesystem      Size  Used Avail Use% Mounted on
overlay          96G   22G   69G  25% /
```

Besides, I modified the `config.mk` according to https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/202#issuecomment-953195443. Otherwise yosys will end with `Segment Fault`.
Here is my `riscvi/config.mk`:
```
export DESIGN_NICKNAME = riscv32i
export DESIGN_NAME = riscv_top
export PLATFORM    = asap7

export DFF_LIB_FILE = $($(CORNER)_DFF_LIB_FILE)

export SYNTH_HIERARCHICAL = 1
export RTLMP_FLOW = True

export RTLMP_MIN_INST = 1000
export RTLMP_MAX_INST = 3500
export RTLMP_MIN_MACRO = 1
export RTLMP_MAX_MACRO = 5

export MAX_UNGROUP_SIZE ?= 1000

export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc
export ADDITIONAL_LEFS = ./platforms/$(PLATFORM)/lef/fakeram7_256x32.lef
export ADDITIONAL_LIBS = ./platforms/$(PLATFORM)/lib/fakeram7_256x32.lib

export DIE_AREA = 0 0 80 90
export CORE_AREA = 5 5 75 85

export PLACE_DENSITY_LB_ADDON = 0.10

export HAS_IO_CONSTRAINTS = 1
export PLACE_PINS_ARGS    = -exclude left:* -exclude right:* -exclude top:*
export MACRO_PLACE_HALO    = 3 3
export MACRO_PLACE_CHANNEL = 6 6
#
export TNS_END_PERCENT   = 100
```

### Expected Behavior

The flow runs without such error.

### Environment

```shell
I install OpenROAD and yosys manually, and map the install directory to OpenROAD-flow-scripts directory `./tools/install` by `ln -s` command.
```

### To Reproduce

As showed.

mi-and-core, 2024-01-02:
Cannot be reproduced with:
yosys 0.36+40 (build from source code – ABC UC Berkeley 1.01) and
openroad (prebuilt binary, Dec 18, 23)

and also cannot be reproduced with:
yosys 0.33 and openroad v2.0-11613 ( both build with openroad_build.sh )

What does yosys version say?
What does yosys-abc say?

By the way: the extra “export DFF_LIB_FILE = $($(CORNER)_DFF_LIB_FILE)” in config.mk
is not needed there anymore.

dearmrlv, 2024-01-02:
Thanks for replying!

`yosys -version`:
```
Yosys 0.33 (git sha1 2584903a0, clang 14.0.0-1ubuntu1.1 -fPIC -Os)
```
`yosys-abc`:
```
UC Berkeley, ABC 1.01 (compiled Jan  1 2024 06:48:35)
abc 01>
```

mi-and-core, 2024-01-02:
A quick look, i would expect in your log something like this:

```
<deleted>

35.1. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
abc -script /cyberpi/OpenROAD-flow-scripts/flow/scripts/abc_speed.script -liberty ./objects/asap7/riscv32i/base/lib/merged.lib -constr ./objects/asap7/riscv32i/base/abc.constr -D 1660
36. Executing ABC pass (technology mapping using ABC).

<deleted>
```

So your never reached line 66 in scripts/synth.tcl, before abc will be called.

(a) does the temporary file /tmp/... exists ?
(b) write permission ?

Alternatively: 
hide systemwide your pre-build openroad and pre-build yosys, try a fresh apt clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts.git and a fresh local build. Use the unmodified config.mk

maliberty, 2024-01-02:
This might be better filed with the yosys team.  Either you have a filesystem problem or an abc crash.

closed 2024-03-06 by maliberty