The-OpenROAD-Project/OpenROAD-flow-scripts #1741: About Autotunner

jack25946705, 2024-01-05:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I want to use autotunner.But it seems that it doesn't work.
But I can use OpenROAD-flow-scripts using docker image I built.
I think autotunner must also be run under that image or other image??
I modify the autotunner dokerfile on the github.I can successfully build,but get some error.
Attached
[error.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13836906/error.txt)
 is the error message.

### Expected Behavior

I think I should tune the parameter correctly. But using the image I built by myself,I cannot tune the parameter.

### Environment

```shell
FROM rayproject/ray:1.11.0

RUN pip install -U ax-platform hyperopt nevergrad optuna pandas

USER root

RUN wget -qO - https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add -

RUN echo "deb http://apt.kubernetes.io kubernetes-xenial main" > /etc/apt/sources.list.d/kubernetes.list

RUN sudo apt-get update -y \
    && sudo apt-get install -y vim htop gcc time libffi-dev

RUN wget https://www.klayout.org/downloads/Ubuntu-20/klayout_0.27.13-1_amd64.deb

RUN sudo apt-get install -y --fix-broken ./klayout_0.27.13-1_amd64.deb \
    && rm ./klayout_0.27.13-1_amd64.deb

RUN sudo apt-get install -y curl

RUN curl -O https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/etc/DependencyInstaller.sh
RUN sudo bash ./DependencyInstaller.sh -dev
RUN rm DependencyInstaller.sh

USER $USER

The above is my modified dockerfile.
In the ~/OpenROAD-flow-scripts/tools/Autotunner,I execute "docker build -t autotunner ." to get my dockerfile
Then I use the following instruction in OpenROAD-flow-scripts to run the docker
docker run --rm -it  -u $(id -u ${USER}):$(id -g ${USER})  -v $(pwd)/flow:/OpenROAD-flow-scripts/flow            -e DISPLAY=${DISPLAY}  -v /tmp/.X11-unix:/tmp/.X11-unix -v ${HOME}/.Xauthority:/.Xauthority            --network host  --security-opt seccomp=unconfined  autotunner
But I get the above error message.
```

### To Reproduce

I don't know what to upload....

vijayank88, 2024-01-30:
@vvbandeira 
Is that autotuner can be installed inside docker environment?
```
pip3 install -U --user 'ray[default,tune]==1.11.0' ax-platform hyperopt nevergrad optuna pandas
pip3 install -U --user colorama==0.4.4 bayesian-optimization==1.4.0
```

I tried running inside docker and got following error
```
bash-4.2$ pip3 install -U --user colorama==0.4.4 bayesian-optimization==1.4.0
WARNING: The directory '/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: The directory '/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting colorama==0.4.4
  Downloading https://files.pythonhosted.org/packages/44/98/5b86278fbbf250d239ae0ecb724f8572af1c91f4a11edf4d36a206189440/colorama-0.4.4-py2.py3-none-any.whl
Collecting bayesian-optimization==1.4.0
  Downloading https://files.pythonhosted.org/packages/2c/af/478deae10abb1cc1034993064b41dbc6015734b4f8cafa2c3d1a05861f75/bayesian_optimization-1.4.0-py3-none-any.whl
Requirement already satisfied, skipping upgrade: numpy>=1.9.0 in /opt/rh/rh-python38/root/usr/local/lib64/python3.8/site-packages (from bayesian-optimization==1.4.0) (1.24.4)
Collecting scipy>=1.0.0
  Downloading https://files.pythonhosted.org/packages/69/f0/fb07a9548e48b687b8bf2fa81d71aba9cfc548d365046ca1c791e24db99d/scipy-1.10.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (34.5MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 34.5MB 9.6MB/s 
Collecting scikit-learn>=0.18.0
  Downloading https://files.pythonhosted.org/packages/3f/48/6fdd99f5717045f9984616b5c2ec683d6286d30c0ac234563062132b83ab/scikit_learn-1.3.2-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.1MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 11.1MB 40.6MB/s 
Collecting threadpoolctl>=2.0.0
  Downloading https://files.pythonhosted.org/packages/81/12/fd4dea011af9d69e1cad05c75f3f7202cdcbeac9b712eea58ca779a72865/threadpoolctl-3.2.0-py3-none-any.whl
Collecting joblib>=1.1.1
  Downloading https://files.pythonhosted.org/packages/10/40/d551139c85db202f1f384ba8bcf96aca2f329440a844f924c8a0040b6d02/joblib-1.3.2-py3-none-any.whl (302kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 307kB 63.4MB/s 
Installing collected packages: colorama, scipy, threadpoolctl, joblib, scikit-learn, bayesian-optimization
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/.local'
Check the permissions.
```

vvbandeira, 2024-02-01:
> `The directory '/.cache/pip/http'`

@jack25946705 
Ray `1.11.0` is quite old, the latest version is `2.9.1`. We don't have active users for AutoTuner so the dependencies version are outdated, which looks like the cause of the error messages. You could try with `2.9.1`, hopefully there were not many breaking API changes between the versions.

@vijayank88
Looks like you are either trying to run `pip3` with `--user` as the root user or the user your are using does not have a `$HOME` folder. Both cases are not encouraged. Either use `pip3` withtout `--user` if you are the root user or create the user inside the docker container/image.

vijayank88, 2024-02-02:
Also inside docker, with default centos7 using `python3.6.8` it is not supported by AutoTuner. 
So better to install `python3.10` as default python version is useful.

vijayank88, 2024-02-02:
@vvbandeira 
Even without `--user ` it is end up with same error
```
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/.local'
Check the permissions.
```

luarss, 2024-03-15:
In case anyone stumbles on this in the future:

Guide for setting up Autotuner locally: https://openroad-flow-scripts.readthedocs.io/en/latest/user/InstructionsForAutoTuner.html
Colab interactive notebook: https://colab.research.google.com/drive/1wye0osn34YVWPvTrfBTftjOfGOtF3ABe

closed 2024-04-01 by vvbandeira