The-OpenROAD-Project/OpenROAD-flow-scripts #1747: No access point for pins which are put inside the die

b224hisl, 2024-01-10:
### Subject

[Stage]: Detail Router.

### Describe the bug

I have put some of the pins on the top layer inside the core region,  like:
[image]
But there seems no access points for these pins. I wonder how to route through these pins.

### Expected Behavior

finish routing successfully

### Environment

```shell
kernel: Linux 3.10.0-1160.76.1.el7.x86_64
os: Ubuntu 20.04.6 LTS (Focal Fossa)
```

### To Reproduce

This file is a little big, so I upload to google drive
https://drive.google.com/file/d/1xvWOmVQyiXQpLw5eb84rhsbmHzanPte6/view?usp=sharing

### Relevant log output

```shell
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 100 pins.
[INFO DRT-0076]   Complete 200 pins.
[INFO DRT-0076]   Complete 300 pins.
[INFO DRT-0076]   Complete 400 pins.
[INFO DRT-0076]   Complete 500 pins.
[INFO DRT-0076]   Complete 600 pins.
[INFO DRT-0076]   Complete 700 pins.
[INFO DRT-0076]   Complete 800 pins.
[INFO DRT-0076]   Complete 900 pins.
[INFO DRT-0077]   Complete 1000 pins.
[ERROR DRT-0074] No access point for PIN/pc[13].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[6].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[5].
[ERROR DRT-0074] No access point for PIN/pc[17].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[0].
[ERROR DRT-0074] No access point for PIN/pc[19].
[ERROR DRT-0074] No access point for PIN/dataadr[0].
[ERROR DRT-0074] No access point for PIN/dataadr[4].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[1].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[1].
[ERROR DRT-0074] No access point for PIN/dataadr[2].
[ERROR DRT-0074] No access point for PIN/ce_mem[1].
[ERROR DRT-0074] No access point for PIN/pc[10].
[ERROR DRT-0074] No access point for PIN/dataadr[6].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[2].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[7].
[ERROR DRT-0074] No access point for PIN/dataadr[7].
[ERROR DRT-0074] No access point for PIN/pc[18].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[7].
[ERROR DRT-0074] No access point for PIN/pc[14].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[0].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[3].
[ERROR DRT-0074] No access point for PIN/memwrite.
[ERROR DRT-0074] No access point for PIN/dataadr[5].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[0].
[ERROR DRT-0074] No access point for PIN/pc[12].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[7].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[3].
[ERROR DRT-0074] No access point for PIN/pc[15].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[3].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[0].
[ERROR DRT-0074] No access point for PIN/pc[11].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[6].
[ERROR DRT-0074] No access point for PIN/pc[8].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[7].
[ERROR DRT-0074] No access point for PIN/dataadr[1].
[ERROR DRT-0074] No access point for PIN/ce_mem[3].
[ERROR DRT-0074] No access point for PIN/we_mem[3].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[4].
[ERROR DRT-0074] No access point for PIN/reset.
[ERROR DRT-0074] No access point for PIN/pc[9].
[ERROR DRT-0074] No access point for PIN/pc[20].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[4].
[ERROR DRT-0074] No access point for PIN/writedata[0].
[ERROR DRT-0074] No access point for PIN/pc[21].
[ERROR DRT-0074] No access point for PIN/writedata[5].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[5].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[6].
[ERROR DRT-0074] No access point for PIN/writedata[1].
[ERROR DRT-0074] No access point for PIN/we_mem[1].
[ERROR DRT-0074] No access point for PIN/pc[16].
[ERROR DRT-0074] No access point for PIN/writedata[4].
[ERROR DRT-0074] No access point for PIN/writedata[7].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[1].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[2].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[1].
[ERROR DRT-0074] No access point for PIN/pc[0].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[5].
[ERROR DRT-0074] No access point for PIN/inter_dmem2[4].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[4].
[ERROR DRT-0074] No access point for PIN/pc[2].
[ERROR DRT-0074] No access point for PIN/ce_mem[0].
[ERROR DRT-0074] No access point for PIN/pc[1].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[5].
[ERROR DRT-0074] No access point for PIN/pc[5].
[ERROR DRT-0074] No access point for PIN/pc[3].
[ERROR DRT-0074] No access point for PIN/pc[7].
[ERROR DRT-0074] No access point for PIN/ce_mem[2].
[ERROR DRT-0074] No access point for PIN/pc[4].
[ERROR DRT-0074] No access point for PIN/dataadr[3].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[2].
[ERROR DRT-0074] No access point for PIN/we_mem[0].
[ERROR DRT-0074] No access point for PIN/we_mem[2].
[ERROR DRT-0074] No access point for PIN/inter_dmem3[3].
[ERROR DRT-0074] No access point for PIN/pc[22].
[ERROR DRT-0074] No access point for PIN/writedata[3].
[ERROR DRT-0074] No access point for PIN/pc[6].
[ERROR DRT-0074] No access point for PIN/writedata[2].
[ERROR DRT-0074] No access point for PIN/writedata[6].
[ERROR DRT-0074] No access point for PIN/pc[23].
[ERROR DRT-0074] No access point for PIN/inter_dmem0[6].
[ERROR DRT-0074] No access point for PIN/inter_dmem1[2].
[ERROR DRT-0074] No access point for PIN/clk.
Error: detail_route.tcl, 70 DRT-0074
Command exited with non-zero status 1
Elapsed time: 0:06.53[h:]min:sec. CPU time: user 67.82 sys 0.95 (1053%). Peak memory: 501564KB.
make[1]: *** [Makefile:745: do-5_3_route] Error 1
make: *** [Makefile:745: results/sky130hd/riscv32i_3d/sky130_gf180/5_3_route.odb] Error 2
```

vijayank88, 2024-01-10:
Can you share io constraints used for pin placement?
Also make test case for pin placement stage and attach to debug. 
IO's should be placed on core boundary only, not inside core area.

b224hisl, 2024-01-10:
> Can you share io constraints used for pin placement? Also make test case for pin placement stage and attach to debug. IO's should be placed on core boundary only, not inside core area.

here is my io.tcl:
```
# Helper function to split a string into a list of strings and numbers
proc split_strings_and_numbers {str} {
    set result {}
    foreach {all letters numbers} [regexp -all -inline {(\D*)(\d*)} $str] {
        if {$letters ne ""} {
            lappend result $letters
        }
        if {$numbers ne ""} {
            lappend result [expr {$numbers + 0}]  ;# Convert to integer
        }
    }
    return $result
}

# Custom comparison function
proc natural_compare {str1 str2} {
    set list1 [split_strings_and_numbers $str1]
    set list2 [split_strings_and_numbers $str2]
    set len [expr {min([llength $list1], [llength $list2])}]
    for {set i 0} {$i < $len} {incr i} {
        set part1 [lindex $list1 $i]
        set part2 [lindex $list2 $i]
        if {$part1 ne $part2} {
            if {[string is integer -strict $part1] && [string is integer -strict $part2]} {
                return [expr {$part1 - $part2}]
            } else {
                return [string compare $part1 $part2]
            }
        }
    }
    return [expr {[llength $list1] - [llength $list2]}]  ;# If all parts are equal, compare by length
}

proc natural_sort {list} {
    return [lsort -command natural_compare $list]
}

proc match_pins { regex } {
    set pins {}
    # The regex for get_ports is not the tcl regex
    foreach pin [get_ports -regex .*] {
        set input [get_property $pin name]
        # We want the Tcl regex
        if {![regexp $regex $input]} {
            continue
        }
        lappend pins [get_property $pin name]
    }
    return [natural_sort $pins]
}

############################################################################################
set writedata [match_pins writedata*]
set writedata_list [concat {*}$writedata]

set dataadr [match_pins dataadr*]
set dataadr_list [concat {*}$dataadr]

set mem [match_pins _mem*]
set mem_list [concat {*}$mem]

set inter_dmem [match_pins inter_dmem*]
set inter_dmem_list [concat {*}$inter_dmem]

define_pin_shape_pattern -layer met5 -region {20 20 1180 980} -size {0.5 0.5} -x_step 10 -y_step 10

set_io_pin_constraint -region "up:{20 20 1180 40}" -pin_names $writedata_list
set_io_pin_constraint -region "up:{20 20 40 980}" -pin_names $mem_list
set_io_pin_constraint -region "up:{20 960 1180 980}" -pin_names $dataadr_list
set_io_pin_constraint -region "up:{1100 20 1180 980}" -pin_names $inter_dmem_list

place_pins -hor_layers met5 -ver_layers met4
```

Here's one of your unit tests that performs exactly what I'm trying to do: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/ppl/test/top_layer7.tcl. So I think the pins can also be placed inside the core area.

b224hisl, 2024-01-10:
I also found that in the error log,  [ERROR DRT-0074] does not occur for every pin, which means that some pins must have access points.

b224hisl, 2024-01-11:
I think I have found what's wrong with that😄. 
The size of the pins must be set according to the pdk itself. 
Btw, how can I know the size of the pin of each metal layer? For me, I just place the pins automatically to a specific layer then I measure the sizes according to the def file. But I think there must be another way to see the reasonable range of pin size.
Anyway, thank you~ now you can close the issue.

rovinski, 2024-01-11:
I think generally pins have to satisfy minimum width and minimum area rules. They also have to be on the manufacturing grid (but preferably on the routing grid). If you have pins with DRC violations, then any connections to them will likely still have violations. That is likely why DRT cannot find an access point.

b224hisl, 2024-01-11:
Yes, after I asjust the pin size, every thing goes well.  
but where can I find the minimum width and minimum area rules?

rovinski, 2024-01-11:
They are in the technology LEF file.

osamahammad21, 2024-01-15:
Should this issue be closed?

vvbandeira, 2024-01-16:
Closing as the OP looks to have found the solution. Please re-open if that is not the case.

closed 2024-01-16 by vvbandeira