The-OpenROAD-Project/OpenROAD-flow-scripts #1756: INFO DPL-0034] Detailed placement failed on the following 1421121 instances

chros098, 2024-01-16:
### Subject

[Documentation] for documentation errors.

### Describe the bug

 displayed on log：INFO DPL-0034] Detailed placement failed on the following 1421121 instances。

### Expected Behavior

Complete the correct process

### Environment

```shell
I am using openroad script etc. There is no such shell under etc. The version has just been updated.
```

### To Reproduce

https://drive.google.com/file/d/1ZO4Ob-cm1gejT5ayU-n0xruf4naSJg3s/view?usp=drive_link

vvbandeira, 2024-01-16:
@chros098, this does not look like an issue with the documentation.
Further, after the error message, there should be a list of the instances that could not be placed correctly. You can use the GUI to take a look at them and also attach a print of them here.
@vijayank88 any other suggestions?

vijayank88, 2024-01-17:
@chros098 
[image]
Seems there was issue with your placement density value. Can you share your config.mk and constraint.sdc
Also your clock port not defined properly.

vijayank88, 2024-01-17:
also you're using OpenROAD commit from `on Jul 2, 2023`. Please pull the latest, run and share test case again.

chros098, 2024-01-18:
We have modified FA_X1, HA_X1, DLL_X1, and used a relatively new openroad version, but there is still a long pause in 3_5. Please help solve this problem. The design config has been uploaded to the test example. Due to some special reasons, we can't connect to git now, so we can't update the latest version. Thank you. 
           https://drive.google.com/file/d/191eZEMdLPRdBbUOu_GwO8YKx9D50ozS5/view?usp=sharing

maliberty, 2024-01-22:
Is it intentional that you have no timing constraints?
You also have a vast amount of empty space coming into dpl:
[image]

Is that intentional?  You get a large amount of buffering:

```
[INFO RSZ-0038] Inserted 2221966 buffers in 284671 nets.
[INFO RSZ-0039] Resized 280945 instances.
```
which greatly increases the placement density locally.

maliberty, 2024-01-22:
With placement densities above 1500% the legalizer will have to spend a lot of time trying to resolve the congestion.  I think it is better to deal with the upstream issues.

[image]

chros098, 2024-01-25:
It was not intentional. We began to think that if no constraints were added, the process would run smoothly. However, in the past few days, we have tried adding constraints and still reported the same problem. We are not sure why so many buffers were added in steps 3_4 and 3_3. It seems to be normal, and this problem does not occur when using other process platforms. Can you help analyze the reason? Thank you.

maliberty, 2024-01-25:
I noticed
```
[INFO RSZ-0058] Using max wire length 20um.
[INFO RSZ-0034] Found 277255 slew violations.
[INFO RSZ-0036] Found 137243 capacitance violations.
[INFO RSZ-0037] Found 65479 long wires.
[INFO RSZ-0038] Inserted 2221966 buffers in 284671 nets.
[INFO RSZ-0039] Resized 280945 instances.
```

the 20um length seems quite short.  However I then discovered that though this appears to be an ng45 test case you have actually replaced the guts of the platform files with SMIC data:

```
/*   No part of this information may be reproduced, transmitted, transcribed,      */
/* stored in a retrieval system, or translated into any human or computer          */
/* language, in any form or by any means, electronic, mechanical, magnetic,        */
/* optical, chemical, manual, or otherwise, without the prior written consent of   */
/* SMIC. Any unauthorized use or disclosure of this material is strictly           */
/* prohibited and may be unlawful. By accepting this material, the receiving       */
/* party shall be deemed to have acknowledged, accepted, and agreed to be bound    */
/* by the foregoing limitations and restrictions. Thank you.                       */
/*                                 
```
This appears to be a violation of the license.  I will not look into this any further as we have no NDA in place.

closed 2024-01-25 by maliberty