The-OpenROAD-Project/OpenROAD-flow-scripts #1759: Boost Serialization - "No suitable build variant" during ./build_openroad.sh

dansharkey, 2024-01-16:
### Subject

[Build] for compilation-related issues.

### Describe the bug

During ./build_openroad.sh I receive an error 

```
CMake Error at /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/boost_serialization-1.80.0/boost_serialization-config.cmake

  but it set boost_serialization_FOUND to FALSE so package
  "boost_serialization" is considered to be NOT FOUND.  Reason given by
  package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_serialization.so.1.80.0 (shared, Boost_USE_STATIC_LIBS=ON)

```

This is on a fresh Ubuntu Matic (23.10) install, trying to install OpenROAD locally. I've run `setup.sh` and the dependency installer script before trying to run the ./build_openroad.sh script.

Similar issue: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/472
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/13950830/build_openroad.log)

### Expected Behavior

OpenROAD should successfully build.

### Environment

```shell
Git commit: 4464b1c843e89625e3eec0e5a22ce42162e157b0
kernel: Linux 6.5.0-14-generic
os: Ubuntu 23.10 (Mantic Minotaur)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11862-g4464b1c84
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.11 (found version "3.11.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
[... 161 lines cut ...]
-- GUI is enabled
-- Charts widget is not enabled
CMake Error at /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/boost_serialization-1.80.0/boost_serialization-config.cmake

  but it set boost_serialization_FOUND to FALSE so package
  "boost_serialization" is considered to be NOT FOUND.  Reason given by
  package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_serialization.so.1.80.0 (shared, Boost_USE_STATIC_LIBS=ON)

Call Stack (most recent call first):
  /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/local/share/cmake-3.24/Modules/FindBoost.cmake:594 (find_package)
  src/drt/CMakeLists.txt:49 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
```

vvbandeira, 2024-01-16:
> This is on a fresh Ubuntu Matic (23.10) install, trying to install OpenROAD locally. I've run setup.sh and the dependency installer script before trying to run the ./build_openroad.sh script.

Could you share exactly the commands you have run for the `setup.sh` and dependency installer in the order that you ran?

dansharkey, 2024-01-16:
Thanks for your support! Please see below an extract of my bash history. As you can see, I initially tried ./build_openroad.sh, which failed due to missing ortools. I tried installing the missing dependency through the package manager, before discovering the documentation(!) and attempting to run setup.sh and the dependency installer. This is probably all out of order vs a "proper" install. Do you think that's likely to be responsible for the issue?

```
  153  cd OpenROAD-flow-scripts/
  154  ./build_openroad.sh --local
  155  sudo apt install ortools
  156  ./build_openroad.sh --local
  157  sudo ./setup.sh 
  158  ./build_openroad.sh --local
  159  less /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log
  160  ./etc/DependencyInstaller.sh
  161  cd Work/vlsi/tools/OpenROAD-flow-scripts/
  162  ./etc/DependencyInstaller.sh
  163  sudo ./etc/DependencyInstaller.sh
  164  ./build_openroad.sh --local
  165  exit
  166  ls
  167  cd Work/vlsi/tools/OpenROAD-flow-scripts/
  168  ./build_openroad.sh 
  169  sudo apt install build-essential cmake libboost-dev libboost-serialization-dev libboost-filesystem-dev libboost-iostreams-dev libboost-program-options-dev zlib1g-dev rapidjson-dev libquadmath0
  170  ./build_openroad.sh 
  171  sudo apt install build-essential cmake libboost-dev libboost-serialization-dev libboost-filesystem-dev libboost-iostreams-dev libboost-program-options-dev zlib1g-dev rapidjson-dev libquadmath0
  172  ./build_openroad.sh 
```

dansharkey, 2024-01-19:
Hello @vvbandeira - I've removed (rm -rf) and re-cloned the OpenROAD-flow-scripts repo, re-run `setup.sh` and `./build_openroad.sh --local` and it's now completed successfully. I did get quite a lot of segfaults during the build - I suspect there's a hardware issue on my end - I'm using quite an old hard disk drive and suspect it's got some issues, though I'm not 100% certain that's the cause. I repeatedly called `./build_openroad.sh` until it completed successfully. In any case, I would suggest the bug can be closed for now if you're happy with that?

rovinski, 2024-01-19:
@dansharkey is there a chance that your system has a small amount of RAM? The build script builds using all available CPU cores, but if there is not enough RAM in the system then it can cause an out-of-memory failure. A way to check would be to try building with a small number of threads such as with `./build_openroad.sh --threads 2`.

dansharkey, 2024-01-20:
It's possible, although I'm on a 32GB system. It's running a Ryzen 5 5600X, which has an effective 12 cores, so may e that overloaded it? I'd have to check memory usage during compilation, but I'm away at the moment so won't be able to do that until next week I'm afraid.

rovinski, 2024-01-20:
That's probably fine. I've mostly seen it on laptops that have a high number of efficiency cores so the logical core count is high (>20) but the RAM is only like 16 GB. But you could try it anyways.

closed 2024-03-13 by maliberty