The-OpenROAD-Project/OpenROAD-flow-scripts #1762: Build Locally Error

Tyrese-FullStackGenius, 2024-01-16:
### Describe the bug

```
ubuntu@ubuntu:~/OpenROAD-flow-scripts/flow$ make
(trap 'mv ./logs/nangate45/gcd/base/4_1_cts.tmp.log ./logs/nangate45/gcd/base/4_1_cts.log' EXIT; /usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/ubuntu/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  /home/ubuntu/OpenROAD-flow-scripts/flow/scripts/cts.tcl -metrics ./logs/nangate45/gcd/base/4_1_cts.json) 2>&1 | tee ./logs/nangate45/gcd/base/4_1_cts.tmp.log
OpenROAD v2.0-11740-g3b47e7b77
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -sink_clustering_enable -balance_levels
[INFO CTS-0050] Root buffer is CLKBUF_X3.
[INFO CTS-0051] Sink buffer is CLKBUF_X3.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    CLKBUF_X3
[INFO CTS-0049] Characterization buffer is CLKBUF_X3.
[INFO CTS-0007] Net "clk" found for clock "core_clock".
[INFO CTS-0010]  Clock net "clk" has 35 sinks.
[INFO CTS-0008] TritonCTS found 1 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 0 placement blockages have been identified.
[INFO CTS-0201] 0 placed hard macros will be treated like blockages.
[INFO CTS-0027] Generating H-Tree topology for net clk.
[INFO CTS-0028]  Total number of sinks: 35.
[INFO CTS-0090]  Sinks will be clustered based on buffer max cap.
[INFO CTS-0030]  Number of static layers: 0.
[INFO CTS-0020]  Wire segment unit: 14000  dbu (7 um).
[INFO CTS-0023]  Original sink region: [(7410, 9660), (64040, 63170)].
[INFO CTS-0024]  Normalized sink region: [(0.529286, 0.69), (4.57429, 4.51214)].
[INFO CTS-0025]     Width:  4.0450.
[INFO CTS-0026]     Height: 3.8221.
 Level 1
    Direction: Horizontal
    Sinks per sub-region: 18
    Sub-region size: 2.0225 X 3.8221
[INFO CTS-0034]     Segment length (rounded): 1.
 Level 2
    Direction: Vertical
    Sinks per sub-region: 9
    Sub-region size: 2.0225 X 1.9111
[INFO CTS-0034]     Segment length (rounded): 1.
[INFO CTS-0032]  Stop criterion found. Max number of sinks is 15.
[INFO CTS-0035]  Number of sinks covered: 35.
[INFO CTS-0018]     Created 5 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 5 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 7:2, 10:1, 11:1..
[INFO CTS-0017]     Max level of the clock tree: 2.
[INFO CTS-0098] Clock net "clk"
[INFO CTS-0099]  Sinks 35
[INFO CTS-0100]  Leaf buffers 0
[INFO CTS-0101]  Average sink wire length 51.96 um
[INFO CTS-0102]  Path depth 2 - 2

==========================================================================
cts pre-repair report_design_area
--------------------------------------------------------------------------
Design area 555 u^2 53% utilization.
[INFO RSZ-0058] Using max wire length 661um.

==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 555 u^2 53% utilization.
Placement Analysis
---------------------------------
total displacement          6.0 u
average displacement        0.0 u
max displacement            1.7 u
original HPWL            1878.1 u
legalized HPWL           1947.2 u
[... 129 lines cut ...]
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Configuring incomplete, errors occurred!
```

### To Reproduce

I don't how I do,

vvbandeira, 2024-01-16:
@Tyrese-FullStackGenius, looks like you are not using the latest code:

> [WARNING] Your current OpenROAD version is outdated.

Please try again with the latest code and do not forget to source `env.sh` if you used `sudo ./setup.sh` to install the dependencies.

Tyrese-FullStackGenius, 2024-01-16:
I'm using latest version
***@***.***:~/OpenROAD-flow-scripts$ openroad -version
v2.0-11740-g3b47e7b77

and I'm following
https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

vvbandeira, 2024-01-16:
You need the latest version of ORFS not OR. Note also that at the verify step [here](https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html#verify-installation) you are required to `source ./env.sh` before running `make` to run the flow.

Tyrese-FullStackGenius, 2024-01-16:
I downloaded using git clone --recursive
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts, it's false?

vvbandeira, 2024-01-16:
> I downloaded using git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts, it's false?

That is correct, but I have merged a fix for this issue today. So you will need to update your local copy or download it again, to update:

```
cd OpenROAD-flow-scripts
git pull
```

Tyrese-FullStackGenius, 2024-01-16:
Thanks, I'll check it out.

vvbandeira, 2024-01-16:
These should work if you have already run `sudo ./setup.sh` successfully.
```
git pull
source ./env.sh
make -C flow
```

maliberty, 2024-03-06:
Please reopen if still an issue

closed 2024-03-06 by maliberty