The-OpenROAD-Project/OpenROAD-flow-scripts #1767: DRT not converging after 64 iterations

gkamendje, 2024-01-18:
### Subject

[Stage]: Detail Router.

### Describe the bug

After approximately 5 iterations, the number of violations is down to 1 or two (spacing violations on M2). However, although these violations can be easily fixed manually, DRT most of the time runs until the 64th iteration and most of the time fails to fix the single violation remaining. The behavior has been reproduced on Ubuntu 22 as well as on CentOS7 machines.  Sometime however, DRT will run until the ~50th iteration or so and fix the violation. 

### Expected Behavior

DRT should be able to fix those violations (specially when thy can be easily fixed manually) 

### Environment

```shell
Git commit: 50feb216d19d6922e2de10daea82437244ed5fa0
kernel: Linux 5.15.133.1-microsoft-standard-WSL2
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.28.0-rc2
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11900-g50feb216d
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 103 lines cut ...]
    Completing 10% with 1 violations.
    elapsed time = 00:00:00, memory = 960.48 (MB).
    Completing 20% with 1 violations.
    elapsed time = 00:00:00, memory = 960.48 (MB).
    Completing 30% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 40% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 50% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 60% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 70% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 80% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 90% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
    Completing 100% with 1 violations.
    elapsed time = 00:00:01, memory = 960.48 (MB).
[INFO DRT-0199]   Number of violations = 1.
Viol/Layer      METAL2
SpacingTableTw       1
```
```

vijayank88, 2024-01-18:
@gkamendje 
Have you restricted drive access ?

gkamendje, 2024-01-18:
I have updated the permission. Please try again.
G

osamahammad21, 2024-01-18:
The config file is missing

gkamendje, 2024-01-18:
strangely enough the issue script is not putting everything in the tar archive. I have added the following file https://drive.google.com/file/d/129Iw7Lt5ZcEny8rx2vbRk3wjI9IrZAwK/view?usp=sharing that contains the config.mk, the def file used and some other stuff

osamahammad21, 2024-01-21:
Also the config file for the platform is missing

gkamendje, 2024-01-22:
@osamahammad21 I have added some platform files here.
https://drive.google.com/file/d/12JwyZ47u60qxKv96f7k1_-Rsa8NP-aQu/view?usp=sharing

osamahammad21, 2024-01-22:
The placement of the pins is making it hard for the detailed router to reach them. The pins are of min width and their centers are not aligned with the tracks. is there a reason you're placing them that way?

gkamendje, 2024-01-22:
There is no particular reason why they are like that. The position/size of the pins was given by the DEF file provided by the analog team that owns the other macros the block communicates with. This can easily be changed if that improves the convergence.
Could you please point me to  warnings/errors that highlights this issue? 
Is there a way to visualize the tracks in question?

osamahammad21, 2024-01-22:
The warnings look like this:
[WARNING DRT-0422] No routing tracks pass through the center of Term o_eeprom_read
This pin is of size 280x280 which is the min width of metal 4:
[image]

As you can see, the pin's center is not aligned with the routing track. This results in a Non sufficient metal violation when the net tries to access the pin vertically. 

I took the liberty of changing all Metal4 pins sizes to 350x350 and it converged after 4 iterations.

PA finds a horizontal access point for that pin, but DRT is short on understanding the access point direction. It sees no issue with accessing it vertically which is violating. I am working out a fix for that issue but that shouldn't stop you from modifying pin placement

rovinski, 2024-01-22:
FWIW, this is a problem commercial routers struggle with too. I had a tapeout last year where this was an issue (analog team gave us a block with many off-grid pins, router couldn't create clean connections). We asked the analog team to align the pins to the routing and it did significantly better, but still had errors due to congestion. We then asked them to spread out some of the pins (we had tens of thousands of pins) so they weren't at 100% density, and then the violations finally went away.

gkamendje, 2024-01-23:
@osamahammad21 indeed changing the pin size for M4 pins does lead to a very quick convergence. However, changing the size of all pins M1 to M4 to 350x350 shows the issue again. After 31 iterations there is only 1 M2 spacing violation remaining but it takes up to 64 iterations to fix that single violation. 
Moreover, I do see so strange pin access patters such as this one 
[image]

@rovinski any idea why `make detail_route_issue` does not include the platform and design configuration files in the resulting tar archive anymore?

rovinski, 2024-01-23:
> @rovinski any idea why make detail_route_issue does not include the platform and design configuration files in the resulting tar archive anymore?

@vvbandeira

osamahammad21, 2024-01-23:
The M2 access is not strange because the center of the M2 pin is not aligned with the M2 track. If the net tries to access it directly vertically, it will get a Non sufficient metal violation. So it tries to access it horizontally by doing this weird hock shape.

I am working on a fix that makes such pin access a bit smoother.

As for the remaining violation, it went under my nose. One of the cells is placed very close to a PDN via. It makes it impossible for the router to access this cell pin without creating a spacing violation with the M2 PDN via enclosure. There is an option in the detailed_route command to ignore such violations and remove the violating PDN vias at the end of routing, do you think this could be a good solution or you don't want to lose any of the PDN vias?

vvbandeira, 2024-01-23:
> > @rovinski any idea why make detail_route_issue does not include the platform and design configuration files in the resulting tar archive anymore?
> 
> @vvbandeira

#1771 should fix it, just waiting to see the impact on file size before merging

gkamendje, 2024-01-24:
> The M2 access is not strange because the center of the M2 pin is not aligned with the M2 track. If the net tries to access it directly vertically, it will get a Non sufficient metal violation. So it [...]

Getting rid of the violating PDN via could certainly be an option to explore. Could you please remind me how to activate this option? 
It is still not quite clear to me why moving away from min M4 pins only successfully masks the issue.

osamahammad21, 2024-01-28:
This is purely placement luck. Matt pointed out that this is should be an issue in the placer to use a site so close to a PDN via, which makes hard for the router to reach the placed cell. but again, this a design with a 100% utilization, so I am not sure if we're asking too much from the placer.
As for repairing the PDN vias option, I found out that it only works on removing vias if the PDN grid is 2 consecutive layers. I am finalizing a PR that should work on repairing a range of via layers between the 2 PDN layers. it should be ready by the end of the day.

osamahammad21, 2024-02-11:
After discussion, we found that removing pdn vias is a bit risky and going to lead in IR drop. you said when you started the issue that the DRC can be easily fixed, Could you specify how you could manually fix it.
[image]

gkamendje, 2024-02-12:
In our use case, the GDS file is imported into Virtuoso by the analog Team.
The violating nets are manually re-routed within Virtuoso.

maliberty, 2024-02-12:
I'm interested in how these "easily fixed manually" violations are resolved in Virtuoso.  What do you change to make these legal?   Are you moving the cells?  Changing the pdn?  Neither seems trivial and I'm wondering if there is something easy we aren't seeing.

gkamendje, 2024-03-04:
@maliberty feedback from analog team is that they are essentially moving cells around. Agreed that this might not be "trivial" at all! I had certainly misunderstood the statement of the analog team when they said that they had been able to "easily" fix the issue.

maliberty, 2024-03-04:
Better handling of the interaction between placement and pdn vias is an area for improvement.  It isn't 'easy' but we have it on the list.  The placement engine needs some 'mini-drc/pin-access' check to look for pins that are being blocked.

rovinski, 2024-07-24:
@maliberty @osamahammad21 closed?

osamahammad21, 2024-09-19:
@maliberty Should this be closed?

closed 2024-09-19 by maliberty