The-OpenROAD-Project/OpenROAD-flow-scripts #1775: sudo ./setup.sh fails in eqy

oharboe, 2024-01-24:
### Subject

[Build] for compilation-related issues.

### Describe the bug

```
$ sudo ./setup.sh 
[deleted]
Cloning into 'eqy'...
remote: Enumerating objects: 1729, done.
remote: Counting objects: 100% (213/213), done.
remote: Compressing objects: 100% (150/150), done.
remote: Total 1729 (delta 85), reused 97 (delta 60), pack-reused 1516
Receiving objects: 100% (1729/1729), 1.01 MiB | 4.72 MiB/s, done.
Resolving deltas: 100% (1106/1106), done.
Note: switching to '8327ac7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 8327ac7 Merge pull request #49 from YosysHQ/jix/submodule_example
yosys-config --build src/eqy_combine.so  src/eqy_combine.cc 
make: yosys-config: No such file or directory
make: *** [Makefile:26: src/eqy_combine.so] Error 127

```

### Expected Behavior

Should work

### Environment

```shell
$ git describe origin/master 
v3.0-79-g61077866

kernel: Linux 6.5.0-14-generic
os: Ubuntu 23.10 (Mantic Minotaur)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11974-gefb1f59c3
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
[... 108 lines cut ...]
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /home/oyvind/OpenROAD-flow-scripts/dependencies/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /home/oyvind/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /home/oyvind/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/oyvind/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.2aTWsULvK3
```

### To Reproduce

See above

vvbandeira, 2024-01-25:
@oharboe do you have any other version of yosys installed on your machine? maybe via apt?

oharboe, 2024-01-25:
> @oharboe do you have any other version of yosys installed on your machine? maybe via apt?

yes, via apt.

oharboe, 2024-01-25:
> Let me see if I can reproduce it, but I think they might be conflicting.

It should be a very common situation, having apt installed yosys...

vvbandeira, 2024-01-25:
Yes, I can reproduce it if I install `yosys` via `apt`, I got version `0.9` from 2019 which is quite old. We could install `yosys` regardless, but maybe a warning would be better?

oharboe, 2024-01-25:
> Yes, I can reproduce it if I install `yosys` via `apt`, I got version `0.9` from 2019 which is quite old. We could install `yosys` regardless, but maybe a warning would be better

No warning, there is nothing wrong or that the user should do....

The setup procedure should be unaffected by other yosys versions in the path and support side by side install.

oharboe, 2024-03-20:
Works now even if yosys is installed in the system. I'll reopen if the problem resurfaces. I think I saw a fix that went in.

closed 2024-03-20 by oharboe