The-OpenROAD-Project/OpenROAD-flow-scripts #1786: [WARNING DRT-0422] No routing tracks pass through the center of Term  clock/io_*

oharboe, 2024-02-01:
### Subject

[Stage]: Detail Router.

### Describe the bug

untar  [detail-route.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/14123805/detail-route.tar.gz)

```
$ NUM_CORES=16 ./run-me-mock-array_Element-asap7-base.sh 
OpenROAD v2.0-12049-gb0d81bbb0
[deleted]
[INFO DRT-0150] Reading design.

Design:                   Element
Die area:                 ( 0 0 ) ( 43200 43200 )
Number of track patterns: 32
Number of DEF vias:       0
Number of components:     7823
Number of terminals:      528
Number of snets:          2
Number of nets:           1329

[WARNING DRT-0422] No routing tracks pass through the center of Term clock
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[0]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[10]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[11]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[12]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[13]
...
```

### Expected Behavior

No error or warning

### Environment

```shell
OpenROAD v2.0-12049-gb0d81bbb0
```

### To Reproduce

See above

maliberty, 2024-02-01:
Is this different from the usual mock array which has manual macro placement?

oharboe, 2024-02-01:
> Is this different from the usual mock array which has manual macro placement?

No, this happens on master. I dont recall seeing it, but it is a long time since I studied mock array.

I saw it on a gigabyte example and thought it was better to report on this tiny testcase.

maliberty, 2024-02-01:
The warning itself was added some months ago so it may have always been there.  Is your larger test case manually placed or mpl2?  This is more of a macro placement issue that the router is just warning about.

oharboe, 2024-02-01:
> The warning itself was added some months ago so it may have always been there. Is your larger test case manually placed or mpl2? This is more of a macro placement issue that the router is just warning about.

There is no macro placement involved in this test case. This happens on the Element level, not in the array.

maliberty, 2024-02-01:
It sounds like a pin placement issue then as the pins should be on track.

eder-matheus, 2024-02-01:
All the pins are properly placed and aligned with the routing tracks.

Vertical pins:
[image]

Horizontal pins:
[image]

I also checked the manufacturing grid, and it's also correct.

maliberty, 2024-02-01:
Is the warning erroneous then?

eder-matheus, 2024-02-01:
> Is the warning erroneous then?

That's what it seems. I'm looking at the DRT code to understand why the warnings, but it seems to be a bug.

closed 2024-03-16 by maliberty