The-OpenROAD-Project/OpenROAD-flow-scripts #1788: GRT guides high congestion

osamahammad21, 2024-02-04:
### Subject

[Stage]: Global Router.

### Describe the bug

GRT is overusing gcells on top of a macro blockage. In a new branch I made to fix estimated cost calculation in drt [TR-fix-estCost](https://github.com/The-OpenROAD-Project-private/OpenROAD-flow-scripts/tree/secure-TR-fix-estCost), asap7/aes-block is failing with 1 DRV. When I investigated further I found that the issue is with a net routing near a macro blockage. Taking it one step back, track assignment seems to assign the net a very bad track on M2 going through the macro blockage.
[image]
The gcell in this picture, along with other aligned gcells right to it, has 5 nets on M2 with only 4 tracks available. The track that is exactly aligned on the top border is not used because of this line:
https://github.com/The-OpenROAD-Project-private/OpenROAD/blob/abecb770baf1a79201cb5d4bab827b33a4537d05/src/drt/src/ta/FlexTA_assign.cpp#L592
So this leaves it with only 4 tracks and 5 nets, which makes it use a track going through the M2 blockage.
The gcells above that row of the attached gcell picture are not congested to justify GRT's use of those gcells.

### Expected Behavior

GRT better guides

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-39-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11958-gabecb770b
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 82 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.rlCwfCA1Xm
```

### To Reproduce

run asap7/aes_block on https://github.com/The-OpenROAD-Project-private/OpenROAD-flow-scripts/tree/secure-TR-fix-estCost
The location of the GCell is [ 22680, 79920, 23220, 80460 ]

osamahammad21, 2024-02-04:
@maliberty @eder-matheus

maliberty, 2024-02-05:
This seems like an ambiguous case where the track lies right on the gcell boundary - is that correct?

osamahammad21, 2024-02-05:
The upper/right border is not owned by the current worker, so yeah. Tho I don’t see any reason for the grt to use those gcells when there is more room elsewhere.

maliberty, 2024-02-05:
There is an available track, drt just isn't using it.  It seems like a definitional issue between grt & drt

eder-matheus, 2024-02-05:
It's also strange that GRT is using all five tracks. Do we have resource adjustments in this design? GRT should be able to detect 4~5 available tracks, but this value should be reduced with the user-defined resource adjustments. I will look at this design to understand what's being calculated there.

closed 2024-02-16 by maliberty